dgatf/msrc #7: Not Working with castle ESC

emilioeer, 2020-10-25:
I am trying to use it on a Talon 90, Horus X12 Express, I have some issues with the sensors not reading correctly, using with a 6s setup Voltage varies between 24,23,22v. When the motor is running i get values jumping all over the place like, 13, 18, 16 ,25,23, all at the same throttle input. I am somewhat new to Opentx so i dont know what i can do to fix this issue.Also i am using an arduino nano is what i had on hand for this project. I can try to help solve this issue

emilioeer, 2020-10-27:
So i tried using a proper arudino pro mini (5v 16mhz) and now using a Phoenix Edge 80A HV ESC hoping that in this way the telemetry for castle was going to work. however i still get really unstable data. using 6s voltage jumps around 21 to 25 v randomly. i dont know were to start to try to help and fix the issue.

dgatf, 2020-11-10:
Sorry for the delay...

What about the other sensors readings: BEC voltage, temperature and RPM? Are they correct?

Nothing to do with opentx, except for the lua config

Arduino nano should be ok (not tested though) as long as you are not using serial Hobbywing protocol. See https://www.helifreak.com/showthread...09#post8152909

emilioeer, 2020-12-15:
Update on this isse, and sorrt for taking so long.

I tested with a arduino pro mini and i am fairly certain that all my connections are ok. attaching and image of my current setup using a breadboard.

[image]

Changed sensor averaging to 1 see iamge below.

[image]

It is still acting weird, not sure if the data i am getting is normal or really there is some issue

Regarding the questions all values for the sensor seem to randomly send random values, such like temperature jumps from 23 to 27 to 28 back to 24 etc. Alos RPM jumps from 0 to 1600 or 2600, equaly voltage varies from 21  to 20 to 24 etc. I tried logging a session to help you visualize the data I am getting, however trying to open the log in Open Tx companion says that there are invalid lines inside the file. I tested first making a log without the MSRC sensors (session 1 log) and then discovering all MSRC sensors (this should be session 2 inside the same log file. Ill attach the file so that you can see how the data changes. I am using Open TX 2.3.9 with a Horus X10, Receiver is a R8 Pro from the new Archer line running ACCESS 2.1.0 for TX and 2.1.5 for Receiver.

The File attached is a CSV file, i changed the file type so that i was able to upload it.

[Model17-2020-12-14.txt](https://github.com/dgatf/msrc/files/5692855/Model17-2020-12-14.txt) 

se below a short video of how values change inside the sensors screen

[newvideo.zip](https://github.com/dgatf/msrc/files/5692954/newvideo.zip)

dgatf, 2020-12-19:
There was an issue with Timer2 when reading telemetry values close to its maximum. Let me know if this happens again, in particular when any telemetry value is close to the maximum 

Fixed with 069e17c

This is a pro Mini 5V (16mhz) with a Castle ICE2HV80: [video](https://youtu.be/Uiunv6LJZbI)

closed 2020-12-19 by dgatf