dgatf/msrc #8: Project won't fit on arduino pro mini.

VitRuzicka, 2020-12-29:
Hi,
I tried to compile your project and arduino says, that project won't fit... it's 106% of capacity. Arduino Mini Pro here 5V 16MHz 328P. I don't need so many fancy features (specific temperature sensors), so can I shrink it down easily without heavy modifications?
Thanks in advance,
Vit

dgatf, 2020-12-30:
That happened during a short period. After removal of BMP180, it is using 27374 of 30720 bytes (89%)

Please download latest code

closed 2020-12-30 by dgatf

VitRuzicka, 2020-12-30:
Hi, I've downloaded the latest code (an hour ago). But still won't fit. Same problem. Should i delete BMP180 files? What about links in other .h and .cpp files? I eventually found another library, so I build my sensor on that, but for someone facing the same issue, this might be helpful. 
Thanks in advance,
Vit

dgatf, 2020-12-31:
I can't reproduce the issue. Be sure you are using the latest code. You can delete bmp180 files, but is not needed. If line 114 in msrc.h is `#include "bmp180.h"`, you are not using latest code and won't fit

VitRuzicka, 2020-12-31:
Hi, 
this is screenshot of the current (i believe 12days old) code from this github repo. Bmp isn't on 114 but 115. Sorry for bothering you again, but the lib just looks very promising.
Thanks in advance, 
Vit
[image]

dgatf, 2020-12-31:
BMP180 were removed by 19/12. That seems to be the latest code

Please could you provide:

- Compilation environment and version (Arduino IDE?)
- OS System
- Complete output text when compiling

VitRuzicka, 2021-01-03:
Hi,
sorry for answering late, didn't get any notification.
Here's the output, it's partially in czech hope it makes sense to u.
```

/usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/vit/.arduino15/packages -hardware /home/vit/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/vit/.arduino15/packages -libraries /home/vit/Arduino/libraries -fqbn=archlinux-arduino:avr:pro:cpu=16MHzatmega328 -ide-version=10813 -build-path /tmp/arduino_build_6159 -warnings=none -build-cache /tmp/arduino_cache_817573 -prefs=build.warn_data_percentage=75 -verbose /home/vit/Stažené/msrc-master/msrc/msrc.ino
/usr/share/arduino/arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/vit/.arduino15/packages -hardware /home/vit/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/vit/.arduino15/packages -libraries /home/vit/Arduino/libraries -fqbn=archlinux-arduino:avr:pro:cpu=16MHzatmega328 -ide-version=10813 -build-path /tmp/arduino_build_6159 -warnings=none -build-cache /tmp/arduino_cache_817573 -prefs=build.warn_data_percentage=75 -verbose /home/vit/Stažené/msrc-master/msrc/msrc.ino
Using board 'pro' from platform in folder: /usr/share/arduino/hardware/archlinux-arduino/avr
Using core 'arduino' from platform in folder: /usr/share/arduino/hardware/archlinux-arduino/avr
Detecting libraries used...
/usr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR -I/usr/share/arduino/hardware/archlinux-arduino/avr/cores/arduino -I/usr/share/arduino/hardware/archlinux-arduino/avr/variants/eightanaloginputs /tmp/arduino_build_6159/sketch/msrc.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/usr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR -I/usr/share/arduino/hardware/archlinux-arduino/avr/cores/arduino -I/usr/share/arduino/hardware/archlinux-arduino/avr/variants/eightanaloginputs /tmp/arduino_build_6159/sketch/bmp.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
/usr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR -I/usr/share/arduino/hardware/archlinux-arduino/avr/cores/arduino -I/usr/share/arduino/hardware/archlinux-arduino/avr/variants/eightanaloginputs -I/usr/share/arduino/hardware/archlinux-arduino/avr/libraries/Wire/src /tmp/arduino_build_6159/sketch/bmp.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/usr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR -I/usr/share/arduino/hardware/archlinux-arduino/avr/cores/arduino -I/usr/share/arduino/hardware/archlinux-arduino/avr/variants/eightanaloginputs -I/usr/share/arduino/hardware/archlinux-arduino/avr/libraries/Wire/src /tmp/arduino_build_6159/sketch/bmp280.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/usr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR -I/usr/share/arduino/hardware/archlinux-arduino/avr/cores/arduino -I/usr/share/arduino/hardware/archlinux-arduino/avr/variants/eightanaloginputs -I/usr/share/arduino/hardware/archlinux-arduino/avr/libraries/Wire/src /tmp/arduino_build_6159/sketch/bn220.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/usr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR -I/usr/share/arduino/hardware/archlinux-arduino/avr/cores/arduino -I/usr/share/arduino/hardware/archlinux-arduino/avr/variants/eightanaloginputs -I/usr/share/arduino/hardware/archlinux-arduino/avr/libraries/Wire/src /tmp/arduino_build_6159/sketch/device.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/usr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_PRO -DARDUINO_ARCH_AVR -I/usr/share/arduino/hardware/archlinux-arduino/avr/cores/arduino -I/usr/share/arduino/hardware/archlinux-arduino/avr/variants/eightanaloginputs -I/usr/share/arduino/hardware/archlinux-arduino/avr/libraries/Wire/src /tmp/arduino_build_6159/sketch/escCastle.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
[... cut ...]
Projekt zabírá 32680 bytů (106%)  úložného místa pro program. Maximum je 30720 bytů.
Globální proměnné zabírají 925 bytů (45%)  dynamické paměti, 1123 bytů zůstává pro lokální proměnné. Maximum je 2048 bytů.
Projekt je moc velký; na http://www.arduino.cc/en/Guide/Troubleshooting#size naleznete typy jak projekt zmenšit.
text section exceeds available space in board
Nastala chyba při kompilaci u desky Arduino Pro or Pro Mini.
```

I am using Arduino ide 1.8.13 (latest from aur) on Linux Manjaro 5.9.11

Thanks in advance,
Vit

dgatf, 2021-01-04:
I've tested with Arduino Ide 1.8.13 and Ubuntu and I get 28112 bytes (91%)

Have you modified the code (eg. uncommented any #define DEBUG, which increases the required flash). Otherwise could be something related to your environment. You could try in other computer/OS