dgatf/msrc #12: Castle Link: erratic readings

twistedwings, 2021-01-04:
I have made a telemetry log with my heli on the bench.
RPM=0, lipo voltage 23.0V all the time.
The telemetry values look strange.

[image]

Here is the log:
[https://drive.google.com/file/d/1Tes9JJHe6Kpq8qTiYFw9zbXp99co7ER8/view?usp=sharing](url)

Radiomaster TX16S
OpenTX 2.3.10

dgatf, 2021-01-04:
Are you using an Arduino 8mhz (3.3v) or 16mhz (5v)?

twistedwings, 2021-01-04:
16mhz (5v)

dgatf, 2021-01-04:
Ok. Better with 16mhz

In escCastle.h I've added in line 21 `#define FIXED_CALIB` and `#define DEBUG_CALIB`

Please make the following tests:

Test 1: Uncomment `#define FIXED_CALIB` and check the telemetry log again. This fixes the telemetry calibration. Values should be more stable now

Test 2: Uncomment `#define DEBUG_CALIB` and connect TX to a serial monitor at 115200 baud. Connect Arduino and heli GNDs (do not connect ttl vcc to the arduino) and post the output

twistedwings, 2021-01-04:
I will give it a try.
Should  #define FIXED_CALIB be uncommented for Test 2 too ?

dgatf, 2021-01-04:
It doesn't matter

twistedwings, 2021-01-04:
Test 1: I don't see an improvement
[image]
https://drive.google.com/file/d/1QFCHxxZyTn3DXSFlbpYtlt01Juk3M-wP/view?usp=sharing

Test 2:
https://drive.google.com/file/d/1YpEXI038_ywlZ04W7kG70JUPnYJa2uek/view?usp=sharing

dgatf, 2021-01-04:
I addition to the esc, do you have enabled other sensors in the Arduino config?

twistedwings, 2021-01-05:
no, castle esc is the only sensor.

dgatf, 2021-01-05:
Castle telemetry involves measuring time very precisely. Atmega328 has only one 16-bit timer (already used to produce the inverted pwm signal to the esc) which forced me to measure the telemetry with a pin interrupt (less precise as to when it happened) rather than the precise timer interrupt and Test 1 confirms that telemetry readings is sometimes delayed by other interrupts (software serial for smartport makes an intensive use of them). The deviations in Test 1 are only to one side (increasing the values not also decreasing as before the test). The difference is that the calibration scaler is fixed rather than adjusted every 10 frames and delayed scaler means lower telemetry values. Unfortunately until next month or so I won't be able to test with my castle esc, but I doubt if something can be improved in this regard. So all in all castle telemetry is on the limit (or a bit beyond) of what atmega328 can do

I suggest the following:

- Leave `#define FIXED_CALIB` uncommented as it produces less errors given that 1 of 2 variables are fixed. I will probably make this the default behaviour
- Increase all refresh rates to 1000 in order to reduce the chip load
- Increase averaging elements to 16

I'll be porting the project to STM32 soon which will handle this with no issues

dgatf, 2021-01-05:
Pass thru is not possible, because castle esc requires an inverted pwm signal when activating the telemetry

To invert the pwm signal from the RX it is needed:

- Timer interrupt (ICP1) to measure the pulse length
- 16 bit timer to produce the inverted pwm signal

I don't think there is another way without using the 16-bit time, but suggestions are very welcome

twistedwings, 2021-01-05:
IMHO there is no need to measure the pulse length or to generate a pwm signal.
The rc output  (to esc) should be the same as the rc input (from receiver) but inverted.
If the rc input goes high you'll get an interrupt and can switch the rc output to low.
If the rc input goes low you'll get an interrupt and can switch the rc output to high.
So for this "rc pass thru" no timer is needed at all and latency would be almost zero.

dgatf, 2021-01-05:
That has the same issue as the telemetry, it relies on a pin interrupt, that fires and interrupt routine which can be delayed in the same way, resulting in a unstable signal increasing for small periods the motor throttle. Timer interrupt is needed as it keeps the time when it was fired despite the routine will be processed later. Pwm is needed as is the only way that guarantees the length of the pulse, otherwise same delay may occur. It's not latency the issue (indeed current latency is up to 20ms, but not an issue), is the accuracy of the signal. The delays producing this issue are around 0.1ms only

twistedwings, 2021-01-05:
OK, I see.
Then I hope to see a STM32F103 port someday.
An alternative solution might be a hardware inverter for the rc input.

twistedwings, 2021-01-05:
Maybe the SBUS solution is worth thinking about since
many FrSky receivers have an inverted SBUS signal.

dgatf, 2021-01-15:
Support for ATMega328PB and ATMega2560 has been added, though not tested and not merged to master yet. Those boards solves the issue of the erratic reading for Castle ESC. See [comment](https://github.com/dgatf/msrc/issues/13#issuecomment-760765542)

closed 2021-01-16 by dgatf