dgatf/msrc #13: (Feature Request) Add SoftSerial to add a second serial port to the Arduino Pro Mini

stefaanbolle, 2021-01-05:
I have a feature request which might be useful when considering creation of an all-in-one sensor board.
SoftSerial library is available for the Arduino Pro Mini, so maybe it might be possible to integrate softserial to unused pins to allow both GPS and ESC via a serial connection?
Any thoughts?

dgatf, 2021-01-07:
Softwareserial is already in use becuase smartport uses an inverted serial signal. This avoids the use of external inverter circuit as is not possible to invert the hardware serial logic in the Atmega328. It would be feasible to have multiple serial ports with softwareserial but you can listen to one at the time. This won't work in this project as serial ports are constantly communicating and some information will be lost. A port for STM32F103 with multiple serial ports is on the way

closed 2021-01-08 by stefaanbolle

stefaanbolle, 2021-01-10:
Can you elaborate a little more on "A port for STM32F103 with multiple serial ports is on the way"?
What board should I already purchase and when is the code expected? Glad to help to test.

dgatf, 2021-01-10:
I'm adding support for ATMega2560 and STM32F103C8. Code is in this [branch](https://github.com/dgatf/msrc/tree/Add_atmega2560_and_stm32f103c8) but untested. I won't be able to test it until few weeks ahead. If you want to try and test the new boards, choose the ATMega2560

Advantages of new boards:

- You can use ESC serial telemetry and GPS at the same time
- You get accurate telemetry with Castle ESC. See issue #12

If you don't need any of the previous better choose the Pro Mini. It is still supported and it's cheaper, lighter and smaller

stefaanbolle, 2021-01-11:
Thanks!
I have a board on order. Want to test it out :-)

dgatf, 2021-01-15:
Update:

ATMega2560 implemented in this [branch](https://github.com/dgatf/msrc/tree/Add_atmega2560)
STM32 has been discarded
Found a better board, Pololu ATMega328PB, which is the same size and weight as the Arduino Pro Mini and almost same pins. Implemented [here](https://github.com/dgatf/msrc/tree/Add_atmega328pb)