dgatf/msrc #18: Syntax error if #define CONFIG_ESC_PROTOCOL PROTOCOL_CASTLE

twistedwings, 2021-01-23:
Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Pololu A-Star 328PB, 5V, 16MHz"

sketch\srxl.cpp: In member function 'void Srxl::send()':

srxl.cpp:78:40: error: 'xbusEsc' was not declared in this scope

         memcpy(buffer + 3, (uint8_t *)&xbusEsc, sizeof(xbusEsc));

                                        ^~~~~~~

sketch\srxl.cpp:78:40: note: suggested alternative: 'Xbus_Esc'

         memcpy(buffer + 3, (uint8_t *)&xbusEsc, sizeof(xbusEsc));

                                        ^~~~~~~

                                        Xbus_Esc

xbus.cpp:4:16: error: 'Xbus_Esc Xbus::xbusEsc' is not a static data member of 'class Xbus'

 Xbus_Esc Xbus::xbusEsc;

                ^~~~~~~

sketch\xbus.cpp: In static member function 'static void Xbus::i2c_request_handler()':

xbus.cpp:56:36: error: 'xbusEsc' was not declared in this scope

         memcpy(buffer, (uint8_t *)&xbusEsc, sizeof(xbusEsc));

                                    ^~~~~~~

sketch\xbus.cpp:56:36: note: suggested alternative: 'Xbus_Esc'

         memcpy(buffer, (uint8_t *)&xbusEsc, sizeof(xbusEsc));

                                    ^~~~~~~

                                    Xbus_Esc

sketch\xbus.cpp: In member function 'void Xbus::update()':

xbus.cpp:128:5: error: 'xbusEsc' was not declared in this scope

     xbusEsc.RPM = __builtin_bswap16(esc.read(CASTLE_RPM));

     ^~~~~~~

sketch\xbus.cpp:128:5: note: suggested alternative: 'Xbus_Esc'

     xbusEsc.RPM = __builtin_bswap16(esc.read(CASTLE_RPM));

     ^~~~~~~

     Xbus_Esc

exit status 1

'xbusEsc' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

dgatf, 2021-01-23:
Bug fixed

Please try now

twistedwings, 2021-01-23:
syntax error is gone

closed 2021-01-23 by twistedwings

dgatf, 2021-01-24:
There was a bug that was causing to read 0 on most of the values. See #19

You'll need to flash with latest code