dgatf/msrc #19: Spektrum X-bus  --  Hobbywing HV V4 issue

J-charles-C, 2021-01-23:
Hello,

I uploaded the firmware and the arduino is pafaitement recognized in the telemetry in "auto-config".

In config.h I have activated "#define CONFIG_ESC_PROTOCOL PROTOCOL_HW_V4_HV".

however, when I plug the "hobbywing rpm" wire into my Arduino, I have no data going back.

cordially

dgatf, 2021-01-24:
Bug fixed

Please try again

J-charles-C, 2021-01-24:
Ok i test.

thanks

J-charles-C, 2021-01-24:
I, 

not better

best

J-charles-C, 2021-01-24:
I just reread the readme and there may be some problem with the RX and arduino with USB port.

I have this model and the ESC is only plugged into the RX0 port. Is it well connected?

https://www.amazon.fr/gp/product/B0722YYBSS/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

dgatf, 2021-01-24:
Yes, that's the problem. The serial signal from the esc is not strong enough for a serial port shared with usb on the arduino board

J-charles-C, 2021-01-24:
is there a solution ?

dgatf, 2021-01-24:
Yes. I've added more flexibility to use software serial with any serial interface. Now you should be able to select softwareserial for the ESC. This can be done as you are using XBUS. With smartport this wouldn't be feasible as software serial is required for smartport to invert the serial signal

Update line 71 from config.h as follows:

`#define ESC_SERIAL softSerial` and connect serial to pin 7 (_PIN_SOFTSERIAL_RX_)

J-charles-C, 2021-01-25:
I,

Thanks ok, i test today  with D7.

best

J-charles-C, 2021-01-26:
Hello,

sorry, I couldn't test yesterday because i work.

I used the version from yesterday. I welded in D7, modified line 71.

So I no longer have the tirests but I now have a value of 0 everywhere.

If it is worth testing, can you give me the line to modify on today's version?
cordially

dgatf, 2021-01-26:
Modify line 30 in _config.h_:
`#define ESC_SERIAL softSerial`

The reason behind is that ESC serial uses 3.3v logic level whereas your Arduino board is 5v. Thus a 3.3v (8mhz) board is recommended. There are versions of Pro Mini and Pro Micro for 3.3v (8mhz)

As a workaround you can try with the software serial (by modifying config.h as explained before) which usually gives better results with this issue

dgatf, 2021-01-26:
To verify that the ESC can be read you can use the sniffer.ino sketch (under tools) and connect ESC data to pin 10, ESC GND to GND and see debug on serial monitor at 19200 baud

J-charles-C, 2021-01-30:
Good evening,

I tested without success. I don't know how to use the debug.

So, is this model the right one?

https://www.amazon.fr/Microcontrôleur-ATmega328P-développement-compatible-Arduino/dp/B01DUSDQQ8/ref=sr_1_2?__mk_fr_FR=ÅMÅŽÕÑ&dchild=1&keywords=arduino+mini+3.320&qid=4444+mini+3.320&qid=

Thank you

dgatf, 2021-01-30:
> I tested without success

Download and flash the latest code

Check that ESC data is connected to pin 7

Check that line 30 in config.h is:

`#define ESC_SERIAL softSerial`

> I don't know how to use the debug.

Also connect ESC GND to Arduino GND

Enable debug mode in config.h uncommenting line 107:

`#define DEBUG_ESC`

Open Serial Monitor in Arduino IDE at 115200bps. Check the output

> So, is this model the right one?

Yes, but I recommend to try debug mode before to discard any other problem

J-charles-C, 2021-01-30:
i,

On serial monitor, i have only :

⸮⸮⸮⸮⸮
DEBUG
V0.8.0

dgatf, 2021-01-30:
Then if the connections are ok, definitely your board is not reading the ESC. Go buy the Mini 3.3v

I'm still surprised that is not working even with softwareserial. You could also post a picture of the connections to check if everything is ok

J-charles-C, 2021-01-30:
[image]

dgatf, 2021-01-30:
Ok, looks good. Last check...

Upload the sketch under tools folder [sniffer](https://github.com/dgatf/msrc/tree/master/tools/sniffer)

Be sure you change the pin in line 10:

`#define PIN_SNIFFER 7`

Connect the ESC to the Arduino and open a serial monitor at 19200bps and check the output. If there is no output, absolutely sure your board is not reading the ESC

J-charles-C, 2021-01-30:
Ok,

INIT
20:13:19.993 -> ⸮  ⸮  ⸮⸮  ⸮ ⸮⸮⸮⸮⸮  ⸮ 

Nothing else

dgatf, 2021-01-30:
Which firmware version is installed in the ESC? You may try with another version. See [post](https://www.helifreak.com/showthread.php?t=835243&page=14#post8244995)

J-charles-C, 2021-01-31:
I have the 160A HV with the latest firmware.

thanks

dgatf, 2021-01-31:
Just check it doesn't have the VBAR in the title of the firmware version. See [post](https://www.helifreak.com/showthread.php?p=8245353#post8245359)

J-charles-C, 2021-02-01:
Good evening. The firmware for Vbar does not exist for my esc. I order the arduino in 3.3v. thanks

closed 2021-02-01 by J-charles-C