dgatf/msrc #22: Atmega2560 pins

stefaanbolle, 2021-01-28:
In your table I notice:
ESC serial 	RX1
GPS 		RX2
PWM out 	7
S.port/SRXL	4/12
XBUS/sensor SDA20
XBUS/sensor SCL21

What pins do you mean on a Mega2560 Pro Mini board because I do not find these pin references on my board
[https://pg-cdn-a2.datacaciques.com/00/NDAy/19/01/02/r4tvb3u808096tbn/ab2ddc6ef3b8ad95.jpg](url)

stefaanbolle, 2021-01-28:
Found the pin mapping I guess
[https://robotdyn.com/pub/media/0G-00005642==MEGA-PROMINI-ATmega2560/DOCS/PINOUT==0G-00005642==MEGA-PROMINI-ATmega2560.pdf](url)

dgatf, 2021-01-28:
I've improved the pin table with usual pin numbers

stefaanbolle, 2021-02-16:
I am using a Radiomaster R161. Smart port wire and VCC and GND go to the Mega2560 pin VCC/GND and D4/D12 with pullup). S.Bus + VCC and GND go to FBL
msrc.lua does not detect the board and no sensors are detected. Was working with 328P pro micro before. Are ports D4 and D12 (4 and 12) the correct ones for smartport?

dgatf, 2021-02-17:
There was an error on the readme. Fixed. Correct pins for ATmega2560 are 10 and 15

[image]

stefaanbolle, 2021-02-17:
Thanks, will try it out

stefaanbolle, 2021-02-17:
Indeed, much better now :-)

stefaanbolle, 2021-02-17:
Is it correct that BN-220 GPS TX goes to pin D17 on Mega2560? I do not get a GPS signal (VCC is connected to 5V and GND to GND)

dgatf, 2021-02-18:
Yes, D17 is the correct pin for GPS. Be sure you have the default baud rate 9600

I'll check GPS and BMP280 for ATmega2560 in the following days

stefaanbolle, 2021-02-18:
> Yes, D17 is the correct pin for GPS. Be sure you have set the default baud rate 9600
> 
> I'll check GPS and BMP280 for ATmega2560 in the following days

Do I have to set baudrate in your code somewhere?

dgatf, 2021-02-18:
If you are using other baud rate than the default 9600 in BN220 you can change its baud rate (with ucenter app from ublox) or you can change line 450 in _smartport.cpp_

`GPS_SERIAL.begin(9600);`

with the desired baud rate

dgatf, 2021-02-18:
To debug the gps, download latest code as debug for gps was broken

Enable gps debug in _config.h_, line 119:

`#define DEBUG_GPS`

 and post the output (with serial monitor at 115200)

dgatf, 2021-02-18:
Please download latest code: now all gps values are initialized to 0 so you won't get garbage values if there is no data input from the gps

dgatf, 2021-02-19:
Same applies for BMP280. You were getting garbage values because BMP280 is not communicating with arduino

Now altitude will be 0 even if there is no data input from bmp280

Review connections and config

dgatf, 2021-02-20:
There was an issue with the altitude calculation. Fixed

dgatf, 2021-02-21:
Date/time and lat/log values were twisted. Fixed
Also altitude is now limited to values bigger than 0

stefaanbolle, 2021-02-25:
GPS looks OK now, still 0m with BMP280 (address 76 since pin determining address is not connected to anything)

dgatf, 2021-02-25:
Do you get a valid value for bmp280 temperature at sensor Tmp1 (0x401)?

stefaanbolle, 2021-02-25:
temp is 0 deg
alt = 0m

stefaanbolle, 2021-02-25:
Seems BMP is not working. Connection is straaight forward if SCL and SDA pins are corresponding in schema vs code

dgatf, 2021-02-25:
If you are using lua for config, i2c address is in decimal, 0x76 is 118 and 0x77 is 119

stefaanbolle, 2021-02-25:
Hmm, address might be wrong indeed

stefaanbolle, 2021-02-25:
Since the last commit I do notice when clicking update in LUA, UPDATING takes forever. Wi have to stop it with ret key on TX. When restarting msrc.lua, values are retained.

dgatf, 2021-02-28:
There was a bug in the lua script that was causing wrong I2C config. Please upload the latest lua script in the radio, update config and check if BMP is working

dgatf, 2021-02-28:
> Since the last commit I do notice when clicking update in LUA, UPDATING takes forever. Wi have to stop it with ret key on TX. When restarting msrc.lua, values are retained.

Fixed. Please flash latest code

stefaanbolle, 2021-02-28:
Updating still keeps flashing. I have to quit it with return.
Alt and Tmp1 remains 0.
Awaiting new sensor module which should arrive any day now. Will swap module to try this option.

dgatf, 2021-02-28:
Could you flash latest code and enable in _config.h_:
```
#define DEBUG_EEPROM_WRITE
#define DEBUG_EEPROM_READ
```
And post the result when booting and also when updating from lua

dgatf, 2021-02-28:
Sorry, now is updated for debug

stefaanbolle, 2021-03-02:
Can I connect both the S.Port connector and the usb connector to the Mega2560 board at the same time to output the debug? Won't overload board nor PC USB Port? Just to be sure

dgatf, 2021-03-02:
Disconnect Vcc from smartport, keep only GND and data lines

stefaanbolle, 2021-03-02:
>     1. Could you flash the latest code and enable in _config.h_:
> 
> 
> ```
> #define DEBUG_EEPROM_WRITE
> #define DEBUG_EEPROM_READ
> ```

Gives
`
V0.8.0
RC
Pr:4
As:0
Gp:1
V1:1
V2:0
Cu:1
T1:0
T2:0
Po:1
rR:10
rV:10
rC:10
rT:10
aR:3
aV:3
aC:3
aT:3
I1:1
A1:118
I2:2
A2:0
RC
Pr:4
As:0
Gp:1
V1:1
V2:0
Cu:1
T1:0
T2:0
Po:1
rR:10
rV:10
rC:10
rT:10
aR:3
aV:3
aC:3
aT:3
I1:1
A1:118
I2:2
A2:0
9⸮&⸮$⸮⸮⸮ą⸮⸮R⸮⸮⸮vq⸮⸮M'⸮)t6⸮H⸮ɉ⸮⸮⸮⸮⸮⸮
V0.8.0
RC
Pr:4
As:0
Gp:1
V1:1
V2:0
Cu:1
T1:0
T2:0
Po:1
rR:10
rV:10
rC:10
rT:10
aR:3
aV:3
aC:3
aT:3
I1:1
A1:118
I2:2
A2:0
RC
Pr:4
As:0
Gp:1
V1:1
V2:0
Cu:1
T1:0
T2:0
Po:1
rR:10
rV:10
rC:10
rT:10
aR:3
aV:3
aC:3
aT:3
I1:1
A1:118
I2:2
A2:0
`

> And post the result when booting and also when updating from lua
> 
>     1. Disable the previous and enable `#define DEBUG` and post the output when updating from lua
gives a lot, but when clicking update I get
`M+
<F:31 D:5000 V:AAAA96F1
<F:31 D:5000 V:43333F2
<F:31 D:5000
V0.8.0
`
after that it remains UPDATING and no further debug output

dgatf, 2021-03-02:
I2C config seems good (I1:1 A1:118). You could try with address 119 (0x77), just in case

The arduino is rebooting when updating from lua. I'll take a look at it (I don't have the radio around and that makes debug more difficult as I have to emulate it). That's why lua keeps trying to update

dgatf, 2021-03-07:
Fixed the rebooting issue when updating from lua. There was an issue deleting objects

Please try latest code

dgatf, 2021-03-07:
I've got BMP280 working at 0x76. I recommend to test BMP280 with an specific sketch and/or I2C scanner

stefaanbolle, 2021-03-07:
Thanks, will try with a new sensor. Should arrive in the mailbox any moment now

stefaanbolle, 2021-03-11:
The LUA on the Transmitter works well again.
The BMP280 (new one does not give any telemetry data via I2C.
It is connected to
GND - GND
SDA - D20
SCL - D21
3V3 - 3V3

In LUA on TX I sellect I2C1 as BMP280 and address 118/119 (tried both)

dgatf, 2021-03-11:
Please flash i2c_scan.ino to check connection and BMP module. No need to power the receiver or anything else

In serial monitor at 115200, you should see:

```
I2C scanner
76
Done
```
Where 76 is the address in hex. If no number, there is an issue with the circuit or module

stefaanbolle, 2021-03-12:
Will have to search, BMP is brand new
```
I2C scanner
Done
```
D20 - SDA and D21 - SCL are correct right?

dgatf, 2021-03-12:
Yes, correct. Be sure they are not interchanged (try interchanging then just un case). Dont use long wires. It looks like a circuit issue

dgatf, 2021-03-12:
I see you are using a HW v5 LH ESC. Could you post the ESC signature by enabling #debug DEBUG_ESC_HW_V4
It will appear berore raising throttle and the output is S: and some hex values. I'm trying to autodetect the ESC model
Are you getting correct values for the current?
Thanks

stefaanbolle, 2021-03-12:
Actually I use a HW V4 => my config was wrong

stefaanbolle, 2021-03-12:
Running out of ideas. Swapped SDA and SCL, same result.
The BMP gets 3,3V
Wires are 7cm

dgatf, 2021-03-12:
Do you have another Arduino to test with?

dgatf, 2021-03-12:
Con you post a picture or a link to your bmp280 module?

dgatf, 2021-03-12:
I think your module is stuck in SPI mode
To select I2C mode, try putting high CS pin for a moment with 3.3v
I've modified the I2C scanner to loop every 3sec

stefaanbolle, 2021-03-12:
There is no CS pin on the module
VCC - GND - SCL - SDA - CS0 and SD0

dgatf, 2021-03-12:
CS=CS0

stefaanbolle, 2021-03-12:
> CS=CS0
CSB I meant. Tried that one, same result

dgatf, 2021-03-12:
Maybe a picture of the circuit you made will help

stefaanbolle, 2021-03-12:
See attached. The yellow wire is going to D20 - SDA and the blue to D21 - SCL
VCC to 3.3V and GND to GND on the Mega2560 board.

[image]
[image]

dgatf, 2021-03-12:
Looks ok... don't know what else...

You can try connecting SD0 to GND or 3.3V. This defines the address (0x76, 0x77). With CSB disconnected or to 3.3v

You can try 5V version of BMP280, which is 4 pins and only I2C. Though I'm using the same board as you, 3.3v 6 pins, without any issue

stefaanbolle, 2021-03-13:
Can this come into play [http://www.varesano.net/blog/fabio/how-disable-internal-arduino-atmega-pullups-sda-and-scl-i2c-bus](url)?
The Mega2560 apparently outputs 5V on its I2C interface while the BMP280 is 3V3 only

stefaanbolle, 2021-03-13:
If the case I probably blew up 2 BMP280-3V3 already :-)
Have 2 BMP280-5V ordered now to test. Will arrive in 6 weeks or so :-p

dgatf, 2021-03-13:
It might be. I always used a 3.3v BMP module without issues, but that does not guarantee other modules wont get burnt. Let me know

dgatf, 2021-03-14:
As a last resort you can try with external pull up resistors (4~5k) on SDA and SCL lines. See [circuit](https://thesolaruniverse.files.wordpress.com/2017/08/029_fig_02_96_dpi.png?w=736)

stefaanbolle, 2021-03-14:
In the meantime tested the BMP280-3V3 with an Arduino Pro Mini 3V3 (SDA -> A4 and SCL -> A5). I2CScan gives no address. Probably both modules are dead. Will try with the 5V version as soon it arrives

stefaanbolle, 2021-04-14:
Even with a new BMP280-5V I get no output on Tmp and Alt

closed 2021-04-27 by stefaanbolle