dgatf/msrc #28: rpm limited ?

twistedwings, 2021-03-08:
I am using an Arduino Pro Mini 3.3V/8Mhz with voltage1 and rpm sensors configured.
Rpm signal goes from a HW phase sensor to pin 8.
Rpm telemertry works up to about 2000 rpm, above 2000 rpm it drops to zero.
The signal on pin 8 is still there (about 1700 Hz).

I am running the same configuration with an 5V/16Mhz Arduino Pro Mini without problems.
Is the  3.3V/8Mhz too slow (I cannot believe that) ?

dgatf, 2021-03-08:
Not too slow. It is because I used a 16Mhz board and it was needed to reduce the timer to avoid an 'overflow' for 8Mhz. Well, at least in theory... I don't have a 8Mhz now

Please try the updated code

twistedwings, 2021-03-08:
still the same problem...

dgatf, 2021-03-08:
Changed timer scaler (maybe timer scaler was too slow for 8Mhz). It will also produce better accuracy at higher rpms

Please try updated code

twistedwings, 2021-03-09:
seems to be fixed, thank you...

closed 2021-03-10 by dgatf