dgatf/msrc #29: BST (TBS Crossfire)

Giorgo961, 2021-03-11:
Hello, please do you plan to add also a BST (TBS Crossfire) telemetry protocol? It would be very helpful for me and probably also for other FPV fans not using FC.

dgatf, 2021-03-12:
No plans yet. I'll take a look though

Giorgo961, 2021-03-13:
If then please take a look also here:  https://github.com/pkurucz/opentelem_to_bst_bridge/issues/2

dgatf, 2021-03-14:
Which board do you plan to use? What sensors are you going to attach?

dgatf, 2021-03-14:
Telemetry seems to be limited to gps, battery and attitude

It seems that there are 2 protocols. One I2C and other serial. I plan to implement I2C, which is very similar to Spektrum XBUS. Could you confirm

dgatf, 2021-03-14:
It is unclear how BST protocol works. Whether sensors should be connected to I2C in slave or master mode. I expect them to be connected as slave, not as master as seems to be in the link you post

Giorgo961, 2021-03-14:
I plan to use Arduino board with APM32F103C. For me it would be the Rx supplying battery voltage enough. Yes, BST I2C is ok. Probably the board should be a slave and Crossfire master.

dgatf, 2021-03-14:
What is the APM32F103C for?

Giorgo961, 2021-03-15:
This one: https://www.aliexpress.com/item/4000807291937.html?spm=a2g0s.9042311.0.0.5a134c4dOWpE5i

dgatf, 2021-03-15:
Here is the [branch](https://github.com/dgatf/msrc/tree/bst) with BST support

This first attempt is with I2C as slave. Flash as is. You need to connect BST SDA, SCL, and GND  to arduino A4, A5 and GND respectively (don't connect Vcc between BST and Arduino as you will need to debug, so it will be feed from usb or usb-ttl, depending on your arduino board. You don't need to connect any sensor yet. Debug is enabled. It will try to send telemetry for battery and gps. I'd be surprised if this works at first. Post the output of the arduino with a serial monitor at 115200

Giorgo961, 2021-03-15:
Thanks a lot, I will try it, but unfortunately the Ardu board still did not arrive. Then I will let you know.

Giorgo961, 2021-03-16:
Do you think it will work also on this board? https://www.banggood.com/Wholesale-ATMEGA328-328p-5V-16MHz-Pro-Mini-PCB-Module-Board-p-68534.html?rmmds=myorder&cur_warehouse=CN Two of them I have at home.

dgatf, 2021-03-16:
Yes, that's a good choice (unless you plan to use a clastle esc or esc serial + gps serial). It's cheap and light

Giorgo961, 2021-03-17:
I checked it, but it seems it does not work. No new sensors in telemetry, only that from the Rx. I do not know, how to display output  from Arduino with serial monitor, I am not familiar with Arduino IDE. I can only flash the board by it. Please, advice.
[image]
[image]

dgatf, 2021-03-18:
I'd suggest to solder angled pins for easy fashing and debug with serial monitor. See [image](../tree/master/images/bottom.jpg). But be can try this without debugging

To confirm you are flashing the correct branch, please could you confirm that line 240 in _msrc.cpp_ is `DEBUG_SERIAL.println("BST");`

Giorgo961, 2021-03-18:
I do not understand to what I should connect the board - to the board FTDI232 with USB converter, which I used for flashing? Do not know for what are the wires on the picture...
Yes, I can confirm the right line 240 in msrc.cpp

dgatf, 2021-03-19:
I've made some fixes to the [bst](https://github.com/dgatf/msrc/tree/bst) branch

Please try with the updated code

Let me know if it works. If it doesn't I'll change the code to broadcast as a master on the i2c

Yes, for debug you have to connect only two wires: arduino-tx to ftdi-rx and arduino-gnd to ftdi-gnd (as well as bst tx on the other side)

Giorgo961, 2021-03-19:
I tried, but still only that old 10 sensors. The difference is, that the red LED on the Ardu board is not blinking now.

dgatf, 2021-03-20:
Ok. Then try with this branch [bst_master](https://github.com/dgatf/msrc/tree/bst_master)

The telemetry is send by I2C master broadcast

Giorgo961, 2021-03-20:
I am sorry, but still the same :-(

Giorgo961, 2021-03-24:
Daniel, any idea, how to proceed? In the Eagletree Vector project they are using pull-up resistors due to weak rise time at 328P. But do not understand, why are resistors connected to A2, A3 and not to Vcc. I tried A2, A3, but no change in behavior. Tomorrow I will check the signal by oscilloscope.

dgatf, 2021-03-24:
> why are resistors connected to A2, A3 and not to Vcc

It would be the same. I believe it is used A2, A3 for easier soldering. for To use A2, A3, it is needed to enable pins as output and set them high (lines 167-170, bst_telemetry.cpp)

But, yes, you can try with external pull ups, either Vcc or analog pins

Another check would be to debug as explained

If I have some time, I would check if the data send by msrc equals to other project

dgatf, 2021-03-24:
There is an issue with the crc. Checking it...

Giorgo961, 2021-03-26:
With the last (master) version it arrived signal on the outputs. With the previous versions not. But no new sensors in the telemetry, still only that old 10 from the receiver. Signals SCL, SDA on pictures:
[image]
[image]

dgatf, 2021-03-26:
Crc is correct (verified with betaflight) and it is emitting the same as bst bridge

As last attempt, I've made minor changes to [bst_master](https://github.com/dgatf/msrc/tree/bst_master). Try it

Giorgo961, 2021-03-26:
Still no new sensors:

[image]

When flashing I am opening file msrc.ino and program uses only 22% of memory. Is it ok?

[image]

Please, what should I launch and what should I see in the Arduino IDE, when I will connect it for debugging?

dgatf, 2021-03-26:
Yes, program size is correct

You have to open _Tools->Serial Monitor_ and set baud rate to 115200. Do not connect Vcc between programmer and arduino as is already powered from the receiver

You should see
```
V0.8.0
BST MASTER
8 0 7B 0 7B 0 0 0 0 86 
8 0 7B 0 7B 0 0 0 0 86
```

Giorgo961, 2021-03-26:
I can confirm, it is like you described:

[image]

Giorgo961, 2021-03-28:
I tried to use a Frsky receiver R-XSR and with S.PORT it is working well. But by Crossfire the lua script is still trying to connect:

[image]

Please, is the 0.8 version ok?

dgatf, 2021-03-28:
Better flash the master branch, not bst
Be sure CONFIG_LUA is not commented un config.h

Giorgo961, 2021-03-28:
ah, ok