dgatf/msrc #31: Castle Link: BecV 0

twistedwings, 2021-03-13:
BEC voltage is sent as zero.
[image]

I don't know if the esc sends a value since I don't have an official telemetry documentation.
According to Spirit  it should.
[http://manual.spirit-system.com/index.php?title=Castle_Telemetry](url)

Castle Creations Talon 90
Pololu ATmega328PB

dgatf, 2021-03-13:
My Castle is an opto so no bec to test. But yours should be sending in theory. I've verified the code and seems ok

You can check raw telemetry of the castle by enabling in config.h `#define DEBUG_ESC_CASTLE` and with a serial monitor at 115200. Post the output, I'll check if it is sending bec telemetry. When debugging, please do not connect Vcc between the Pololu and the USB-TTL. This is important

twistedwings, 2021-03-13:
seems like the esc is sending zero.

20:58:04.143 -> 2 3814 2027 3247 1014 1013 2939 1013 1013 1013 1013 2656 1013 
20:58:04.378 -> 2 4054 2025 3260 1013 1013 2939 1013 1013 1013 1013 2657 1013 
20:58:04.610 -> 2 4294 2026 3260 1013 1013 2939 1013 1013 1013 1014 2658 1013 
20:58:04.842 -> 2 4534 2025 3259 1013 1013 2938 1013 1013 1013 1014 2657 1014 
20:58:05.072 -> 2 4774 2026 3247 1013 1013 2938 1013 1013 1013 1013 2657 1013 
20:58:05.353 -> 2 5014 2025 3247 1014 1013 2939 1013 1013 1013 1013 2656 1013 
20:58:05.586 -> 2 5254 2025 3247 1013 1013 2940 1013 1013 1013 1013 2657 1014 
20:58:05.817 -> 2 5494 2025 3261 1013 1013 2939 1013 1013 1013 1013 2862 1013 
20:58:06.051 -> 2 5734 2025 3261 1014 1013 2940 1014 1013 1014 1013 2657 1013 
20:58:06.285 -> 2 5975 2025 3260 1013 1013 2939 1013 1013 1013 1013 2862 1013 
20:58:06.522 -> 2 6214 2026 3260 1013 1013 2940 1013 1013 1014 1013 2657 1014

dgatf, 2021-03-13:
Correct, your model is not sending bec telemetry

closed 2021-03-13 by dgatf