dgatf/msrc #36: current scale

twistedwings, 2021-03-22:
I want to connect a current sensor to A6.
The current scale of the sensor is 19.1A/V.
I have entered a scale of 19.1 in OpenTX.
Then I tested msrc by applying voltage to pin A6.

These were the measurements:
1V 0.7A (expected 19.1A)
2V 1.4A (expected  38.2A)

I had to enter a scale of 484 to get the correct current displayed.

Arduino 3,3V/8MHz
Sensors HW3, current

dgatf, 2021-03-23:
Code is ok. There was an error in theformula for analog current in the readme. The way is managed the ratio in opentx sensors is a bit counterintuitive for other sensors than voltage dividers (ratio = max sensor voltage x voltage divider ratio). I still need to understand why multiplier 25.5 is needed to get multiplier = 1

closed 2021-03-23 by dgatf

twistedwings, 2021-03-23:
thanks.
An explanation for the 25.5 magic is here
[https://openrcforums.com/forum/viewtopic.php?f=86&t=12483&p=148399#p148399](url)

dgatf, 2021-03-23:
Interesting. Thanks