dgatf/msrc #38: No data Kontronik Jive

Hobby4life, 2021-04-13:
Issue:  No telemetry data Kontronik Jive.

TX: Frskty X10S Express
RX: X4R (Smartport)

I have Smartport running, configured it with the lua script to use the Kontronik Protocol.
That works.. got all sensors discovered.

Connected the Jive to RX0 of my Nano.

I see there is data coming in.. but not passed thru to smartport..

I know the telemetry port works, because I use a Jlog S32 logger to read out it's telemetry.

Am I missing something?

Hobby4life, 2021-04-14:
I have used the sniffer tool to monitor the jive output.
All baudrates except 115200bd show data.

I assume the jive's dont sent out telemetry @ 115200bd, but more like 9600bd

dgatf, 2021-08-31:
Because your esc uses an older Kontronik protocol. There is no official protocol but it has been revered engineered [here](https://www.helifreak.com/showthread.php?t=721180). I may add support for this protocol in the future

The implemented Kontronic protocol in msrc is at 115200. You can find the specs on this [link](https://www.kontronik.com/fileadmin/kontronik-sobek/Public/Content/Images/Content/Downloads/Software/Kontronik_TelMe_V4.12_1.12_EN.pdf)

closed 2021-08-31 by dgatf