dgatf/msrc #55: HobbyWing V4 120A LV gives not date when motor off

LuHaGH, 2021-11-01:
> You are right, only when the motor is is running the ESC is transferring data.

[https://github.com/dgatf/msrc/files/7449668/ESC-XBus.txt](url)

That is not right, but the format is different when the motor is stopped.
I took your sniffer and as you can see in the logfile there are two different data lines if the motor is not runnig and only one if the motor is running.

[HW_120A_V4.txt](https://github.com/dgatf/msrc/files/7452524/HW_120A_V4.txt)

Maybe you could optimize the data processing?

dgatf, 2021-12-12:
Fixed with 2d8250a26736583c5ac949de7acf573e012c01c4

closed 2021-12-12 by dgatf

dgatf, 2022-02-13:
Indeed it was not solved. When motor is stopped, packet length is 1 byte longer

Fixed now with 5c6c9c98b25005eb8c09cc64a048f630b779bb17