dgatf/msrc #56: bad quantization of analog "Curr" sensor ?

frittna, 2021-12-06:
Hi, when i try to use "Curr" with a analog signal on ADC pin20 (on my TeensyLC) i get really bad quantization setps, like only a few steps for 0-100%.. 
BUT when i simply swap the pins in the MSRC code (pin20 and 26 in my case)  to use the Airspeed pin instead of the Current pin it is quantisizing fine like it should. 

I mean without any hardware change. 

any idea if that could be a bug beside the hardware which is (i guess) working good with TeensyLC 3.3V, HW ESC v4 RPM. BMP280, 2xTemp analog - at last!
For the current - since the Hobbywing 40A ESC has no built in shunt - I now use 1mOhm shunt resisor with a LM358N as diff-amp placed in the GND Cable, 0-40A equals 1.6V and it is working good, no bad noise with 1u and 220n caps, nice level shifing on a scope. Shunt is ok i guess. 
btw: my ACS711EX hall current sensor was not usable because of far to heavy noise on the signal line on my try

AND: Great project!! everything is working superb after dealing with many many issues but that was perhaps all my fault and mainly Pololu 5V related

regards

dgatf, 2021-12-11:
> Hi, when i try to use "Curr" with an analog signal on ADC pin20 (on my TeensyLC) i get really bad quantization setps, like only a few steps for 0-100%.. BUT when i simply swap the pins in the MSRC code at constants.h (pin20 and 26 in my case) to use the current sensor on the Airspeed-input instead of the Current-input it is quantisizing fine like it should.

Thanks. There was a mistake on the pin numbers for current and pressure for the Teensy. It is fixed now. Pins 18 and 19 are used for the I2C port and can't be used for analog readings. Pin 20 should be fine for Current now.

> btw: my ACS711EX hall current sensor was not usable because of far to heavy noise on the signal line on my try

You can reduce the noise in two ways:
- Software. Increase averaging elements in config.h (CONFIG_AVERAGING_ELEMENTS_CURR) or lua script for current. Assuming there is one measurement for current every 2ms, to get 1s average, increase averaging elements to 500
- Hardware. Use capacitor at Vout (Cload, 1nF). [Circuit](http://www.12580ic.com/upload/ACS711-Typical-Application.gif)

frittna, 2021-12-12:
Thanks !
 I made a picture for you. It is a LOG where i try to explain what i mean with the wrong calculated values of current, no matter what is connected, regards.
[image]

dgatf, 2021-12-13:
It looks like it is enabled the pressure pin instead of the current pin

Please could you attach or post your config.h and the debug when enabling DEBUG_EEPROM_READ (if using lua script)

frittna, 2021-12-14:
i will enable and post a debug log soon.

but i explain it again in a very short version:

1:  i have an analog sensor connected to Pin20, which is "Curr" at default , my PIN26 is free 
     Result --> received "Curr" values are choppy.  

2:  Then, i do not change anything on the connections/sensors/etc
     I ONLY change PIN numbers for Curr and Pressure in Arduino (constants.h) where Port connections are set.
     Result --> Pressure is showing the correct value from pin20 and it is smooth and nice.

no unit, no calculation, and same precision for both SPort sensors are set in OpenTX 2.3.14 (Q7)
there must be a difference in handling analog values in current in comparison to pressure.
if i would connect 2 identical sensors to pressure and curr at the same time it would be interesting, have not done that because i can't try at the moment..

regards

**here is a part of my config.h**
/* Sensors */
...
#define CONFIG_ESC_PROTOCOL PROTOCOL_HW_V4_LV // PROTOCOL_NONE, PROTOCOL_HW_V3, PROTOCOL_HW_V4_LV, PROTOCOL_HW_V4_HV, PROTOCOL_HW_V5_LV, PROTOCOL_HW_V5_HV, PROTOCOL_PWM, PROTOCOL_CASTLE, PROTOCOL_KONTRONIK
#define CONFIG_GPS false
#define GPS_BAUD_RATE 9600
#define CONFIG_VOLTAGE1 false
#define CONFIG_VOLTAGE2 false
#define CONFIG_NTC1 true
#define CONFIG_NTC2 true
#define CONFIG_CURRENT true
#define CONFIG_AIRSPEED true
#define CONFIG_I2C1_TYPE I2C_BMP280 // I2C_NONE, I2C_BMP280
#define CONFIG_I2C1_ADDRESS 118 //in dezimal format
/* Refresh rate in 0.1s (1 = 100ms) */
#define CONFIG_REFRESH_RPM 1
#define CONFIG_REFRESH_VOLT 1
#define CONFIG_REFRESH_CURR 1
#define CONFIG_REFRESH_TEMP 1
#define CONFIG_REFRESH_DEF 1
/* Averaging elements (1 = no averaging) */
#define CONFIG_AVERAGING_ELEMENTS_RPM 3
#define CONFIG_AVERAGING_ELEMENTS_VOLT 3
#define CONFIG_AVERAGING_ELEMENTS_CURR 3
#define CONFIG_AVERAGING_ELEMENTS_TEMP 3
#define CONFIG_AVERAGING_ELEMENTS_DEF 3
...

**here is the part in my constants.h**
...
// Teensy LC/3.x
#if defined(__MKL26Z64__) || defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MK64FX512__) || defined(__MK66FX1M0__) || defined(__IMXRT1062__)
#define PIN_NTC1 14
#define PIN_NTC2 15
#define PIN_VOLTAGE1 16
#define PIN_VOLTAGE2 17
#define PIN_CURRENT 26 //20 davor, aber da cuurent nicht gut funktioniert wird gegen pressure (VSpd) vertauscht
#define PIN_PRESSURE 20 //26 davor
#define SMARTPORT_FRSKY_SBUS_SERIAL hardSerial0
#define SRXL_IBUS_SERIAL hardSerial0
#define ESC_SERIAL hardSerial1
#define GPS_SERIAL hardSerial2
#define DEBUG_SERIAL Serial
#endif
...

dgatf, 2021-12-14:
Indeed there was an issue with the analog current resolution. Also with analog voltage. Fixed with ded3b53af5ad7094489262927ec9beee0258729d

I've added multipliers for analog current and analog voltage sensors. This allows to transform to the correct values from code, rather than from opentx, which was causing a poor value resolution.

This also allows other rx protocols to adjust the current and voltage values to the real ones.

So rather using opentx to adjust the current value, use CURRENT_MULTIPLIER in config.h to transform Volts to Amps

One comment is that you are getting a low Vout in your circuit for current measurement. Based on log data I suspect you are getiing a max Vout of 0.4V or so, while maximum is 3.3v. You may need to change the diffamp for a bigger gain. If I'm correct you resolution will be about 400 points using CURRENT_MULTIPLIER (with opentx you were getting a resolution about 4 levels). Not too bad at all

closed 2021-12-14 by dgatf