dgatf/msrc #57: msrc

Klippeneck, 2022-01-15:
Hello Daniel,
came across a link to your sensor on RC Network.
Since I fly Jeti, I wanted to build a sensor with 2 voltage measurements as a start.
So Arduino Pro Mini 3.3V, 8MHz flashed with the configured SW and plugged into the RX.
The sensor does not appear in the telemetry selection of the transmitter.
To exclude a defect on the Arduino board, I have then aud the board a sketch of RCT 
which works. Do you have a tip that the sensor works?
Actually I wanted to use the sensor with an ESC. But I am not yet so far.
I am still relatively inexperienced in Arduino history.

Jürgen

dgatf, 2022-01-15:
Please post your config.h

Which ESC are you connecting to?

dgatf, 2022-01-15:
To enable debug with Jeti protocol, change in _constants.h_, line 51:

`#define DEBUG_SERIAL softSerial`

And in _config.h_ line 68, enable debug:

`#define DEBUG`

Then connect arduino pin 12 to programmer pin Rx and arduino gnd to programmer gnd

Use serial monitor at 115200 and post the output

Klippeneck, 2022-01-15:
Hallo,
meine config.h :
#ifndef CONFIG_H
#define CONFIG_H
#include "constants.h"

/* Empfangsprotokoll */
#define RX_PROTOCOL RX_JETIEX

/* Sensoren */
#define CONFIG_ESC_PROTOCOL PROTOCOL_NONE
#define CONFIG_GPS false
#define GPS_BAUD_RATE 9600 
#define CONFIG_VOLTAGE1 true 
#define CONFIG_VOLTAGE2 true 
#define CONFIG_NTC1 false 
#define CONFIG_NTC2 false 
#define CONFIG_CURRENT false 
#define CONFIG_AIRSPEED false 
#define CONFIG_I2C1_TYPE I2C_NONE 
#define CONFIG_I2C1_ADDRESS 119
/* Aktualisierungsrate in 0,1s (1 = 100ms) */
#define CONFIG_REFRESH_RPM 1
#define CONFIG_REFRESH_VOLT 1
#define CONFIG_REFRESH_CURR 1
#define CONFIG_REFRESH_TEMP 1
#define CONFIG_REFRESH_DEF 1
/* Mittelwertbildungselemente (1 = keine Mittelwertbildung) */
#define CONFIG_AVERAGING_ELEMENTS_RPM 3
#define CONFIG_AVERAGING_ELEMENTS_VOLT 3
#define CONFIG_AVERAGING_ELEMENTS_CURR 3
#define CONFIG_AVERAGING_ELEMENTS_TEMP 3
#define CONFIG_AVERAGING_ELEMENTS_DEF 3
/* Analoge Multiplikatoren */
#define VOLTAGE1_MULTIPLIER 1
#define VOLTAGE2_MULTIPLIER 1
#define CURRENT_MULTIPLIER 1

/* Pwm out */
#define CONFIG_PWMOUT false
#define PWMOUT_DUTY 0.5 // 0.5 = 50%

/* Nur Smartport und Opentx */
#define SENSOR_ID 10 // Sensor Id
#define DATA_ID 0x5000 // DataId (Sensortyp)
//#define ESC_SIGNATURE // HW V4 Signatur (nur Smartport). Dies gibt die ESC-Signatur und den Rohstrom an die Sensoren 5100, 5101 und 5102 aus
#define CONFIG_LUA // Kommentar, wenn kein Lua-Skript zur Konfiguration verwendet wird (nur Smartport)

/* XBus */
//#define XBUS_CLOCK_STRECH_SWITCH

/* Bibliothek I2C_T3 für Teensy LC/3.X verwenden */
#define I2C_T3_TEENSY

/* Init-Verzögerung für FlyFun ESC hinzufügen. Unkommentiert lassen, wenn der ESC nicht scharf geschaltet wird */
//#define ESC_INIT_DELAY 10000

/* HW V4/V5 Stromschwellenwert Drossel (0-100%) */
#define CURRENT_THRESHOLD 25

/* Eeprom-Schreiben erzwingen */
//#define FORCE_EEPROM_WRITE // Uncomment to force write eeprom as defined in config.h. Nützlich, wenn man lua benutzt und eeprom ist durcheinander. Bei jedem erneuten Flashen muss die Zeile auskommentiert werden, sonst wird die Konfiguration beim Einschalten zurückgesetzt.

/* Fehlersuche
   Trennen Sie die Vcc-Verbindung zwischen dem RC-Modell und dem Arduino
   Verbinden Sie nicht gleichzeitig Vcc vom Modell und usb (TTL)
   Telemetrie könnte im Debug-Modus nicht richtig funktionieren
   Verbinden Sie den Arduino Rx mit TTL Tx zum Flashen, dann, falls zutreffend, verbinden Sie den Arduino Rx mit esc oder gps
*/

//#define DEBUG
//#define DEBUG_PACKET
//#define DEBUG_EEPROM_WRITE
//#define DEBUG_EEPROM_READ

//#define DEBUG_HW3
//#define DEBUG_HW4
//#define DEBUG_ESC_KONTRONIK
//#define DEBUG_GPS
//#define DEBUG_PWM
//#define DEBUG_CASTLE
//#define DEBUG_CASTLE_RX

//#define SIM_RX
//#define SIM_SENSORS
//#define SIM_LUA_SEND
//#define SIM_LUA_RECEIVE

#endif

Devices used:
DS12 and DS 16 transmitter
Receiver REX 10 Assist

*** Übersetzt mit www.DeepL.com/Translator (kostenlose Version) ***

dgatf, 2022-01-15:
_config.h_ is ok and your transmitter and receiver are compatible with Jeti Ex Telemetry

Check that receiver output pin, _Ext,_ is configured as _Ex Bus_

You'll need to enable the debug mode as explained and post the output

Klippeneck, 2022-01-15:
I do not use ext. output there hangs a vario, have the E1 output set to EX bus.
With the debug I have to read in first, will next week.

dgatf, 2022-01-15:
> I do not use ext. output there hangs a vario, have the E1 output set to EX bus. 

That's ok then

Klippeneck, 2022-01-16:
Debug went easier than I thought.
Here is the output: 11:19:01.638 ->     
                             11:19:01.638 -> V900

dgatf, 2022-01-16:
This means that Arduino is not reading the Ex Bus data frames from the receiver, either because the receiver is not sending Ex frames or because the connection between the receiver and the Arduino is faulty.

Check the receiver config and be sure it is sending ex frames. And check connection between receiver and arduino

You should see a similar debug output:

```
V900
<3D 1 8 E0 3A 0 E0 5E 
<3D 1 8 E1 3A 0 3C 4 
T>3B 1 1C E1 3A 14 F 12 0 A4 0 A4 0 1 49 56 6F 6C 74 61 67 65 20 31 56 A3 4 1E 
<3D 1 8 E2 3A 0 58 EB 
<3D 1 8 E3 3A 0 84 B1 
V>3B 1 16 E3 3A E F 4C 0 A4 0 A4 0 11 D1 44 21 D1 44 2 AE 4E 
```

Klippeneck, 2022-01-17:
Next try:
Sensor at E1 output of receiver, serial log21:16:49.928 -> V900
21:16:50.677 -> <3E 3 28 0 31 20 80 3E 
21:16:50.677 -> <3E 33 33 0 33 33 0 70 
21:16:50.724 -> <3 28 0 31 20 80 3E D9 
21:16:50.724 -> <3E 3 28 0 31 20 80 3E 
21:16:50.724 -> <3E 3 28 0 31 20 80 3E 
21:16:50.724 -> <3E 3 28 0 31 20 80 3E 
21:16:50.724 -> <3E 3 28 0 31 20 80 3E 
21:16:50.771 -> <3E 3 28 0 31 20 80 3E 
21:16:50.771 -> <3E 3 28 0 31 20 80 3E 
21:16:50.771 -> <3E 3 28 0 31 20 80 3E 
21:16:50.771 -> <3E 3 28 0 31 20 80 3E 
21:16:50.771 -> <3E 3 28 0 31 20 80 3E 
21:16:50.818 -> <3E 3 28 0 31 20 80 3E 
21:16:50.818 -> <3E 3 28 0 31 20 80 3E 
21:16:50.818 -> <3E 3 28 0 31 20 80 3E 
21:16:50.818 -> <3E 3 28 0 31 20 80 3E 
V900V

dgatf, 2022-01-17:
There is a mistake in debug mode. Could you flash latest code or update line 278 in _jetiex.cpp_:

`for (uint8_t i = 0; i < length; i++)`

Klippeneck, 2022-01-17:
With latest Software;
14:58:54.201 -> V900
14:58:54.717 -> <EA 2E EB 2E EC 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E B7 E4 3D 1 8 0 3A 0 41 57 
14:58:55.279 -> <3E 3 28 0 31 20 80 3E DD 2E EB 2E EC 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E B7 E4 3D 1 8 0 3A 0 41 57 
14:58:55.279 -> <3E 1 28 2 31 20 80 3E DD 2E EB 2E EC 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E 6A 82 
14:58:55.279 -> <3E 3 28 2 31 20 80 3E DD 2E EB 2E EC 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E C5 FE 3D 1 8 2 3A 0 F9 E2 
14:58:55.279 -> <3E 3 28 2 31 20 80 3E DD 2E EB 2E EC 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E C5 FE 3D 1 8 2 3A 0 F9 E2 
14:58:55.279 -> <3E 3 28 2 31 20 80 3E DD 2E EB 2E EC 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E C5 FE 3D 1 8 2 3A 0 F9 E2 
14:58:55.279 -> <3E 3 28 2 31 20 80 3E DD 2E EB 2E EC 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E C5 FE 3D 1 8 2 3A 0 F9 E2 
14:58:55.279 -> <3E 3 28 3 31 20 80 3E DD 2E EB 2E EC 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E FC F3 3D 1 8 3 40 0 91 B5 
14:58:55.279 -> <3E 3 28 3 31 20 80 3E DD 2E EB 2E EC 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E FC F3 3D 1 8 3 3A 0 25 B8 
14:58:55.279 -> <3E 3 28 3 31 20 80 3E DD 2E EB 2E EC 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E E0 2E FC F3 3D 1 8 3 3A 0 25 B8

dgatf, 2022-01-17:
Please try updated code. Should be fixed now

Klippeneck, 2022-01-17:
Test with Jeti EX Bus successful.
2 voltages are transmitted as configured.
Next Teststep with ESC.

dgatf, 2022-01-17:
Great

closed 2022-01-17 by dgatf

dgatf, 2022-01-17:
As you are using the Pro Mini with Jeti Ex, you'll need to change line 71 in _config.h_:

`#define ESC_SERIAL softSerial`

And connect ESC telemetry to pin 7

dgatf, 2022-01-17:
Well, just made that automatic for ATmega328P and ATmega32U4. So either, you can download the updated code

Klippeneck, 2022-01-23:
Hello,
performed first tests with a Fun Fly V5 ESC.
Values supplied by the ESC for current and voltage have significant deposits, sometimes more than 30%.
Is there a way to adjust the values in the sensor?
Voltage measurements via Analog In go well.
However, the sensor runs with me (with Jeti ) only with 5V Arduinos, at 3.3V the sensor reports briefly and stops after 2-3 seconds. What else can I give away?

dgatf, 2022-01-23:
>  Values supplied by the ESC for current and voltage have significant deposits, sometimes more than 30%.

Usually the voltage values from HW ESCs are more or less accurate and stable

Does not happen the same with the current, where there are big fluctuations when throttle value is low (thr<25%, maybe higher) and offset and gain may vary from one model to another

Can you describe better the discrepancy or send a telemetry log (including throttle). Also you could enable debug for HW ESC in config.h, `#define DEBUG_HW4`

>  However, the sensor runs with me (with Jeti ) only with 5V Arduinos, at 3.3V the sensor reports briefly and stops after 2-3 seconds.

Maybe is too much (Jeti + ESC) for the slower 8Mhz board:

- Check debug mode is disabled
- Does this happen when enabling the ESC or even with only one or two analog sensors?
- Try increasing the serial buffer size in serial.h: `#define FIFO_SIZE 128`

Klippeneck, 2022-01-24:
Next tests:
3.3V Ardino Pro Mini runs with only 2 voltage measurements once ESC or GPS is configured,
the sensor crashes despite the changes.
5V Arduino Pro Mini works with 2 voltage measurements and ESC or GPS. Measured values of the ESC 
are very inaccurate and would probably have to be calibrated.
Further tests with Pololu 328 PB are planned. Hope that it works with it.

dgatf, 2022-01-24:
> 3.3V Ardino Pro Mini runs with only 2 voltage measurements once ESC or GPS is configured,
the sensor crashes despite the changes.

Seems to be too much for this board. You can try reducing the ESC timeout to 1ms, in escHW4.cpp, line 10:     `serial_.setTimeout(2);`
 
> 5V Arduino Pro Mini works with 2 voltage measurements and ESC or GPS. Measured values of the ESC
are very inaccurate and would probably have to be calibrated.

Are RPM and temp also innacurate?

You can adjust the ratio for voltage and current in escHW4.h, line 21-22, third or fourth value, depending if your Fly Fun is LV or HV:
```
const uint8_t voltageDivisor_[4] = {11, 21, 11, 21};
const float ampGain_[4] = {10, 16.9, 10, 16.9};
```
Some telemetry logs will be useful

>  Further tests with Pololu 328 PB are planned. Hope that it works with it.

That's fine, but I don't expect any improvement. Innacurate values from ESC is an issue of the ESC sending those values