dgatf/msrc #58: msrc with Jeti

Klippeneck, 2022-01-25:
Next test:
Voltage and current values can be trimmed to a usable accuracy by adjusting the divisor and ampGain in escHW4.h  useful accuracy by adjusting the divisor or ampGain. Remaining measurement deviations probably come from the ESC.
I can't measure RPM at the moment, but it could be correct. After further fine tuning of the coefficients 
I will test the combination in flight.
Attached is a short LOG file.
# Sensortest
# Dienstag, 25. Januar 2022 09:16:37
000000000;4292021870;0;Tx; 
000000000;4390921784;0;REX10A; 
000000000;4390921784;1;U Rx;V
000000000;4390921784;2;A1;
000000000;4390921784;3;A2;
000000000;4390921784;4;Q;%
000000000;4390921784;10;Vario;m/s
000000000;4390921784;11;Abs. Hoehe;m
000000000;4390921784;12;Hoehe;m
000000000;4390921784;13;Luftdruck;hPa
000000000;4390921784;14;Temperatur;°C
000000000;4390921784;15;Roll;°
000000000;4390921784;16;Nick;°
000000000;4390921784;17;Heck;°
000000000;4390921784;18;G-Kraft;G
000000000;4390921784;19;Status;
000000000;4198441984;0;Sensor; 
000000000;4198441984;1;RPM;
000000000;4198441984;2;Strom;A
000000000;4198441984;3;Spannung;V
000000000;4198441984;6;Cell Voltage;V
000000000;4198441984;7;Consumption;mAh
000000000;4198441984;8;Voltage 1;V
000000000;4198441984;9;Voltage 2;V
000000001;4390921784;10;1;2;-5;11;4;1;6568;12;4;1;-1;13;4;1;9366;14;1;1;203;15;1;1;10;16;1;1;12;17;1;1;0;18;1;2;99;19;0;0;2
000000001;4198441984;1;4;0;3059;2;1;1;0;3;1;2;535;6;1;2;2;7;1;0;0;8;1;2;94;9;1;2;92
000000424;4390921784;1;1;2;526;2;1;0;9;3;1;0;9;4;1;0;100
000000484;4198441984;1;4;0;3059;2;1;1;0;3;1;2;535;6;1;2;2
000000524;4390921784;10;1;2;-4;11;4;1;6567;12;4;1;-1;15;1;1;9;16;1;1;12;17;1;1;0
000000544;4198441984;6;1;2;2;7;1;0;0;8;1;2;94;9;1;2;92
000000584;4390921784;10;1;2;-4;13;4;1;9366;15;1;1;9;16;1;1;12;17;1;1;0
000000724;4390921784;10;1;2;-4;14;1;1;203;15;1;1;9;16;1;1;12;17;1;1;0
000000787;4390921784;10;1;2;-4;15;1;1;9;16;1;1;12;17;1;1;0;18;1;2;99
000000885;4390921784;10;1;2;-5;15;1;1;9;16;1;1;12;17;1;1;0;19;0;0;2
000000905;4198441984;1;4;0;3059;2;1;1;0;3;1;2;535;6;1;2;2
000000945;4390921784;1;1;2;525;2;1;0;9;3;1;0;9;4;1;0;100
000001285;4390921784;10;1;2;-7;14;1;1;204;15;1;1;9;16;1;1;12;17;1;1;0
000001447;4390921784;10;1;2;4;15;1;1;9;16;1;1;12;17;1;1;0;18;1;2;99
000001466;4390921784;1;1;2;526;2;1;0;9;3;1;0;9;4;1;0;100
000001466;4198441984;6;1;2;2;7;1;0;0;8;1;2;94;9;1;2;92
000001486;4198441984;1;4;0;3059;2;1;1;0;3;1;2;535;6;1;2;2
000001506;4390921784;10;1;2;13;15;1;1;9;16;1;1;12;17;1;1;0;19;0;0;2
000001586;4390921784;10;1;2;23;11;4;1;6578;15;1;1;9;16;1;1;12;17;1;1;0
000001606;4198441984;1;4;0;5724;2;1;1;0;3;1;2;801;6;1;2;400
000001686;4390921784;10;1;2;32;12;4;1;7;15;1;1;9;16;1;1;12;17;1;1;0
000001786;4390921784;10;1;2;38;13;4;1;9364;15;1;1;9;16;1;1;12;17;1;1;0
000001888;4390921784;10;1;2;45;14;1;1;204;15;1;1;9;16;1;1;12;17;1;1;0
000001991;4390921784;1;1;2;526;2;1;0;9;3;1;0;9;4;1;0;100;10;1;2;54;15;1;1;9;16;1;1;12;17;1;1;0;18;1;2;99
000002090;4390921784;10;1;2;59;15;1;1;9;16;1;1;12;17;1;1;0;19;0;0;2
000002111;4198441984;6;1;2;400;7;1;0;0;8;1;2;94;9;1;2;92
000002190;4390921784;10;1;2;65;11;4;1;6588;15;1;1;9;16;1;1;13;17;1;1;0
000002290;4390921784;10;1;2;72;12;4;1;16;15;1;1;9;16;1;1;12;17;1;1;0
000002371;4198441984;1;4;0;5724;2;1;1;0;3;1;2;801;7;1;0;0;8;1;2;94;9;1;2;92
000002391;4390921784;10;1;2;77;13;4;1;9363;15;1;1;9;16;1;1;13;17;1;1;0
000002511;4390921784;1;1;2;526;2;1;0;9;3;1;0;9;4;1;0;100
000002513;4390921784;10;1;2;84;14;1;1;204;15;1;1;9;16;1;1;13;17;1;1;0
000002611;4390921784;10;1;2;92;15;1;1;8;16;1;1;13;17;1;1;0;18;1;2;100
000002711;4390921784;10;1;2;101;15;1;1;8;16;1;1;13;17;1;1;-1;19;0;0;2
000002811;4390921784;10;1;2;106;11;4;1;6596;15;1;1;8;16;1;1;13;17;1;1;-1
000002851;4198441984;6;1;2;400;7;1;0;0;8;1;2;94;9;1;2;92
000002912;4390921784;10;1;2;117;12;4;1;29;15;1;1;8;16;1;1;13;17;1;1;-1
000002992;4390921784;10;1;2;125;13;4;1;9362;15;1;1;7;16;1;1;13;17;1;1;-1
000003031;4390921784;1;1;2;526;2;1;0;9;3;1;0;9;4;1;0;100
[... 96 lines cut ...]
000009895;4198441984;6;1;2;432;7;1;0;113;8;1;2;94;9;1;2;92
000009976;4390921784;10;1;2;-39;13;4;1;9363;15;1;1;8;16;1;1;13;17;1;1;-1
000010075;4390921784;10;1;2;-53;14;1;1;210;15;1;1;8;16;1;1;14;17;1;1;-1
000010175;4390921784;10;1;2;-65;15;1;1;8;16;1;1;13;17;1;1;-1;18;1;2;99
000010315;4390921784;1;1;2;526;2;1;0;9;3;1;0;9;4;1;0;100
000010476;4390921784;10;1;2;-98;12;4;1;12;15;1;1;8;16;1;1;13;17;1;1;-1
000010675;4390921784;10;1;2;-117;14;1;1;210;15;1;1;8;16;1;1;13;17;1;1;-1
000010795;4390921784;10;1;2;-122;15;1;1;8;16;1;1;13;17;1;1;-1;18;1;2;99
000010795;4198441984;6;1;2;444;7;1;0;113;8;1;2;94;9;1;2;92
000010835;4390921784;1;1;2;526;2;1;0;9;3;1;0;9;4;1;0;100
000010875;4390921784;10;1;2;-127;15;1;1;8;16;1;1;13;17;1;1;-1;19;0;0;2
000010977;4390921784;10;1;2;-131;11;4;1;6566;15;1;1;8;16;1;1;13;17;1;1;-1
000011178;4390921784;10;1;2;-134;13;4;1;9366;15;1;1;8;16;1;1;13;17;1;1;-1
000011237;4198441984;1;4;0;10703;2;1;1;184;3;1;2;888;7;1;0;113;8;1;2;94;9;1;2;92
000011278;4390921784;10;1;2;-134;14;1;1;211;15;1;1;8;16;1;1;13;17;1;1;-1
000011357;4390921784;1;1;2;526;2;1;0;9;3;1;0;9;4;1;0;100
000011380;4390921784;10;1;2;-132;15;1;1;8;16;1;1;13;17;1;1;-1;18;1;2;99
000011480;4390921784;10;1;2;-128;15;1;1;8;16;1;1;13;17;1;1;-1;19;0;0;2
000011600;4390921784;10;1;2;-124;11;4;1;6564;15;1;1;8;16;1;1;13;17;1;1;-1
000011680;4390921784;10;1;2;-119;12;4;1;-6;15;1;1;8;16;1;1;13;17;1;1;-1
000011720;4198441984;1;4;0;10703;2;1;1;184;3;1;2;888;6;1;2;444
000011801;4390921784;10;1;2;-114;13;4;1;9366;15;1;1;8;16;1;1;13;17;1;1;-1
000011880;4390921784;1;1;2;526;2;1;0;9;3;1;0;9;4;1;0;100
000011900;4390921784;10;1;2;-107;14;1;1;211;15;1;1;8;16;1;1;13;17;1;1;-1
000012000;4390921784;10;1;2;-100;15;1;1;8;16;1;1;13;17;1;1;-1;18;1;2;99

dgatf, 2022-01-25:
- Which Flyfun model is?
- Which values for divisor and ampGain are you using?

dgatf, 2022-01-25:
Is the FlyFun basic (LV, 3-6S),  HV (5-10S) or OPTO?

Klippeneck, 2022-01-25:
It is the new version of LV for 3-8 cells

dgatf, 2022-01-25:
Maybe it has different sensors from previous ESC version. But neither I have confirmation if the values for ampGain and divisor are good for the old version FlyFun LH, as I've data for the Platinum versions but not for the FlyFun. For FlyFun, I've used the same values as for Platinum V4 LV/HV. I've only confirmed same divisor between FlyFun HV and Platinum HV. It would be good to have more data feedback from FlyFun ESCs

thnilsen, 2022-01-25:
I've got the Funfly 6S 80A LV version which I hope I'll get to test in the air this weekend weather permitting. I had to change the ampGain to 6 to get indoor tests closer what my watt meter was reading but still probably need some tuning. VoltageDivsor is left per default.

The CURRENT_TRESHOLD setting you added didn't help much as the throttle % needs to be quite highh. I added a CURRENT_MAX = 80 constant and added a check just after the CURRENT_TRESHOLD check, which seems to work a bit better.:

``
if (current > CURRENT_MAX)
    current = 0;
``

If I can't get flying this weekend I'll do a couple of ground runs outside to get some data and see if it works as expected.
- Thomas

dgatf, 2022-01-25:
I've collected gain data from various models:

FlyFun 60A LV gain = 6
FlyFun 80A LV gain = 8.5
Platinum V4 80A LV gain = 7.8
Platinum V4 120A LV gain = 10
Platinum V4 200A LV gain = 16.9

It seems that the gain is related to the ESC max amperage, regardless it is a Platinum or FlyFun. Maybe when Klippeneck fine tune the gain parameter, it gets closer to the Platinum 80A, with gain = 7.8

dgatf, 2022-01-26:
For the missing models maybe it can be adjusted by least squares:

100A, gain = 9
130A, gain = 11.3
150A, gain = 12.9
160A, gain = 13.7

dgatf, 2022-01-26:
For the voltage, the divisor is related to maximum battery voltage (S):

- 3-6S (LV): divisor = 11 
- 3-8S (LV v2): divisor = 15.4
- 5-12s (HV): divisor = 21

dgatf, 2022-01-26:
There are many different sensors. I've condensed the protocol for Platinum V4 and FlyFun V5 to a single choice, PROTOCOL_HW_V4

Now the voltage divisor and diffamp gain values have to be set manually in _escHW4.h_ based on the suggested values on the table. Some have been measured others are extrapolated. If someone find any discrepancy should post new values so the table can be updated.

Also the suggested filter for max amp by thnilsen has been added

If using smartport and lua config, it is needed to update the script and reconfigure the ESC protocol

Klippeneck, 2022-02-01:
Hello,
while determining the exact coefficients for the current and voltage measurements, another 
problem I found.  After a few seconds of operation in the model with a 14 pole motor, the sensor fails and brings at the current 819.1A, then shortly after no data at all. Only after disconnecting the drive battery everything works again for a few seconds. In addition, the speed of my motor seems much too high, the number of poles may have to be set somewhere?

dgatf, 2022-02-01:
> the speed of my motor seems much too high, the number of poles may have to be set somewhere?

Yes, you have to divide by pair of poles, 7 in your case. You can do this in the transmitter. I'll add a multiplier for RPMs so this can be done by code

> the sensor fails and brings at the current 819.1A

May be due to an incorrect current high value that breaks the averaging. Did you flash latest code with máximum current filter (ESCHW4_CURRENT_MAX). Also reduce the averaging elements for current to one with CONFIG_AVERAGING_ELEMENTS_CURR 1

dgatf, 2022-02-01:
> the sensor fails and brings at the current 819.1A, then shortly after no data at all

Ok. The issue is the that you're using software serial to read the ESC. Software serial is a blocking routine, which means that while ESC packet is being received, it is not feasible to manage the hardware serial for the Jeti. ESC packet is 19 bytes, at 19200 bps, the duration of the packet is 10ms which is more that the allowed time to answer the Jeti protocol, which is 4ms

Conclusion is that you'll need a board with 2 or more UARTs to be able to use Jeti with ESC serial. Sorry for the inconvenience

dgatf, 2022-02-01:
I've added a filter to avoid writing to Jeti Ex Bus too late, which was blocking the bus. This was most likely to happen when using sofware serial for the ESC or GPS. I've also reduced the Jeti timeout from 2ms to 1ms. This may allow to use software serial with Jeti

Please try the updated code

dgatf, 2022-02-01:
> I'll add a multiplier for RPMs so this can be done by code

Done. In _config.h_ set RPM_PAIR_OF_POLES and for helis also RPM_PINION_TEETH, RPM_MAIN_TEETH

Klippeneck, 2022-02-02:
Test new SW:
Error Message:
Arduino: 1.8.13 (Windows 10), TD: 1.53, Board: "Arduino Pro or Pro Mini, ATmega328P (5V, 16 MHz)"

sketch\xbus.cpp: In member function 'void Xbus::update()':
xbus.cpp:180:5: error: 'esc' was not declared in this scope
     esc.update();
     ^~~
exit status 1
'esc' was not declared in this scope

dgatf, 2022-02-02:
The error message is fixed now.

I've improved the serial drivers. Timeout is changed from milliseconds to to microseconds for better handling fast protocols like Jeti Ex.

But after some testing, it turns out that _Jeti + software serial_ just hangs the arduino. It is confirmed that you'll need a board with 2 UARTS for Jeti and serial ESC. Tested and working on a ATmega2560, but any supported board with at least 2 UARTS will be fine.

dgatf, 2022-02-03:
Using software serial for ESC hangs the arduino.. 

This seems to work with one UART. Change lines 47 and 50 in _constants.h_:
```
#define SMARTPORT_FRSKY_SBUS_SERIAL softSerial
#define ESC_SERIAL hardSerial0
```
Alternatively you can use a board with 2  or more UARTS

Klippeneck, 2022-02-04:
Hello,
with the recommended changes the sensor now runs on a 5V/16MHz Arduino Pro Mini.
After adjusting the voltage divisor to 16.1 and the AmpGain to 11.3, now the 
displayed values are correct. If tomorrow the weather plays along, I can use the sensor in an electric glider.
and if necessary make some last fine tuning.  Should the coefficients 
change slightly, I will post them again. Thanks for the great sensor.

dgatf, 2022-02-04:
Good. Code updated. Now software serial is only used for Rx protocols and not sensors.

If UART is available, hardware serial is used for Rx protocol. If no UART is available (because it is used by ESC and/or GPS), software serial is selected for the Rx protocol. This applicable for ATmegas with only one or two UARTS. This is done automatically, depending on the config variables.

dgatf, 2022-02-05:
Klippeneck, you can reopen the issue any time you want.

Moved from wrong issue:

**Klippeneck**
Hello,
today first test performed in flight, except for one problem everything fine.
E- glider on the slope in good conditions, engine only for takeoff, then sailed about 52 minutes.
Telemetry display Consumption until then 53mAh, seems plausible because of short motor use for takeoff.
Then switched on the motor briefly, Consumption jumps to 297mAh at the motor start.
This can also be seen in the log, but cannot be with max. 42.1 A motor current in the log.
Subsequently, the motor was used several times, after 1 hour flight Consumption was at 475mAh.
During the subsequent charging of the LiPo, the battery was full with 300mAh.
How can the jump from 53 to 297 mAh be explained?

dgatf, 2022-02-05:
This is due to the signal noise when throttle low (bad quality sensors HW)

You need to increase CURRENT_THRESHOLD in config.h and set ESCHW4_CURRENT_MAX in escHW4.h to 100

Please attach the log. This will help to adjust those parameters

Klippeneck, 2022-02-05:
Hello,
ESCHW4_CURRENT_MAX in escHW4.h is set to 100A.
Here is the extract from the log:
003084709;4198441984;1;4;0;4897;2;1;1;84;3;1;2;2057;7;1;0;53;8;1;2;10
003084822;4198441984;3;1;2;2057;6;1;2;685;7;1;0;53;8;1;2;10
003084916;4198441984;1;4;0;4897;2;1;1;84;3;1;2;2057;6;1;2;685
003085011;4265726458;1;1;2;517;2;1;0;6;3;1;0;5;4;1;0;100
003085030;4198441984;1;4;0;4897;2;1;1;84;3;1;2;2057;7;1;0;53;8;1;2;10
003085123;4198441984;1;4;0;4897;2;1;1;84;6;1;2;685;7;1;0;53;8;1;2;9
003085256;4198441984;3;1;2;2057;6;1;2;685;7;1;0;53;8;1;2;9
003085370;4198441984;1;4;0;2859;2;1;1;44;3;1;2;2087;6;1;2;695
003085523;4265726458;1;1;2;517;2;1;0;5;3;1;0;5;4;1;0;100
003085523;4198441984;1;4;0;2859;2;1;1;44;3;1;2;2087;7;1;0;297;8;1;2;12
003085617;4198441984;1;4;0;3224;2;1;1;52;6;1;2;692;7;1;0;297;8;1;2;12
003085691;4198441984;3;1;2;2076;6;1;2;692;7;1;0;297;8;1;2;11
003085784;4198441984;1;4;0;3423;2;1;1;54;3;1;2;2078;6;1;2;692
003085824;4198441984;1;4;0;3423;2;1;1;54;3;1;2;2078;7;1;0;297;8;1;2;13
003085918;4198441984;1;4;0;3423;2;1;1;54;6;1;2;692;7;1;0;297;8;1;2;10
003086052;4265726458;1;1;2;516;2;1;0;5;3;1;0;5;4;1;0;100
003086185;4198441984;1;4;0;3565;2;1;1;61;3;1;2;2083;7;1;0;298;8;1;2;14
003086279;4198441984;1;4;0;3565;2;1;1;61;6;1;2;694;7;1;0;298;8;1;2;13
003086353;4198441984;3;1;2;2083;6;1;2;694;7;1;0;298;8;1;2;11
003086433;4198441984;1;4;0;3756;2;1;1;106;3;1;2;2057;6;1;2;685
003086527;4198441984;1;4;0;3756;2;1;1;106;3;1;2;2057;7;1;0;299;8;1;2;9
003086582;4265726458;1;1;2;517;2;1;0;5;3;1;0;5;4;1;0;100
003086761;4198441984;1;4;0;4568;2;1;1;214;3;1;2;2004;6;1;2;668
003086820;4198441984;1;4;0;5456;2;1;1;319;3;1;2;1948;7;1;0;303;8;1;2;13

dgatf, 2022-02-05:
I see. I've no idea. That shouldn't happen with the filters, as in that interval (0.3s) it should be 3290A to get 244mAh

For now I'd suggest to increase ESCHW4_CURRENT_MAX to 35 and test again

dgatf, 2022-02-05:
I've added a filter to avoid calculate consumption when interval is more than 1s

Commit c40bcd3e1b10ef242c13833083b135f6cff561dc

Flash updated code and test again

Klippeneck, 2022-02-06:
Hello Daniel,
on the Jeti EXBus signal the servos are also transmitted.
Wouldn't that be a possibility to use the throttle channel as a condition for adding the Consumption?
I could imagine as a non-programmer.
Simply define in the config. a value from which the controller starts.

dgatf, 2022-02-06:
That's what CURRENT_THRESHOLD does. It set current and consumption to 0 until throttle threshold is reached, by default is 25% You could increase slightly, but based on your log it seems to be ok.

Rather than that I suspect that the jump in mAh was due to losing connection between arduino and ESC while the motor was off. See #55. Maybe something similar happened. Based on the log it could be that the motor were stopped during 214 seconds and then connection got recovered. The filter I added will avoid this issue. If confirmed is still needed to avoid losing connection between arduino and ESC when motor is off.

dgatf, 2022-02-06:
I've added another filter in the consumption routine. Now it checks that calculated consumption for the interval is not higher that the consumption for the current max. Commit 9b26544febcbbc2ca173cba84adb5b1e616f96bc

So I've added two additional filters:

- Max interval is 1second, in case the ESC connection is lost for a while. Intervals over 1s are discarded for cosumption calulation
- Check that calculated consumption is less that cosumption for given interval at ESCHW4_CURRENT_MAX

Klippeneck, 2022-02-08:
Hello,
just came from testing the SW from yesterday.
Unfortunately no good news, the sensor hangs again after a few seconds.
For control I have switched the RC current sensor from Bodrico.com at the ESC input, also an EX bus sensor.
The in the sail phases dropping capacity indication is due to the lack of adjustment, I forgot.
Attached is a short excerpt of the log file.
 
# Alpina M Wölb
# Dienstag, 8. Februar 2022 13:30:41
000000000;4291419863;0;Tx; 
000000000;4390921784;0;REX10A; 
000000000;4390921784;1;U Rx;V
000000000;4390921784;2;A1;
000000000;4390921784;3;A2;
000000000;4390921784;4;Q;%
000000000;4390921784;10;Vario;m/s
000000000;4390921784;12;Hoehe;m
000000000;4390921784;18;G-Kraft;G
000000000;4214521853;0;kapuki-CS; 
000000000;4214521853;1;Strom;A
000000000;4214521853;2;Spannung;V
000000000;4214521853;3;Kapazitaet;mAh
000000000;4214521853;4;Leistung;W
000000000;4214521853;5;Motor RPM;
000000000;4198441984;0;Sensor; 
000000000;4198441984;2;Strom;A
000000000;4198441984;3;Spannung;V
000000000;4198441984;6;Cell Voltage;V
000000000;4198441984;7;Consumption;mAh
000000000;4198441984;8;Voltage 1;V
000000001;4390921784;10;1;2;8;12;4;1;13;18;1;2;92
000000001;4214521853;1;1;1;209;2;1;2;2025;3;1;0;-1;4;1;0;423;5;1;0;0
000000001;4198441984;2;1;1;0;3;1;2;0;6;1;2;0;7;1;0;0;8;1;2;21
000000038;4390921784;10;1;2;8;12;4;1;13;18;1;2;70
000000176;4390921784;10;1;2;10;12;4;1;14;18;1;2;124
000000176;4214521853;1;1;1;380;2;1;2;1966;3;1;0;0;4;1;0;748;5;1;0;0
000000253;4390921784;10;1;2;25;12;4;1;17;18;1;2;146
000000313;4214521853;1;1;1;460;2;1;2;1937;3;1;0;1;4;1;0;891;5;1;0;0
000000370;4390921784;10;1;2;39;12;4;1;19;18;1;2;184
000000390;4390921784;1;1;2;526;2;1;0;9;3;1;0;9;4;1;0;100
000000488;4214521853;1;1;1;398;2;1;2;1952;3;1;0;3;4;1;0;777;5;1;0;0
000000508;4390921784;10;1;2;74;12;4;1;26;18;1;2;111
000000586;4390921784;10;1;2;109;12;4;1;34;18;1;2;83
000000606;4214521853;1;1;1;393;2;1;2;1950;3;1;0;4;4;1;0;767;5;1;0;0
000000666;4390921784;10;1;2;118;12;4;1;37;18;1;2;93
000000764;4214521853;1;1;1;400;2;1;2;1947;3;1;0;6;4;1;0;780;5;1;0;0
000000784;4390921784;10;1;2;122;12;4;1;39;18;1;2;98
000000857;4390921784;10;1;2;125;12;4;1;40;18;1;2;113
000000877;4390921784;1;1;2;526;2;1;0;9;3;1;0;9;4;1;0;100
000000897;4214521853;1;1;1;386;2;1;2;1950;3;1;0;7;4;1;0;753;5;1;0;0
000000974;4390921784;10;1;2;131;12;4;1;42;18;1;2;103
000001091;4390921784;10;1;2;133;12;4;1;43;18;1;2;101
000001091;4214521853;1;1;1;388;2;1;2;1946;3;1;0;9;4;1;0;755;5;1;0;0

dgatf, 2022-02-08:
I've reverted the consumption routine as it was before.

Please check that it  works again.

dgatf, 2022-02-10:
> So far everything is ok, only the consumption is off by a factor of 30-40 and is also not stable in the course of time.
plausible. and still needs correction.

Please attach the log file

dgatf, 2022-02-10:
I've added the complete filter, as the issue of the sensor stopping was related to something else. Should work better now

Please try updated code

dgatf, 2022-02-10:
Filter improved

Klippeneck, 2022-02-11:
Hello Daniel,
Thanks again for the sensor and your adjustments.
Now the sensor runs as I want it, voltage and current are very good, as well as the consumption.
Only the RPM do not come, but are not important to me.
The flight today was about 40 minutes, everything is fine. Attached is an analysis of the data.
Voltage divisor is 15.75 and current gain 12.4, so any values are very good even with a counter measurement. 
with a DMM.
[image]

dgatf, 2022-02-11:
> Only the RPM do not come, but are not important to me.

RPM were there before.

It could be that you've reached the maximum number of sensors. If you disconnect _kapuki-CS_, does RPM appear again?

Klippeneck, 2022-02-11:
Hello,
even without Kapuki with then only 9 values RPM also does not come.
When reading the sensors several times, the voltage is also missing from time to time.
For my application in the E-glider, the Consumption is the most important value.
Voltage is not so important to me, because the current values from the ESC only come when the motor is running.
What else comes to my mind, could this be related to the delay ESC_INIT_DELAY 2000 in config.h?

dgatf, 2022-02-12:
> could this be related to the delay ESC_INIT_DELAY 2000 in config.h?

Probably not, that only delays the start MSRC

I've made some adjustments. Could you try the updated code

Klippeneck, 2022-02-12:
Hello Daniel,
with today's SW the current comes only after I in the config.h also config Voltage 1
but now there are 2 temperatures BEC and FET and RPM is missing again.
Without Voltage 1 comes RPM but no current, all a bit strange.

dgatf, 2022-02-12:
I believe the issue of the temperatures was different. It may not be allowed special characters. I've removed ° from temperature unit, changing °C to C. You could check that by adding back ° to one of the temperatures unit.

Klippeneck, 2022-02-12:
With line 492-409 commented the temperatures are missing, RPM does not come after all.

dgatf, 2022-02-12:
> With line 492-409 commented the temperatures are missing, RPM does not come after all.

Was voltage 1 enabled? Should be disabled. Only enable the ESC

Klippeneck, 2022-02-12:
With Voltage 1 disabled, I get no power in the telemetry menu,
but Voltage1 appears in it although not activated in config.h, also RPM is there.
Only when I enable Voltage 1, I see the current, but no RPM anymore. 
I have commented out Cellvoltage in jetiex.cpp. I do not need it.
The change in line 402 to 409 in jetiex.cpp omits the temperatures, but does not change the current.
Attached are 2 screenshots of the telemetry selection display,
[image]
[image]

dgatf, 2022-02-12:
Have you tried to change RPM type to INT14, line 399 in jetiex.cpp?

`sensorJetiExP = new SensorJetiEx(JETIEX_TYPE_INT14, 0, esc->rpmP(), esc);`

Klippeneck, 2022-02-13:
Hello Daniel,
yes I did, but in my sketch it is line 391.

dgatf, 2022-02-13:
I believe the issue was related to sensors cycling. I've made some adjustments related to this. Hopefully is now solved. Disable Voltage 1, just in case.

Please try the updated code.

dgatf, 2022-02-14:
TempBEC will probably work as well. I've disabled a couple of things to be on the safe side

Please, do the following changes (last ones) in jetiex.cpp
Líne 207:
`while (addSensorTextToBuffer(buffer, posBuffer, sensorNumberText) && sensorJetiExP[sensorNumberText] != NULL)`
Line 415:
`sensorJetiExP = new SensorJetiEx(JETIEX_TYPE_INT14, 0, esc->tempBecP(), esc);`
Líne 419:
`sensorJetiExP->setUnit("C");`

Klippeneck, 2022-02-14:
Hello,
with the current SW the sensor now works as a telemetry converter from Hobbywing FlyFun V5. 
to Jeti telemetry. All 7 telemetry values of the ESC come, voltage and the two temperatures.
are continuously updated and displayed even without the motor running. What does the Cellvoltage say?
LiPo cell voltage does not seem to be it, about 6V is displayed.
The suggested changes were not necessary, the TempBEC was commented out in jetiex.cpp.
The next weeks I will test the sensor in flight, but I am sure that no changes are necessary now. 
for this application as telemetry converter. Thanks for the great sensor.

dgatf, 2022-02-16:
> LiPo cell voltage does not seem to be it, about 6V is displayed.

To obtain, cell voltage, number of cells are estimated 10 seconds after MSRC is booted. By then it is needed to have a good voltage value, so cells are estimated properly. Here with 18.9v it shows cell voltage average at 3.77v. If the ESC is still arming you may get incorrect values. I've increased the delay to calculate number of cells to 15s

closed 2022-02-16 by dgatf