dgatf/msrc #59: 328P, Ibus protocol and GPS

erwann56, 2022-02-04:
I am on NANO 328p in USB, GPS BN220 and flysky receiver.
I noticed 2 problems with the GPS that I can't get to work.
As soon as I define "GPS true" in config.h, all the sensors are at zero in my telemetry. (In ibus.cpp, when I put // in front of the LAT, LON and ALT lines, the other sensors work again) Could the problem come from the "IBUS_TYPE_S32" format?
Moreover when I connect the Tx GPS on the PIND7, my card is blocked. Do you think of a solution?
Merci

dgatf, 2022-02-04:
There was a similar issue with Jeti #58. I've updated the code to avoid using software serial for sensors as it blocks the Rx protocol. If there is no enough UARTS, software serial will be used for the Rx protocol. Now this should be done automatically by code. 

Flash latest code and swap IBUS and GPS, connecting IBUS to pins 7 and 12 (with resistor) and GPS to RX

Not tested yet, though

dgatf, 2022-02-04:
Tested and working (1 UART, IBUS+GPS). Swap connections as suggested. Needed to fix a bug. Flash latest code

closed 2022-02-04 by dgatf