dgatf/msrc #60: Problem with "IBUS_TYPE_S32"?

erwann56, 2022-02-05:
Thank you for your work.
The last code is better (less latency) however the telemetry for LAT, LON and ALT blocks the feedback from all sensors. When I comment out the first 3 gps sensors in IBUS.cpp, the telemetry works again. A problem with "IBUS_TYPE_S32"?

[image]
[image]
[image]

dgatf, 2022-02-05:
There was some issues with the IBUS types and also lat/lon where swapped. Fixed with f0f213df78c5358053702289dd7a25e72e4757d6
Please try latest code and check if the problem persists

dgatf, 2022-02-05:
If still doesn't work, change type to _IBUS_TYPE_S16_:
```
sensorIbusP = new SensorIbus(AFHDS2A_ID_GPS_LAT, IBUS_TYPE_S32, gps->latP(), gps);
addSensor(sensorIbusP);
sensorIbusP = new SensorIbus(AFHDS2A_ID_GPS_LON, IBUS_TYPE_S32, gps->lonP(), gps);
addSensor(sensorIbusP);
sensorIbusP = new SensorIbus(AFHDS2A_ID_GPS_ALT, IBUS_TYPE_S32, gps->altP(), gps);
```
I want to check if 4 bytes telemetry values are processed in the transmitter

Klippeneck, 2022-02-05:
Hello,
today first test performed in flight, except for one problem everything fine.
E- glider on the slope in good conditions, engine only for takeoff, then sailed about 52 minutes.
Telemetry display Consumption until then 53mAh, seems plausible because of short motor use for takeoff.
Then switched on the motor briefly, Consumption jumps to 297mAh at the motor start.
This can also be seen in the log, but cannot be with max. 42.1 A motor current in the log.
Subsequently, the motor was used several times, after 1 hour flight Consumption was at 475mAh.
During the subsequent charging of the LiPo, the battery was full with 300mAh.
How can the jump from 53 to 297 mAh be explained?

erwann56, 2022-02-06:
I tried with 2 GPS to be sure.
With the modification f0f213d, same result. All values ​​are sensors are frozen at 0.
By modifying S32 by S16, the sensors work but for the GPS I have as before:
LAT, LON and SPE at 0. COG and ALT work.
I also have a compilation error (see photo).
I had tested another code some time ago. All GPS data uploaded correctly in my remote control.
(https://create.arduino.cc/projecthub/yvesmorele/low-cost-telemetry-and-data-logger-flysky-turnigy-8491d1)
[image]

dgatf, 2022-02-06:
Based on opentx code [flysky_ibus.cpp](https://github.com/opentx/opentx/blob/2.3/radio/src/telemetry/flysky_ibus.cpp), 4 bytes values alone are not processed, only 2 bytes and AC packets (multi value packets)

I've implemented AC packets to be able to send GPS_FULL packet. Commit 924d1e6206e8fed0f8a646994b0a5dbce328f309

Please try updated code

erwann56, 2022-02-06:
speed, cog and voltage are set to zero. Lat, Lon, Alt data are displayed but with inconsistent values
[image]

dgatf, 2022-02-06:
Please upload the image again

erwann56, 2022-02-06:
I will try.
Thank you

dgatf, 2022-02-06:
> Lat, Lon, Alt data are displayed but with inconsistent values

Please post the correct values?

erwann56, 2022-02-06:
cog and voltage values ​​are ok. speed never worked.
the actual values ​​should be:
alt 202m
43.37
-0.35
[image]

dgatf, 2022-02-06:
This will be trial error. It is not clear the format for the multi packet for GPS

I've modified the GPS packet format. Please flash updated code. This time do not connect the GPS and enable SIM_SENSORS in config.h. It will be easier for me the check the values if needed

erwann56, 2022-02-06:
I will try tomorrow.

erwann56, 2022-02-07:
[image]
With the last code and SIM_sensors. GPS was unplugged.

dgatf, 2022-02-07:
There was an issue in the definition of AC packets. Most likely is fixed now. Please try updated code

erwann56, 2022-02-07:
First of all, I have to comment some lines because I have blockages at compilation because of "rpm multiplier" "rpm pinion"...
With the last code, I have the GPS values ​​at 0. The voltage sensor is missing.
In order to confirm the hardware, I tested the other code. There is just a problem with the speed which must be multiplied by 100 (I tested with my car)

[image]
)
[image]

dgatf, 2022-02-08:
Please try updated code. Should be ok now

dgatf, 2022-02-08:
>PS: MRSC_gui win does not generate lines for rpm pinion, teeth...

Binaries updated. Fixed

erwann56, 2022-02-08:
I'm not at home. I will test tomorrow evening. Thank you

erwann56, 2022-02-09:
Great!!
Values ​​are going up nicely. There is just an adjustment to be made with the GPS speed which is divided by 200 (0.2km/h instead of 40km/h).
So I will soon be able to test the BMP280 sensor (-;
Thanks again for your project.
[image]

erwann56, 2022-02-11:
Perfect.
The speed data is correct.
If I can, I test the BMP280 this afternoon.
(Do you know if there is a LUA script or a widget that would allow to have a graphic feedback on the TX16S as in the photo?)

dgatf, 2022-02-11:
There are several scripts lua scripts in this [link](https://giters.com/EdgeTX/lua-scripts)

closed 2022-02-11 by dgatf