dgatf/msrc #61: BMP280 and Ibus

erwann56, 2022-02-11:
I quickly tested the "BMP280 module 3.3V (6 pins, I2C/SPI)" with GPS and voltage.
I have 3 additional sensors which appear TEMP1, 00Gp and Vspd).
Temp1 displays 40° and the others are at 0.
GPS and voltage are OK.
I connected the BMP280 to the arduino nano at 3.3v, GND, A3 and A4. Did I forget something?

dgatf, 2022-02-11:
Fixed now. Flash updated code

erwann56, 2022-02-11:
I always have the BMP values ​​at 0 except TEMP1 at -40.
I did a scan of the i2c port, I can't find my sensor.
I do not understand why. I'll keep you informed.

erwann56, 2022-02-11:
The i2c scan finds it with the address 076. (dry solder on A4)
The telemetry is always identical temp1 -40 and alt at 0.

dgatf, 2022-02-11:
- Check connections
- Check SDA/SCL are not swapped
- You are using arduino at 5V, so you should use the 5V version (4 pins) of the BMP280. See [remark](https://github.com/dgatf/msrc#34-i2c-sensors). If it is the 3.3v version (6 pins), you may have burnt it (it may not get burn if sensor Vcc is connected to 3.3v, but still could be due to I2C at 5v)

erwann56, 2022-02-11:
[image]

erwann56, 2022-02-11:
I will test again but everything seems OK at the sensor level.
For the I2C address, should I leave 119 or put 76?

[image]

erwann56, 2022-02-11:
[image]

It was an address problem. I had to put 118.
However, there is a lag for temperature.
Instead of 21°c I have -18°c.

erwann56, 2022-02-12:
Everything is perfect.

closed 2022-02-12 by dgatf