dgatf/msrc #63: FPort support

matteo-campanella, 2022-02-13:
Any chance to get FPort2 protocol supported ? That being bidirectional would allow to use a single board both for telemetry and ppm demux (obviously with limitations related to available pins)

dgatf, 2022-02-13:
FPort2 receivers (Archers) can not be configured as smartport as well?

So an Archer receiver (like Archer R6) with servos connectors configured with smartport won't do the work as well? An this is compatible with MSRC

matteo-campanella, 2022-02-14:
That is true, it can be configured as smart port, but in the cases when you need also channels output (especially channels usually not present as direct outputs on the receivers), then you need to use both smart port for telemetry and sbus for channels. With FPort you manage channels output and telemetry all in one single stream.
As of now this is feasible but with two separate arduinos, one powered by MSRC and another running the sbus decoding software, which makes the solution a bit bulky.

dgatf, 2022-02-14:
It can be done. Fport telemetry is similar to smartport. Though I'll need to implement sbus as well. I'll add to the todo list, without date

dgatf, 2022-03-14:
I've added support for FPort. I've created a branch until is tested. Could you test the code as I don't have access. You can download from  [https://github.com/dgatf/msrc/tree/fport](url) You can use any board. Once this is verified, I'll add the pwm signals

oilytin, 2025-07-17:
Did this get merged? cant find the branch.