dgatf/msrc #66: Inclusion of number of Sats fixed onto please

Mike-HRC, 2022-02-15:
Hi Daniel - any chance that you could add in the number of satelites fixed onto into the data stream either transmitted as SATS or as TMP2 please.  The GPS widget that I was hoping to try using uses the number of satelites to start GPS logging etc.

Cheers,
Mike

dgatf, 2022-02-15:
Smartport doesn't have a specific  sensor for GPS SATS. Also no one called TMP2. Anyway I've added GPS SATS for smartport to a generic smartport sensor. See 495a80669f5f69303114de26740205986fc31f53
Re-discover sensors and you'll find sensor at 0x5003. Rename the sensor as SATS and this may work with your widget if the condition is against the name sensor string. Otherwise I need to know also the sensor number for SATS expected by the widget

Mike-HRC, 2022-02-15:
Thanks Daniel

Mike-HRC, 2022-02-15:
I'm doing something wrong - but I get an error if I try to change in IDE

Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "Arduino Pro or Pro Mini, ATmega328P (5V, 16 MHz)"

C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\smartport.cpp: In member function 'void Smartport::setConfig(Config&)':

smartport.cpp:483:66: error: no matching function for call to 'Sensor::Sensor(int, uint8_t*, int, Bn220*&)'

       sensorP = new Sensor(DIY_FIRST_ID + 3, gps->satP(), 10, gps);

                                                                  ^

In file included from C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\smartport.h:60:0,

                 from C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\smartport.cpp:1:

C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\sensor.h:18:5: note: candidate: Sensor::Sensor(uint16_t, float*, uint8_t, AbstractDevice*)

     Sensor(uint16_t dataId, float *valueLP, uint8_t refresh, AbstractDevice *deviceP);

     ^~~~~~

C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\sensor.h:18:5: note:   no known conversion for argument 2 from 'uint8_t* {aka unsigned char*}' to 'float*'

C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\sensor.h:8:7: note: candidate: constexpr Sensor::Sensor(const Sensor&)

 class Sensor : public FormatData

       ^~~~~~

C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\sensor.h:8:7: note:   candidate expects 1 argument, 4 provided

exit status 1

no matching function for call to 'Sensor::Sensor(int, uint8_t*, int, Bn220*&)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Mike-HRC, 2022-02-15:
Or if I make change in Notepad and then load smartport.cpp into the folder I get this

Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "Arduino Pro or Pro Mini, ATmega328P (5V, 16 MHz)"

C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\smartport.cpp: In member function 'void Smartport::setConfig(Config&)':

smartport.cpp:483:62: error: no matching function for call to 'Sensor::Sensor(int, uint8_t*, int, Bn220*&)'

   sensorP = new Sensor(DIY_FIRST_ID + 3, gps->satP(), 10, gps);

                                                              ^

In file included from C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\smartport.h:60:0,

                 from C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\smartport.cpp:1:

C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\sensor.h:18:5: note: candidate: Sensor::Sensor(uint16_t, float*, uint8_t, AbstractDevice*)

     Sensor(uint16_t dataId, float *valueLP, uint8_t refresh, AbstractDevice *deviceP);

     ^~~~~~

C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\sensor.h:18:5: note:   no known conversion for argument 2 from 'uint8_t* {aka unsigned char*}' to 'float*'

C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\sensor.h:8:7: note: candidate: constexpr Sensor::Sensor(const Sensor&)

 class Sensor : public FormatData

       ^~~~~~

C:\Users\User\Documents\Arduino\MSRC 0.8 - GPS Sat added\msrc-master S.Port\msrc\sensor.h:8:7: note:   candidate expects 1 argument, 4 provided

exit status 1

no matching function for call to 'Sensor::Sensor(int, uint8_t*, int, Bn220*&)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Mike-HRC, 2022-02-15:
That's worked a treat Daniel - thanks, all sorted now

closed 2022-02-16 by dgatf