dgatf/msrc #71: APD ESC and Jeti telemetry

gamingthemarket, 2022-03-05:
I have an APD F120X ESC and Jeti radio. I want to decode the telem using PWM over a single wire. Can you help me with this diagram? I'm not sure about 5V and ground lines. Thanks kindly!

[image]

dgatf, 2022-03-05:
- telemetry does not need ground
- 5v is ok
- pwm? You meant serial telemtry, don't you?
- jeti goes to pins 7&12 if using ESC with serial telemetry
- esc serial telemetry goes to rx1

gamingthemarket, 2022-03-05:
You are defining the ESC T output (telemetry) signal as serial? Just to clarify, APD F series only use UART as input for firmware. I'll try the other pins. Thanks for a fast response
[image]
!

gamingthemarket, 2022-03-05:
Also, I have read some people use an opto isolator, but I'm not sure if it's necessary on the APD F120X. My max load to the ESC is 150 amps and 35V.

dgatf, 2022-03-05:
With [Beta firmware](https://docs.powerdrives.net/products/configuration/f-series-configuration#telemetry-type) it is feasible to send also full serial telemetry.

Depending on telemetry type, pins varies:

- RPM output (PWM signal). Jeti to pins Rx and Tx. PWM signal to pin 8
- Serial telemetry. Jeti to pins 7 and 12. Serial telemetry to pin Rx

gamingthemarket, 2022-03-05:
Ok we're getting closer. Does Rx1 need a resistor?
[image]

dgatf, 2022-03-05:
> RPM output (PWM signal). Jeti to pins Rx and Tx. PWM signal to pin 8

It is highly recommended 1k resistor at Tx. Though it will work without it

dgatf, 2022-03-05:
- Remove GND and RAW cables from UART and replace with cables from Vin and
GND
- Resistor is on pin TX0, not RX1
- PWM is on pin8 without resistor

gamingthemarket, 2022-03-06:
Still don't get a signal. I am not sure if it is the wiring, software, or Jeti config. The Rx is on Ex Bus. When this works the radio's SENSOR MENU will show a new telemetry option?

Is the baud rate wong?

[image]

My config.h:
#ifndef CONFIG_H
#define CONFIG_H
#include "constants.h"
#define RX_PROTOCOL RX_JETIEX
#define CONFIG_ESC_PROTOCOL PROTOCOL_PWM
#define CONFIG_GPS false
#define GPS_BAUD_RATE 9600 
#define CONFIG_VOLTAGE1 true 
#define CONFIG_VOLTAGE2 false 
#define CONFIG_NTC1 true 
#define CONFIG_NTC2 false 
#define CONFIG_CURRENT true 
#define CONFIG_AIRSPEED false 
#define CONFIG_REFRESH_RPM 1
#define CONFIG_REFRESH_VOLT 1
#define CONFIG_REFRESH_CURR 1
#define CONFIG_REFRESH_TEMP 1
#define CONFIG_REFRESH_DEF 1
#define CONFIG_AVERAGING_ELEMENTS_RPM 1
#define CONFIG_AVERAGING_ELEMENTS_VOLT 1
#define CONFIG_AVERAGING_ELEMENTS_CURR 1
#define CONFIG_AVERAGING_ELEMENTS_TEMP 1
#define CONFIG_AVERAGING_ELEMENTS_DEF 1
#define VOLTAGE1_MULTIPLIER 1
#define CURRENT_MULTIPLIER 1

#define CONFIG_PWMOUT false
#define PWMOUT_DUTY 0.5 // 0.5 = 50%

#define SENSOR_ID 10 // Sensor Id
#define DATA_ID 0x5000 // DataId (sensor type)
#define CURRENT_THRESHOLD 25

gamingthemarket, 2022-03-06:
APD F telemetry is PWM -> pin 8. Jeti Ex Bus output -> Tx. Wiring is the same as my prior post. I got the IDE upload working w/ the full library and config.h.  

I tried this version:
RX_PROTOCOL RX_JETIEX
CONFIG_ESC_PROTOCOL PROTOCOL_APD_F

Also tried:
RX_PROTOCOL RX_JETIEX
CONFIG_ESC_PROTOCOL PROTOCOL_PWM

Jeti radio -> Apps -> Jetibox Emulator = no connection to MX for telemetry. Am I missing something?

dgatf, 2022-03-07:
>  CONFIG_ESC_PROTOCOL PROTOCOL_PWM

This is the correct protocol for telemetry as RPM signal. Moreover, if you select PROTOCOL_APD_F, then Jeti would go other pins

You'll need to enter in debug mode

Change line 77 in constants.h to:

`#define DEBUG_SERIAL softSerial`

Connect arduino GND to programmer GND and arduino pin 12 to programmer Rx

Enable DEBUG_PWM in config.h and post output with serial monitor at 115200

Disable DEBUG_PWM and enable DEBUG_PACKET and post output

gamingthemarket, 2022-03-09:
The ESC is faulty has to be replaced. I am going to table this project. Thanks for working with me amigo, and helping to push the hobby in a good direction.

dgatf, 2022-03-09:
You're welcome

closed 2022-03-09 by dgatf