dgatf/msrc #72: Problem when read RPM from the tachometer socket of a CDI petrol engine

roscor61, 2022-03-07:
Hello,

I tried to read the signal from the tachometer socket of the CDI of an RC petrol engine.
On the oscilloscope, It is normal square wave signal with 35% duty cycle and 5vpp signal.
The useful frequency range goes from 7 hz to 370 hz approximately (420 rpm to 22,200 rpm).
From 250hz upwards msrc sends the right value in telemetry, below 250hz (15.000rpm) approximately the value always drop to 0 rpm.
Why?

Settings: Multipl: 1, Blade: 1

Is the system set up for a higher frequency range produced by the ESCs?

Is a hw fequency multiplier necessary?

Thank you
Rosario

dgatf, 2022-03-07:
I've increased the lower rpm range lately. Be sure you download the latest code

Which board for MSRC are you using?

djatila666, 2022-03-07:
> Hello,
> I am using Arduino Nano 16Mhz / 5V for the final layout and an Arduino R3
> for testing.
> I'll download the new fw soon and I'll let you know ...
> Thank you for the attention (xscuse me for the bad English)

Hi, how do you Connect CDI to Arduino?
Thanks

roscor61, 2022-03-07:
I tried connecting the thacometer output directly to both the D8 and GND pin.
I also tried using an optocoupler with 10k pullup but the result is the same ..
Below 250hz the telemetry go to 0 rpm.

dgatf, 2022-03-07:
I believe 250Hz was the old code.

I've just tested with the latest code and arduino nano at 16MHz, can read as low as 60 rpm (1Hz)

closed 2022-03-12 by dgatf