dgatf/msrc #75: Add CRSF (Crossfire) support

pierrotm777, 2022-03-11:
Hello,

Is it possible to add CRSF telemetry for use with ExpressLRS and TBS devices ?

Thanks,

Pierre

pierrotm777, 2022-03-12:
Hello,
I have ELRS and TBS systems.
I can help you for these tests. Thanks

dgatf, 2022-03-12:
MSRC will send only GPS and battery telemetry. There are no other sensors available in CRSF telemetry protocol. Only GPS and  analog voltage and current can be sent. No vario, airspeed, temperature and ESC telemetry will be sent. Would that be ok?

pierrotm777, 2022-03-12:
I think it's possible to have also vario.
But gps and vbat are ok for me.

dgatf, 2022-03-12:
That's right. I'll include vario speed

pierrotm777, 2022-03-12:
Cool,
Thanks :-)

dgatf, 2022-03-13:
I've created a branch for CRSF. I'll merge when tested

Please download and flash from [https://github.com/dgatf/msrc/tree/crsf](url)

Which board are you going to use?

pierrotm777, 2022-03-13:
Sorry for the delay,
I think to use an Teensy 3.2 or LC ?
I suppose they can work in 400K bauds ?

dgatf, 2022-03-13:
Yes. Better the LC, as I have one LC around. Easier for me to check in case of issues

Also arduino (atmega328, atmega32u4...) will work, if not connecting to GPS or ESC serial. Otherwise you need a Teensy

erwinf, 2022-03-15:
I would like to try this out as well with elrs and a pro mini board

dgatf, 2022-03-15:
[crsf branch](https://github.com/dgatf/msrc/tree/crsf) is already configured with a simulated voltage sensor. You don't need to install the sensor yet. Only connect the crsf data to pin 1 for the teensy and to pins rx and tx for the pro mini. Also connect Vcc and gnd. Check for voltage telemetry value 12.34v in the transmitter.

dgatf, 2022-03-15:
> Hello,
> I have ELRS and TBS systems.
> I can help you for these tests. Thanks

I though you were able to test it

pierrotm777, 2022-03-15:
Yes, i have well tbs and elrs systems but OpenAVRc isn't ready for the crsf telemetry. We work on actually.

dgatf, 2022-03-15:
No worries. When you're ready

secretman28, 2022-04-25:
i can help here , im want to convert s-port telemetry to crsf to get telemetry from my turbine jet , that send frsky telemetry, remote is ready, crsf have some sensor available https://github.com/ArduPilot/ardupilot_wiki/blob/master/images/crossfire-telemetry-meaning.jpg we can induce a change in name to get the sensor we want and in the lua of the remote we can manage to get it as we wanted lets say we send the fuel consumption from the frsky in roll imput to the crossfire and in the remote we use the roll as a fuel comsumption  easy manage by the user for what he want to do, im far from a coder but i have a base to make thing work as i want lol

m0p3d, 2023-09-23:
Any news? ELRS adaption increasing would be very helpful to get CRSF Telemetry working

CarlouT, 2024-09-01:
Just want to know if Telemetry for Express LRS is already supported?

ahfynjj, 2024-10-28:
Is there any latest progress? Thank you

ahfynjj, 2024-10-29:
How to connect CRS? The receiver usually has two wires, TX and RX, but the msrcconfig shows only one wire.

ahfynjj, 2024-10-29:
I connected gpio0 to the receiver's TX, and gpio1 to the receiver's RX. The vspeed was displayed normally and locked at 0.05m. I tried to connect the BL32 ESC's TX to gpio5, and tried to turn on the ADP HV and ADP F options, but all failed.
Below is the debug,
[image]

USB. Debug enabled
USB (96)
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 
CRSF. Bad CRC or header
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x1A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0xA7 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0xC 0x14 0x24 0x20 0x64 0xA 0x0 0x8 0x2 0x0 0x0 0x0 0xF0 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
[... cut ...]
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x9A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0x7B 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x1A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0xA7 
CRSF (222) > 0xC8 0x4 0x7 0x0 0x5 0x8 
CRSF (222) < 0xC8 0x18 0x16 0xE0 0xC3 0x1A 0x2B 0xC0 0xF7 0xB 0xC 0x82 0xF 0x7C 0xE0 0x3 0x0 0x0 0x0 0x0 0x0 0x0 0x4C 0x7C 0xE2 0xA7

dgatf, 2024-10-29:
> I connected gpio0 to the receiver's TX, and gpio1 to the receiver's RX. The vspeed was displayed normally and locked at 0.05m. I tried to connect the BL32 ESC's（flycolor XcrossHV 80A) TX to gpio5, and tried to turn on the ADP HV and ADP F options, but all failed.

Great. If you see vspeed at 0.05m/s in your transmitter, then it's perfectly as expected. Only vario is enabled with 0.05m/s value, despite of sensors defined in msrc_gui. This was to test the protocdl.

That means I can easily implement the rest of sensors. I'll et you kow when it's completed so you can test the rest of telemetry values for CRSF.

dgatf, 2024-10-29:
Please download updated [firmware](https://drive.google.com/file/d/1pzIt4XdcLxiSUfSTfyoh_w5tLTFd9JN2/view?usp=drive_link).

Hopefully all capable sensors for CRSF are implemented. Though the telemetry for CRSF is limited. If using a blheli esc with kiss telemetry you have to select _APD F_. I'll update to a more general designation: _Kiss_.

For CRSF with any esc you will get as much as voltage, current and consumption.

Available sensors for CRSF are: GPS, vario and battery.

Let me know if any issue. Haven't checked it deeply.

dgatf, 2024-10-31:
Do you also see Rx Batt, Rx Curr and Rx Cons without msrc connected?

ahfynjj, 2024-11-01:
[log.txt](https://github.com/user-attachments/files/17601256/log.txt)
Unlike last time, I did not connect the ESC today, and the Rx Batt, Rx Curr and Rx Cons of msrc remained fixed values ​​without change, which was the same as the last time the ESC was connected.
The receiver I used was BETAFPV SuperP 14CH. I used a u-blox M10 GPS from GEPRC, turned on the GPS option of MSRC, and the baud rate was 115200. The relevant content of GPS did appear in the telemetry interface of the remote control, but the value remained fixed without change, as shown in the figure below. Because of the typhoon, I couldn't go outdoors, so I put the GPS next to the window. The PPS light of the GPS showed that it seemed to have obtained inaccurate coordinate information, and from my experience.
The above is translated using Google Translate. I hope it is accurate. Thank you!
[image]

dgatf, 2024-11-01:
Ok. We are using three type of telemetry packets. Let's review them.

**Vario**
This vario is a different telemetry packet than the first test. This one shows the vspeed but also the altitude. Even if not connected, could you confirm that it shows vspeed and altitude (it will show zero values if vario sensor not connected).

**GPS**
- Latitude and longitude. It is shown but with incorrect values, isn't it? I'll check that.
- Speed, heading and sats. Are the values shown in transmitter correct?

**Battery**
There are two possibilities. Either the packet is malformed or this packet is already used by the receiver. Could ask someone if they can see both, receiver voltage, current and consumption and at the same time voltage, current and consumption of the battery?

dgatf, 2024-11-01:
Regarding gps and based on your log, your gps module only outputs GPRMC packets, which includes time (but not available in crsf gps telemetry packet), lat, lon, speed and heading but no sats. Sats won't be available in telemetry.

dgatf, 2024-11-01:
Ok. There were some issues with the telemetry values. Should fixed with commit 51ac5645e7a95652f17c1eeb54d51149421cd6f7.

Please, check if the three telemetry packets are correct. Check also battery (with esc for example).

Let me know if any issue.

dgatf, 2024-11-02:
It seems that telemetry battery packet is already used by the receiver. See [tbs-crossfire-manual.pdf](https://www.team-blacksheep.com/media/files/tbs-crossfire-manual.pdf), page 56. At least there is only one voltage, current and consumption sensor. Don't know if it is feasible to disable the battery telemetry from the receiver, so you could use to measure the battery voltage, current and consumption. 

If you could confirm that gps and vario works, I'll consider this protocol as completed.

ahfynjj, 2024-11-02:
The feedback from GPS and ESC (bl32) are displayed normally. Thank you very much.
[log.txt](https://github.com/user-attachments/files/17607235/log.txt)

dgatf, 2024-11-02:
Ok. Great. So indeed, now you also see voltage, current and consumption besides the receiver voltage currant and consumption as well. Don't you?

Anyway, I consider CRSF completed.

Thank you for the tsting.

closed 2024-11-02 by dgatf