dgatf/msrc #76: Can the GPS speed be converted from KTS to KM/H

ahfynjj, 2022-03-12:
Can the GPS speed be converted from KTS to KM/H, and what is the current GPS refresh rate? Is it possible to increase the refresh rate to 10HZ per second!
Greetings from eastern countries, best wishes！
Thanks ！

dgatf, 2022-03-12:
Speed is read in kts from the GPS, then is converted to the units for the specific receiver protocol (frsky expects kts, ibus expects kmh...). If using opentx you can change the units by editting the telemetry sensor properties. The only exception is the Jeti protocol. Are you using Jeti? In this case, I could add an option to select the speed units.

dgatf, 2022-03-12:
GPS refresh rate can be adjusted for frsky (d or smartport). For other receiver protocols is the time between telemetry frames (at maximum feasible rate) and depends on how many other sensors are enabled.

I'll add a parameter for frsky, as now is hardcoded to 500ms

Note that the default refresh rate for GPS is 1Hz. You'll need to increase that as well. Also I recommentd to increase GPS baud rate to 115200

I'll let you know when gps refresh rate parameter is implemented (for frsky)

ahfynjj, 2022-03-12:
Thank you very much for your reply
I'm using JETI ExBus, really looking forward to if you can add an option to select the speed units for KM/H

Is it possible to add a 10hz refresh rate option to JETI ExBus? I expect to DIY a sensor and put it on the jet. The 10hz refresh rate is an ideal choice. Or there is an option to choose the refresh rate at compile time

Thanks again

dgatf, 2022-03-12:
I've added a parameter to select gps speed units for Jeti. This can be selected with msrc_gui or manually in config.h (#define JETI_GPS_SPEED_UNITS_KMH). I've set default to km/h

I've also added a parameter for the gps refresh rate, that can be selected from msrc_gui or in config.h (CONFIG_REFRESH_GPS). This only apply for frsky protocols

The gps refresh rate parameter doesn't matter if using Jeti protocol, as is already to the maximum rate. There are 10 sensors for gps and MSRC sends a telemetry packet every 2 Jeti frames (this is to alow other sensors out of MSRC to send telemetry if connected). So gps sensor refresh rate now is be 200ms, which is 50Hz (if there are other sensors enabled refresh frecuency will lower)

If telemetry packet were send with every Jeti frame, then it could reach your desired refresh rate, 100ms (10Hz). Also some gps sensors may be removed, like date, time, sats... But to send telemetry with every Jeti frame or remove some gps sensors, you would need to do manually.

You will need to change the configuration for the GPS module (BN220?) and select 5Hz to 10Hz refresh rate and increase baud rate to 115200. With this rates you will need a board with 2 UARTS. It will be too much for software serial. I recommend TeensyLC

Also select correct baud rate for MSRC in accordance with the baud rate you selected in the GPS module

ahfynjj, 2022-03-13:
I have purchased a GPS module with baud rate 115200 baud rate and support 10Hz refresh rate, the model is TAU1201, AliExpress link[]( https://www.aliexpress.com/item/1005002382964905.html)
I will go to the next test after receiving it.

thanks

dgatf, 2022-03-13:
I forgot that Jeti sends multiples sensors per frame. All GPS data is sent in two telemetry packets. Jeti sends 1 frame every 10ms. So refresh rate is 40ms (2 packets with data, 2 packets empty). Which is 25Hz

closed 2022-03-17 by dgatf