dgatf/msrc #79: msrc with Pololu ATmega328PB

Klippeneck, 2022-03-31:
Hello Daniel,
which board type to select in config.h when using a Pololu ATMega328PB board?  
Jürgen

dgatf, 2022-04-01:
You have to add a new board to Arduino IDE. See [instructions](https://www.pololu.com/docs/0J74/all). Then select Pololu A-Star 328PB

You may download the latest code as I just have fixed some issues with this board:
- Changed software serial Tx pin from 12 to 9 as it was conflicting with RX1
- Fixed Castle link issue

Also I plan to add support for the second I2C port. This will allow to use I2c sensor (barometer) with XBUS and Hitec

dgatf, 2022-04-01:
> which board type to select in config.h when using a Pololu ATMega328PB board?

You don't need to define the board in config.h. The board selection in msrc_gui is only used to generate the image of the circuit

closed 2022-04-01 by dgatf