dgatf/msrc #86: Castle Link and Teensy LC

jdowdle, 2022-06-21:
Im getting Telemetry, the values looked saturated, but tried throttle and it is unresponsive now.  I enabled feature "Live Link Enable" on the other page of Castle Link 3.85.00.  It is confusing that the ?help button describes this as a USB telemetry, I hope I enabled the correct option.

I currently have the esc input connected to 2 & 6 on the teensyLC, with a 3.3k to 5 V.  Your doc on the Castle branch shows using 5V pullup in the Castle/Teensly LC schematic, the table is silent on what to pullup to.  The main branch shows pullup to 3.3V for Castle.  I am confused whether I should be using the main branch with Castle link configured, or the Castle branch.  Started with Main, now have Castle programmed.

I initially connected the throttle out from Receiver to pin 22 as shown in the schematic in the Main branch, but  have corrected to pin 16 now.

BTW, I get no ESC beeping when I apply power, but noticed when I disconnect receiver power I get the startup ESC beeps.  But no throttle response with power on after that.  I verified that Live Link Enable is active in the Castle Link programming tool.

I believe thats it, any ideas?  thanks.

ahfynjj, 2022-11-09:
I have the same problem as you

dgatf, 2022-11-29:
> I enabled feature "Live Link Enable" 

Yes, that's correct

>  The main branch shows pullup to 3.3V

Pull up to Teensy pin 3V
 
> I initially connected the throttle out from Receiver to pin 22 as shown in the schematic in the Main branch, but have corrected to pin 16 now.

Sorry, connect Rx to pin 16 as defined in the table. Schematic to pin 22 is wrong

> BTW, I get no ESC beeping when I apply power, but noticed when I disconnect receiver power I get the startup ESC beeps. 

That means ESC is not getting proper signal from Teensy. Check the throttle signal from receiver is connected to pin 16 and the pull up is installed. If this is correct, try with a lower resistance, 1k

Just tested and working with Teensy LC

dgatf, 2022-11-29:
Schematic fixed with 0e726eaea269905623cfb59bc3ca438d9e1a8f03

closed 2022-11-29 by dgatf