dgatf/msrc #90: RP2040 and Smartport with Castle question

ahfynjj, 2022-12-03:
Using RP2040, I connected the castle160 ESC through s.port. When the Board GPIO 4 (Castle. Receiver signal) is not connected to the receiver, the feedback of the remote controller is displayed normally. When I connect the Board GPIO 4 to the receiver, the feedback The transmitted information will be lost, but the remote control pushes the accelerator, and the ESC will drive the motor to rotate. This question is very strange.
English is not good, I hope you can understand.

dgatf, 2022-12-06:
Fixed with ed3d98a1766f31775f09d71143668b9964ae28a7. Please try updated [firmware](https://github.com/dgatf/msrc/blob/master/binaries/MSRC-RP2040.uf2)

ahfynjj, 2022-12-07:
The returned value is displayed, but it seems that the value is wrong. "ESC RPM" stays at "14592/rpm" when the motor stops; "ESC Current" is fixed at "204.80A" without any change; "ESC Concentration" is always "0mAH".

dgatf, 2022-12-08:
I've made some adjusments. Please try updated [firmware](https://github.com/dgatf/msrc/blob/master/binaries/MSRC-RP2040.uf2)

dgatf, 2022-12-08:
Some debug is needed. Please flash this firmware (no need to config it). Disconnect power supply from battery to the rp2040 and connect rp2040 to usb. Then with serial monitor (you can use arduino ide or putty) post output

[MSRC-RP2040_castle.zip](https://github.com/dgatf/msrc/files/10187500/MSRC-RP2040_castle.zip)

Unzip as u2f can't be attached here

ahfynjj, 2022-12-14:
Is there any data, I'm not sure if it's done correctly, please watch the video
https://youtu.be/ODJKWpRl3E4

dgatf, 2023-07-21:
Fixed with 33585ca3531d00a9905ecfa0e9957d370f44e846

closed 2023-07-21 by dgatf