dgatf/msrc #91: RP2040 and Jeti

Klippeneck, 2022-12-05:
Hello Daniel,
a super project with the RP2040.
My test setup with FunFly 80A and MS5611 for Jeti does not run, does not send data and the LED does not flash. Programmed with the UF2 file and confugured with the setuptool. 
Wiring checked several times, everything according to plan. How do I get the sensor to run?

Klippeneck, 2022-12-07:
Hello Daniel,
I have tested the new SW, registration of the sensor in the transmitter now works, but no values come from the ESC, are always at zero. The LED does not flash.
To test the board, I flashed and configured the OXS SW from mstrens, the LED flashes and the telemetry values come. Unfortunately, the sensor can not ESC values.

Klippeneck, 2022-12-07:
Hello, 
just tested mt Vario, no values in the display, stable 0.

dgatf, 2022-12-07:
Check wiring. Enable debug from msrc_gui and post the output (you can use arduino ide serial monitor). Remember to disconnect power supply from the battery to the rp2040 when connected to usb

dgatf, 2022-12-07:
Also, check if enabling voltage analog only, you get values different than 0. For this, disable debug. You don't need to connect anything to that pin, it will be floating, or connect 3.3v from the rp2040

dgatf, 2022-12-07:
There is no reading from the esc. Is it enabled and connected to gpio5?

Klippeneck, 2022-12-07:
Hello Daniel,
ESC is activated and connected to GPIO 5.
Telemetry only works with voltage measurement (floating between 4 and 5V ).
Both ESC values and vario always shows zero.
Temperature of the vario sensor works, but has the unit mAh.
The sensor temperature changes when warming up the same.

dgatf, 2022-12-07:
Then telemetry seems to be ok. It seems to be an issue with the esc only. Are you able to check the output of the esc otherwise?

Klippeneck, 2022-12-07:
Hello,
yes I have tested the ESC telemetry output with a msrc sensor based on Arduino,
works. All telemetry values are ok and update continuously, except for the current, but that must be because of the 25% threshold. Strange is that the temperature of the MS 5611 is continuously updated, but altitude and Vspeed are not. From the ESC no current values come at all, only always zero.

dgatf, 2022-12-07:
You need to connect esc ground to rp2040, is it?

Klippeneck, 2022-12-07:
Yes, over BEC

dgatf, 2022-12-07:
Ok. Check that esc firmware is not the vbar version

Klippeneck, 2022-12-07:
I can not read the SW of the controller, do not have an interface for it.
Actually, the telemetry works via an Arduino MSRC.

dgatf, 2022-12-07:
Then, its ok...

dgatf, 2022-12-07:
Please, save and attach your config from msrc_gui

Klippeneck, 2022-12-07:
[image]

dgatf, 2022-12-07:
I think I've found the issue. Please try updated [firmware](https://github.com/dgatf/msrc/blob/master/binaries/MSRC-RP2040.uf2)

Klippeneck, 2022-12-08:
Hi Daniel,same problem with the new SW, but the debug log looks a bit different.
ESC values stable at zero, as well as variable values.
Only voltage and sensor temperature of the MS 5611 work.
Attached is a debug log.
09:19:24.736 -> USB (138)
09:19:24.924 -> MS5611 (98) < Temp: 18.00 Pressure: 45935 Altitude: 0.00 Vspeed: 0.00
09:19:25.158 -> MS5611 (98) < Temp: 18.00 Pressure: 45935 Altitude: 0.00 Vspeed: 0.00
09:19:25.252 -> Vspeed (108): 0.00
09:19:25.345 -> MS5611 (98) < Temp: 18.00 Pressure: 45934 Altitude: 0.00 Vspeed: 0.00
09:19:25.580 -> MS5611 (90) < Temp: 18.00 Pressure: -74511 Altitude: 0.00 Vspeed: 0.00
09:19:25.767 -> Voltage (104): 4.56
09:19:25.767 -> Vspeed (108): 0.00
09:19:25.814 -> MS5611 (90) < Temp: 18.00 Pressure: 45935 Altitude: 0.00 Vspeed: 0.00
09:19:25.814 -> USB (138)
09:19:26.001 -> MS5611 (90) < Temp: 18.00 Pressure: 45934 Altitude: 0.00 Vspeed: 0.00
09:19:26.236 -> MS5611 (90) < Temp: 18.00 Pressure: 45934 Altitude: 0.00 Vspeed: 0.00
09:19:26.330 -> Vspeed (108): 0.00
09:19:26.423 -> MS5611 (90) < Temp: 18.00 Pressure: 45935 Altitude: 0.00 Vspeed: 0.00
09:19:26.564 -> Jeti Ex(210) < 3D 1 8 0 3A 0 41 57 
09:19:26.564 -> Jeti Ex(128) < 3D 1 8 0 3A 0 41 57 
09:19:26.564 -> Jeti Ex Text (128) > 3B 1 18 0 3A 10 F E 0 A4 0 A4 0 1 1B 52 50 4D 52 50 4D BE 8A 27 
09:19:26.611 -> Jeti Ex(108) < 3D 1 8 46 3A 0 EE 87 
09:19:26.611 -> Jeti Ex(108) < 3D 1 8 46 3A 0 EE 87 
09:19:26.611 -> Jeti Ex Values (108) > 3B 1 23 46 3A 1B F 59 0 A4 0 A4 0 14 0 0 0 21 0 20 31 0 40 41 0 0 51 0 0 61 0 40 1B CE DF 
09:19:26.611 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.611 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.611 -> Jeti Ex Values (102) > 3B 1 21 46 3A 19 F 57 0 A4 0 A4 0 74 0 0 0 81 C8 41 91 12 0 A1 0 0 B4 0 0 0 B1 FD DF 
09:19:26.658 -> MS5611 (90) < Temp: 18.01 Pressure: 45936 Altitude: 0.00 Vspeed: 0.00
09:19:26.658 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.658 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.658 -> Jeti Ex Values (102) > 3B 1 23 46 3A 1B F 59 0 A4 0 A4 0 14 0 0 0 21 0 20 31 0 40 41 0 0 51 0 0 61 0 40 1B CE DF 
09:19:26.658 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.704 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.704 -> Jeti Ex Values (102) > 3B 1 21 46 3A 19 F 57 0 A4 0 A4 0 74 0 0 0 81 C8 41 91 12 0 A1 0 0 B4 0 0 0 B1 FD DF 
09:19:26.704 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.704 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.704 -> Jeti Ex Values (102) > 3B 1 23 46 3A 1B F 59 0 A4 0 A4 0 14 0 0 0 21 0 20 31 0 40 41 0 0 51 0 0 61 0 40 1B CE DF 
09:19:26.751 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.751 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.751 -> Jeti Ex Values (102) > 3B 1 21 46 3A 19 F 57 0 A4 0 A4 0 74 0 0 0 81 C8 41 91 12 0 A1 0 0 B4 0 0 0 B1 FD DF 
09:19:26.751 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.799 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.799 -> Jeti Ex Values (102) > 3B 1 23 46 3A 1B F 59 0 A4 0 A4 0 14 0 0 0 21 0 20 31 0 40 41 0 0 51 0 0 61 0 40 1B CE DF 
09:19:26.799 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.799 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.799 -> Jeti Ex Values (102) > 3B 1 21 46 3A 19 F 57 0 A4 0 A4 0 74 0 0 0 81 C8 41 91 12 0 A1 0 0 B4 0 0 0 B1 FD DF 
09:19:26.845 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.845 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.845 -> Jeti Ex Values (102) > 3B 1 23 46 3A 1B F 59 0 A4 0 A4 0 14 0 0 0 21 0 20 31 0 40 41 0 0 51 0 0 61 0 40 1B CE DF 
09:19:26.845 -> Voltage (104): 4.58
09:19:26.845 -> Vspeed (108): 0.00
09:19:26.845 -> USB (138)
09:19:26.845 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.845 -> MS5611 (90) < Temp: 18.00 Pressure: 45933 Altitude: 0.00 Vspeed: 0.00
09:19:26.845 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.892 -> Jeti Ex Values (102) > 3B 1 21 46 3A 19 F 57 0 A4 0 A4 0 74 0 0 0 81 CA 41 91 12 0 A1 0 0 B4 0 0 0 B 76 CE 
09:19:26.892 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.892 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.892 -> Jeti Ex Values (102) > 3B 1 23 46 3A 1B F 59 0 A4 0 A4 0 14 0 0 0 21 0 20 31 0 40 41 0 0 51 0 0 61 0 40 1B CE DF 
09:19:26.939 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.939 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.939 -> Jeti Ex Values (102) > 3B 1 21 46 3A 19 F 57 0 A4 0 A4 0 74 0 0 0 81 CA 41 91 12 0 A1 0 0 B4 0 0 0 B 76 CE 
09:19:26.939 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.939 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.939 -> Jeti Ex Values (102) > 3B 1 23 46 3A 1B F 59 0 A4 0 A4 0 14 0 0 0 21 0 20 31 0 40 41 0 0 51 0 0 61 0 40 1B CE DF 
09:19:26.986 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.986 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:26.986 -> Jeti Ex Values (102) > 3B 1 21 46 3A 19 F 57 0 A4 0 A4 0 74 0 0 0 81 CA 41 91 12 0 A1 0 0 B4 0 0 0 B 76 CE 
09:19:26.986 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:27.048 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:27.048 -> Jeti Ex Values (102) > 3B 1 23 46 3A 1B F 59 0 A4 0 A4 0 14 0 0 0 21 0 20 31 0 40 41 0 0 51 0 0 61 0 40 1B CE DF 
09:19:27.048 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:27.048 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:27.048 -> Jeti Ex Text (102) > 3B 1 1A 46 3A 12 F 10 0 A4 0 A4 0 2 39 43 75 72 72 65 6E 74 41 B3 1C 92 
09:19:27.048 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:27.095 -> MS5611 (90) < Temp: 18.01 Pressure: 45934 Altitude: 0.00 Vspeed: 0.00
09:19:27.095 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
[... cut ...]
09:19:27.095 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:27.095 -> Jeti Ex Values (102) > 3B 1 23 46 3A 1B F 59 0 A4 0 A4 0 14 0 0 0 21 0 20 31 0 40 41 0 0 51 0 0 61 0 40 1B CE DF 
09:19:27.142 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:27.142 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:27.142 -> Jeti Ex Values (102) > 3B 1 21 46 3A 19 F 57 0 A4 0 A4 0 74 0 0 0 81 CA 41 91 12 0 A1 0 0 B4 0 0 0 B 76 CE 
09:19:27.142 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87 
09:19:27.189 -> Jeti Ex(102) < 3D 1 8 46 3A 0 EE 87

dgatf, 2022-12-08:
Flash this firmware and post output. It is already configured (no need to use msrc_gui) with esc hw4 and some debug that I want to check. Please post the output

[MSRC-RP2040.zip](https://github.com/dgatf/msrc/files/10185737/MSRC-RP2040.zip)

Unzip it, as uf2 files cannot be attached here

Klippeneck, 2022-12-08:
Hi Daniel,
19:51:55.715 -> 0 TO
19:51:55.715 -> OK 0 TO
19:51:55.762 -> OK 0 TO
19:51:55.809 -> OK 1 0 TO
19:51:58.940 -> OK FF TO
19:51:58.940 -> OK 0 TO
19:51:58.940 -> OK 33 20 20 8 8 8 48 2 2 2 2A FA 3 2 6A 1A 35 C1 E5 TO
19:51:58.986 -> OK 0 TO
19:51:58.986 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:51:59.221 -> OK 0 TO
19:51:59.221 -> OK 33 20 A0 8 8 8 48 2 2 2 2A 2 0 0 D 42 D 6F B9 TO
19:51:59.408 -> OK 0 TO
19:51:59.408 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:51:59.596 -> OK 0 TO
19:51:59.643 -> OK 33 20 20 8 8 8 48 2 2 2 2A 2A 0 0 D 42 D 6F B9 TO
19:51:59.830 -> OK 0 TO
19:51:59.830 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:52:00.017 -> OK 0 TO
19:52:00.017 -> OK 33 20 A0 8 8 8 48 2 2 2 2A 22 0 0 D 42 D 6F B9 TO
19:52:00.205 -> OK 0 TO
19:52:00.205 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:52:00.408 -> OK 0 TO
19:52:00.455 -> OK 33 20 20 22 20 20 20 26 20 20 A0 10 1C 10 50 28 AA E8 2B FF TO
19:52:00.595 -> OK 0 TO
19:52:00.595 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:52:00.830 -> OK 0 TO
19:52:00.830 -> OK 33 20 A0 22 20 20 20 26 20 20 A0 10 1C 10 50 48 AA E8 2B FF TO
19:52:01.017 -> OK 0 TO
19:52:01.017 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:52:01.205 -> OK 0 TO
19:52:01.205 -> OK 33 20 20 22 20 20 20 26 20 20 A0 90 1C 10 50 A8 AA E8 2B FF TO
19:52:01.392 -> OK 0 TO
19:52:01.392 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:52:01.626 -> OK 0 TO
19:52:01.626 -> OK 33 20 A0 22 20 20 20 26 20 20 A0 D0 3 2 6A 12 35 BD E5 TO
19:52:01.814 -> OK 0 TO
19:52:01.861 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:52:02.001 -> OK 0 TO
19:52:02.001 -> OK 33 20 20 11 10 10 90 13 10 10 50 10 1C 10 50 8 A8 8 E5 TO
19:52:02.189 -> OK 0 TO
19:52:02.236 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 TO
19:52:02.423 -> OK 33 20 A0 11 10 10 90 13 10 10 50 90 1C 10 50 88 6A 7A E5 TO
19:52:02.611 -> OK 0 TO
19:52:02.657 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:52:02.798 -> OK 0 TO
19:52:02.798 -> OK 33 20 20 11 10 10 90 13 10 10 50 10 1C 10 50 28 AA E8 2B FF 0 TO
19:52:03.032 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:52:03.220 -> OK 0 TO
19:52:03.220 -> OK 33 20 A0 11 10 10 90 13 10 10 50 D0 3 2 6A D2 6A 82 E5 TO
19:52:03.407 -> OK 0 TO
19:52:03.454 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:52:03.642 -> OK 0 TO
19:52:03.642 -> OK 33 20 20 23 20 20 20 26 20 20 A0 D0 3 2 6A EA 6A 7A E5 TO
19:52:03.829 -> OK 0 TO
19:52:03.876 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:52:04.017 -> OK 0 TO
19:52:04.017 -> OK 33 20 A0 23 20 20 20 26 20 20 A0 D0 3 2 6A 12 35 BD E5 TO
19:52:04.204 -> OK 0 TO
19:52:04.204 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:52:04.438 -> OK 0 TO
19:52:04.438 -> OK 33 20 20 23 20 20 20 26 20 20 A0 10 1C 10 50 68 AA 8 E5 TO
19:52:04.626 -> OK 0 TO
19:52:04.673 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:52:04.813 -> OK 0 TO
19:52:04.860 -> OK 33 20 A0 23 20 20 20 26 20 20 A0 10 1C 10 50 48 AA E8 2B FF TO
19:52:05.001 -> OK 0 TO
19:52:05.047 -> OK B3 B3 20 BD A0 A4 21 67 78 21 21 B9 TO
19:52:05.235 -> OK 0 TO
19:52:05.235 -> OK 33 20 20 4 4 4 24 2 2 2 2A 1A 0 0 D 3F D 6F B9 TO
19:52:05.422 -> OK 0 TO
19:52:05.422 -> OK

dgatf, 2022-12-08:
That's weird. I can't reproduce. It is receiving data from the esc, but it seems to be garbage. Please flash this one and post the output.

[MSRC-RP2040_hw4_pio.zip](https://github.com/dgatf/msrc/files/10189734/MSRC-RP2040_hw4_pio.zip)

Klippeneck, 2022-12-09:
Hello Daniel,
the new SW does not work. The ESC logs on but sends no data, stable zero.
Vario no longer logs on, also sends no data, likewise Voltage.
Attached is a debug file. 
13:29:09.314 -> 8A 82 A2 82 AA CA 9A 
13:29:13.516 -> 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
13:29:13.516 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:13.751 -> 0 1 0 0 0 32 0 0 0 5 BE 0 0 D C6 D D8 B9 
13:29:13.938 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:14.126 -> 0 2 0 0 0 32 0 0 0 C1 0 0 D C7 D D9 B9 
13:29:14.313 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:14.547 -> 0 3 0 0 0 32 0 0 0 5 C3 0 0 D CE D D9 B9 
13:29:14.735 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:14.969 -> 0 4 0 0 0 32 0 0 0 5 BF 0 0 D CC D D8 B9 
13:29:15.110 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:15.344 -> 0 5 0 0 0 32 0 0 0 5 C3 0 0 D C9 D DA B9 
13:29:15.532 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:15.766 -> 0 6 0 0 0 32 0 0 0 5 BE 0 0 D CB D D9 B9 
13:29:15.953 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:16.141 -> 0 7 0 0 0 32 0 0 0 5 C2 0 0 D D0 D D9 B9 
13:29:16.328 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:16.563 -> 0 8 0 0 0 32 0 0 0 5 BE 0 0 D C8 D DA B9 
13:29:16.750 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:16.937 -> 0 9 0 0 0 32 0 0 0 5 BF 0 0 D C9 D D8 B9 
13:29:17.125 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:17.359 -> 0 A 0 0 0 32 0 0 0 5 C3 0 0 D CA D D7 B9 
13:29:17.547 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:17.734 -> 0 B 0 0 0 32 0 0 0 5 C2 0 0 D D5 D DA B9 
13:29:17.922 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:18.156 -> 0 C 0 0 0 32 0 0 0 5 C6 0 0 D C4 D D9 B9 
13:29:18.343 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:18.531 -> 0 D 0 0 0 32 0 0 0 5 BE 0 0 D C2 D D8 B9 
13:29:18.718 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:18.953 -> 0 E 0 0 0 32 0 0 0 5 C2 0 0 D D0 D D8 B9 
13:29:19.140 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:19.328 -> 0 F 0 0 0 32 0 0 0 5 C2 0 0 D CE D D8 B9 
13:29:19.515 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:19.749 -> 0 10 0 0 0 32 0 0 0 5 BE 0 0 D C9 D D8 B9 
13:29:19.937 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:20.124 -> 0 11 0 0 0 32 0 0 0 5 C3 0 0 D C7 D D8 B9 
13:29:20.327 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:20.530 -> 0 12 0 0 0 32 0 0 0 5 C3 0 0 D C8 D D8 B9 
13:29:20.765 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:20.952 -> 0 13 0 0 0 32 0 0 0 5 C3 0 0 D CA D D8 B9 
13:29:21.140 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:21.327 -> 0 14 0 0 0 32 0 0 0 5 C4 0 0 D C9 D D8 B9 
13:29:21.561 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:21.749 -> 0 15 0 0 0 32 0 0 0 5 C3 0 0 D C7 D D8 B9 
13:29:21.936 -> B3 B3 21 FF 21 67 78 21 21 B9 
13:29:22.124 -> 0 16 0 0 0 32 0 0 0 5 BF 0 0 D C8 D D7 B9 
13:29:22.358 -> B3 B3 21 FF 21 67 78 21 21 B9

dgatf, 2022-12-09:
At least that is no garbage, though different frames than expected

Please flash following firmwares and post output:

[MSRC-RP2040_hw4_uart1.zip](https://github.com/dgatf/msrc/files/10196894/MSRC-RP2040_hw4_uart1.zip)
[MSRC-RP2040_hw4_pio.zip](https://github.com/dgatf/msrc/files/10196896/MSRC-RP2040_hw4_pio.zip)

Only hw4 is enabled. Config can't be changed. At least sensors should be registered in jeti

Klippeneck, 2022-12-09:
Hello Daniel,
here are the debug files, the ESC has logged into the transmitter for both SW. 
Bei SW .uart
19:24:54.591 -> 8A 82 A2 82 AA CA 9A 6 0 
19:24:59.550 -> 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
19:24:59.596 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
19:24:59.784 -> 33 20 A0 20 20 20 20 26 20 20 A0 D0 1C 10 50 C8 6B 9A B9 0 
19:24:59.971 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
19:25:00.206 -> 33 20 20 20 20 20 20 26 20 20 A0 90 3 2 6A 2 D D3 B9 0 
19:25:00.393 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
19:25:00.627 -> 33 20 A0 20 20 20 20 26 20 20 A0 D0 1C 10 50 8 AE 48 97 7 0 
19:25:00.815 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
19:25:01.002 -> 33 20 20 10 10 10 90 13 10 10 50 50 3 2 6A 12 D D2 B9 0 
19:25:01.190 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
19:25:01.424 -> 33 20 A0 10 10 10 90 13 10 10 50 D0 3 2 6A A D D3 B9 0 
19:25:01.612 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
19:25:01.799 -> 33 20 20 10 10 10 90 13 10 10 50 50 1C 10 50 68 6B 9A B9 0 
19:25:01.986 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
19:25:02.221 -> 33 20 A0 10 10 10 90 13 10 10 50 D0 1C 10 50 28 AE 48 97 7 0 
19:25:02.408 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
19:25:02.596 -> 33 20 20 21 20 20 20 26 20 20 A0 D0 1C 10 50 28 AE 48 97 7 0 
19:25:02.783 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
19:25:03.017 -> 33 20 A0 21 20 20 20 26 20 20 A0 D0 1C 10 50 8 AE 48 97 7 0 
19:25:03.205 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
19:25:03.392 -> 33 20 20 21 20 20 20 26 20 20 A0 90 3 2 6A 2 D D3 B9 0 
19:25:03.580 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
19:25:03.814 -> 33 20 A0 21 20 20 20 26 20 20 A0 D0 1C 10 50 48 6B 92 B9 0 
19:25:04.000 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
19:25:04.228 -> 33 20 20 8 8 8 48 2 2 2 2A 12 0 0 D C1 D D2 B9 0 
19:25:04.411 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
19:25:04.602 -> 33 20 A0 8 8 8 48 2 2 2 2A 1A 0 0 D BE D D2 B9 0 
19:25:04.825 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 

Bei SW pio
19:29:05.741 -> 8A 82 A2 82 AA CA 9A 
19:29:11.043 -> 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
19:29:11.137 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:11.344 -> 0 1 0 0 0 32 0 0 0 5 C6 0 0 D B8 D CF B9 
19:29:11.532 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:11.719 -> 0 2 0 0 0 32 0 0 0 5 C4 0 0 D BD D CE B9 
19:29:11.948 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:12.135 -> 0 3 0 0 0 32 0 0 0 5 BE 0 0 D B8 D CF B9 
19:29:12.323 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:12.549 -> 0 4 0 0 0 32 0 0 0 5 BE 0 0 D B9 D CE B9 
19:29:12.736 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:12.925 -> 0 5 0 0 0 32 0 0 0 5 C3 0 0 D BC D CE B9 
19:29:13.160 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:13.347 -> 0 6 0 0 0 32 0 0 0 5 C0 0 0 D B7 D CF B9 
19:29:13.529 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:13.763 -> 0 7 0 0 0 32 0 0 0 5 C3 0 0 D BA D CE B9 
19:29:13.945 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:14.125 -> 0 8 0 0 0 32 0 0 0 5 C3 0 0 D B6 D CE B9 
19:29:14.359 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:14.548 -> 0 9 0 0 0 32 0 0 0 5 C3 0 0 D BA D CD B9 
19:29:14.736 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:14.923 -> 0 A 0 0 0 32 0 0 0 5 C3 0 0 D B6 D CF B9 
19:29:15.150 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:15.338 -> 0 B 0 0 0 32 0 0 0 5 C3 0 0 D BB D CE B9 
19:29:15.527 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:15.754 -> 0 C 0 0 0 32 0 0 0 5 BE 0 0 D BA D CE B9 
19:29:15.941 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:16.131 -> 0 D 0 0 0 32 0 0 0 5 C2 0 0 D BA D CD B9 
19:29:16.365 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:16.553 -> 0 E 0 0 0 32 0 0 0 5 C3 0 0 D BA D CD B9 
19:29:16.732 -> B3 B3 21 FF 21 67 78 21 21 B9 
19:29:16.919 -> 0 F 0 0 0 32 0 0 0 5 C3 0 0 D BA D CE B9 
19:29:17.155 -> B3 B3 21 FF 21 67 78 21 21 B9

dgatf, 2022-12-09:
There are two concurrent issues:

### Issue 1

Related to uart1 not being at proper baudrate. I can't reproduce it. What RP2040 board are you using?

Hopefully this is fixed with this fw:

[MSRC-RP2040_hw4_uart1.zip](https://github.com/dgatf/msrc/files/10197953/MSRC-RP2040_hw4_uart1.zip)

If not the correct values in telemetry, please post the output.

### Issue 2

Different frames than usual from the esc. I've adapted the hw4 driver to read those frames. Note this firmware doesn't use uart1 but pio for serial

[MSRC-RP2040_hw4_pio.zip](https://github.com/dgatf/msrc/files/10197960/MSRC-RP2040_hw4_pio.zip)

With this fw you should see the correct values. Otherwise please post output

Klippeneck, 2022-12-10:
Hello Daniel,
the board I use has the designation RP2040-Zero from Waveshare.
With the SW uart no changes from previous versions.
With the SW pio updates the voltage in the telemetry display of the transmitter, as well as the two BEC temperatures. That the other ESC values ( current and RPM) do not update is probably due to the 25% threshold for the current ( is set to 250A). Probably the two SWs are special
Versions and can not be configured.
Log at pio:
09:48:17.564 -> 8A 82 A2 82 AA CA 9A 
09:48:21.782 -> 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
09:48:21.969 -> B3 B3 21 FF 21 67 78 21 21 B9 
09:48:22.203 -> 0 1 0 0 0 32 0 0 0 5 BD 0 0 D B4 D C5 B9 
09:48:22.391 -> B3 B3 21 FF 21 67 78 21 21 B9 
09:48:22.578 -> 0 2 0 0 0 32 0 0 0 5 C3 0 0 D B9 D C5 B9 
09:48:22.766 -> B3 B3 21 FF 21 67 78 21 21 B9 
09:48:23.000 -> 0 3 0 0 0 32 0 0 0 5 C2 0 0 D B8 D C3 B9 
09:48:23.188 -> B3 B3 21 FF 21 67 78 21 21 B9 
09:48:23.375 -> 0 4 0 0 0 32 0 0 0 5 C3 0 0 D B8 D C5 B9 
09:48:23.609 -> B3 B3 21 FF 21 67 78 21 21 B9 
09:48:23.797 -> 0 5 0 0 0 32 0 0 0 5 C0 0 0 D B2 D C5 B9 
09:48:23.984 -> B3 B3 21 FF 21 67 78 21 21 B9 
09:48:24.219 -> 0 6 0 0 0 32 0 0 0 5 C3 0 0 D 86 D C5 B9 
09:48:24.406 -> B3 B3 21 FF 21 67 78 21 21 B9 
09:48:24.609 -> 0 7 0 0 0 32 0 0 0 5 C2 0 0 D B1 D C6 B9 
09:48:24.796 -> B3 B3 21 FF 21 67 78 21 21 B9 
09:48:24.984 -> 0 8 0 0 0 32 0 0 0 5 C3 0 0 D B7 D C5 B9 
09:48:25.171 -> B3 B3 21 FF 21 67 78 21 21 B9 
09:48:25.406 -> 0 9 0 0 0 32 0 0 0 5 C3 0 0 D B1 D C5 B9 
09:48:25.593 -> B3 B3 21 FF 21 67 78 21 21 B9 
09:48:25.781 -> 0 A 0 0 0 32 0 0 0 5 C2 0 0 D B6 D C4 B9 
09:48:25.968 -> B3 B3 21 FF 21 67 78 21 21 B9 
09:48:26.202 -> 0 B 0 0 0 32 0 0 0 5 C2 0 0 D B6 D C5 B9 
09:48:26.390 -> B3 B3 21 FF 21 67 78 21 21 B9 
09:48:26.577 -> 0 C 0 0 0 32 0 0 0 5 BD 0 0 D B6 D C4 B9 
09:48:26.812 -> B3 B3 21 FF 21 67 78 21 21 B9
Log at uart
09:33:31.291 -> 8A 82 A2 82 AA CA 9A 2 0 
09:33:35.509 -> 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
09:33:35.697 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
09:33:35.884 -> 33 20 A0 20 20 20 20 26 20 20 A0 D0 1C 10 50 28 35 81 B9 0 
09:33:36.071 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
09:33:36.306 -> 33 20 20 20 20 20 20 26 20 20 A0 D0 1C 10 50 88 35 81 B9 0 
09:33:36.493 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
09:33:36.681 -> 33 20 A0 20 20 20 20 26 20 20 A0 90 1C 10 50 A8 35 7D B9 0 
09:33:36.868 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
09:33:37.103 -> 33 20 20 10 10 10 90 13 10 10 50 90 3 2 6A 9A 35 81 B9 0 
09:33:37.290 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
09:33:37.477 -> 33 20 A0 10 10 10 90 13 10 10 50 D0 1C 10 50 88 35 7D B9 0 
09:33:37.712 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
09:33:37.899 -> 33 20 20 10 10 10 90 13 10 10 50 90 1C 10 50 68 35 81 B9 0 
09:33:38.087 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
09:33:38.321 -> 33 20 A0 10 10 10 90 13 10 10 50 90 3 2 6A A2 35 7D B9 0 
09:33:38.508 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
09:33:38.696 -> 33 20 20 21 20 20 20 26 20 20 A0 D0 1C 10 50 C8 AB E8 96 F 0 
09:33:38.883 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
09:33:39.118 -> 33 20 A0 21 20 20 20 26 20 20 A0 D0 1C 10 50 68 35 79 B9 0 
09:33:39.305 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
09:33:39.493 -> 33 20 20 21 20 20 20 26 20 20 A0 90 3 2 6A A2 35 7D B9 0 
09:33:39.727 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
09:33:39.914 -> 33 20 A0 21 20 20 20 26 20 20 A0 90 3 2 6A 72 35 7D B9 0 
09:33:40.102 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
09:33:40.289 -> 33 20 20 8 8 8 48 2 2 2 2A 1A 0 0 D 73 D 9E B9 0 
09:33:40.524 -> B3 B3 20 BD A0 A4 21 67 78 21 21 B9 0 
09:33:40.711 -> 33 20 A0 8 8 8

dgatf, 2022-12-10:
> That the other ESC values ( current and RPM) do not update is probably due to the 25% threshold for the current ( is set to 250A).

Current in telemetry is be 0 until 25% throttle, but I'll see the actual value in the output. RPMs should show anytime. Maybe packets are different when throttle is up. Please post the output of the pio firmware after throttle is up.

Correct, this firmware config can't be changed

I'm almost sure uart is ok, so I suspect the esc it is sending non standard serial, which somehow does not affect the pio serial.

dgatf, 2022-12-10:
>  The values for voltage and RPM are not yet correct, will be due to the missing configuration.

Yes

> Compared to the last attempts, temperature BEC is missing.

Is it 0 or not registered?

Please confirm if the following sensors are registered:

- RPM
- Current
- Voltage
- Temp FET
- Temp BEC
- Cell Voltage
- Consumption

Before adding this changes to the main branch I need to confirm whether this is an issue with rp2040 uart1 or the esc is sending different types of frames

Could you test the esc with any arduino board with the following scketch. Connect esc signal to rx and esc gnd to gnd an dpost output

[hw4_read.zip](https://github.com/dgatf/msrc/files/10200887/hw4_read.zip)

Alternatively you can test this esc with arduino version of msrc and see if it works

Klippeneck, 2022-12-11:
Hello Daniel,
always registered are speed, current, voltage, temperature FET, voltage cell, consumption.
In 5 out of 10 tests, temperature BEC is also registered.
All registered values are updating.
When trying with the Arduino msrc exactly the same, temperature BEC is sometimes missing, both. 
in the registration and then also as telemetry value.
So far I could not see any systematic when temperature BEC is missing.
Will do more systematic testing tomorrow and report back.

Klippeneck, 2022-12-12:
Hello Daniel,
in today's tests all ESC values both when logging in to the transmitter and. 
with the current sensor values i. O., also temperature BEC always comes.
In the tests with Arduino msrc temperature BEC is missing in 9 out of 10 cases.
In addition, the current values are correct only the first time after flashing the Arduino msrc.
With each subsequent startup, the current shows only about half values.
This but only on the side, when the RP 204 msrc is ready, I will replace the Arduino msrc.
Probably I have installed some error.

dgatf, 2023-07-18:
This is may be fixed with ce9b22b8d6700a00b059e56347e5634fa09ccdcc

closed 2023-07-18 by dgatf