dgatf/msrc #94: RP2040 status LED is not used

NilsRo, 2023-01-08:
It seems that the status led does not work anymore like it is noted in the README. 

"Status led of the board blinks when sending telemetry. If it doesn't blink check connections and config"

Using the provided prebuild firmware with S.Port and Hobbwing V4 ESC.

dgatf, 2023-07-28:
> Looking forward to a version with RGB indicator lights.

Fixed with 1a3f299f7e3c4cdfac2c9b15cf074ee6aff59de2

closed 2023-07-28 by dgatf

dgatf, 2023-07-29:
> MSRC is more suitable for my application, using jtei's remote control. Currently, I have found that:
> 
>     GPS positioning is relatively slow, sometimes longitude and latitude only appear for a long time on JETI, and sometimes when the TX plug is unplugged and plugged in again, data such as longitude and latitude and date will appear;
> 
>     The voltage display data jumps, not at a fixed value;
> 
>     Unable to display the current value, I am not sure if it is the reason for the setting;

I've created a new issue for this. Please see #115