dgatf/msrc #103: Jeti EX Bus GPS not working

andypnz, 2023-02-13:
Hi. Great project!!

Using a RPi PICO (2040). Matek M10-5883 GPS (9600, NMEA).

'Voltage' telem works (pin not connected - floating - but I get numbers)

GPS enabled, 9600. Unfortunately no data on the transmitter (all zeros)

Jeti Ex Bus logic analyzer capture attached if of any use? Really keen to try and get this going - happy to try any suggestions / firmware / do more captures, etc :-)

[Jeti EX Bus telem stream with GPS sensor + voltage floating (2).zip](https://github.com/dgatf/msrc/files/10719453/Jeti.EX.Bus.telem.stream.with.GPS.sensor.%2B.voltage.floating.2.zip)

andypnz, 2023-02-18:
Ok, managed to sort that - the code was not handling southern hemisphere -ve latitude (or -ve long either)......

dgatf, 2023-07-19:
> Ok, managed to sort that - the code was not handling southern hemisphere -ve latitude (or -ve long either)......
> 
> If you like I can share with you the changes I made to fix this, and also improve the displayed gps resolution....

Yes, please share. Thank you

Is it with the RP2040?

andypnz, 2023-07-19:
Its been a while since I looked at this :-)
I have attached the project directory - but from memory it was just changes to these two files:

jeti.c, lines 281-284 change to:
```
                uint8_t degrees = value / 60000;
                uint32_t degreesDecimals = ((value % 60000)); 
                *(buffer + *buffer_index + 1) = (uint16_t)degreesDecimals;      // degrees (dec, l)
                *(buffer + *buffer_index + 2) = (uint16_t)degreesDecimals >> 8; // degrees (dec, h)
```
nmea.c, lines 23-24:
```
    *parameter.lat = 692761.0;               //-692.761166667;  // 11º32'45.67" +N, -S
    *parameter.lon = 10320.0;               //-2500; //-1251.964833333; // 20º51'57.89" +E, -W
```
line 131:
```
            *parameter->lat = (lat_dir * (atof(degrees) * 60 + minutes)) * 1000;
```
line 139:
```
            *parameter->lon = (lon_dir * (atof(degrees) * 60 + minutes)) * 1000;
```

[project.zip](https://github.com/dgatf/msrc/files/12100395/project.zip)

dgatf, 2023-07-19:
Ok. Negative coordinates where not correct. I'll add `value *= -1;` to fix it
I guess multiplier 1000 is to increase accuracy. I may add it later

But based on the debug, the issue was related to gps module not being read by rp2040

dgatf, 2023-07-19:
Ok, I see. Negative coordinates where not correct. I'll add value *= -1; to fix it. Commit https://github.com/dgatf/msrc/commit/6f5e9d93f3eb206e419a58ffc1ece5e114bbde5d
I guess multiplier 1000 is to increase accuracy. I may add it later

But based on the debug, the issue was related to gps module not being read by rp2040 and that's why all were zeros in the transmitter

dgatf, 2023-07-20:
Ok, I see. Negative coordinates where not correct. I'll add `value *= -1;` to fix it. Commit 6f5e9d9
I guess multiplier 1000 is to increase accuracy. I may add it later

But based on the debug, the issue was related to gps module not being read by rp2040 and that's why all were zeros in the transmitter

andypnz, 2023-07-20:
I think you really need to add the multiplier - without this, the GPS accuracy is very poor.......

dgatf, 2023-07-20:
I agree that accuracy for coordinates is not the best. I realized that using a float is not enough for coordinates. Float has a mantissa of 7 digits and for 1m precision on coordinates, 9 digits are needed. Now, in worst case, the accuracy is 100m, depending on the degrees value (bigger, less accurate)

But just multiplying the value by 1000 does not increase accuracy, as the limit is the mantissa for float variables. Another variable type is required to hold coordinates accurately. For coordinates I'll change from float to double, which has a mantissa of 16 digits (int32 will work as well).  But this implies creating new sensors for all rx protocols to hold coordinates with best accuracy. Will do this soon.

Thank you.

dgatf, 2023-07-20:
The issue was that the formatting for coordinates for jetiex was wrong. No need to add multipliers. Fixed with 8bf664fb0cb1fd51307aba826ccfbee37594e224

As a side note:

Jeti uses 3 decimals on minutes. This is 2m accuracy. Float has a mantissa of 7 digits, which is ok in most cases for 3 decimals on minutes, except for longitudes > 160°, where accuracy drops to 20m

NMEA protocol sends 5 decimals on minutes, which is 0.2m accuracy (usual gps modules don't have this accuracy though). This requires 10 digits matissa. A double (16 digits mantissa) instead a float is required to hold all digits. This implies to create sensors for all rx protocols that holds a double instead a float. I may do this as an enhacement

closed 2023-07-21 by dgatf