dgatf/msrc #105: Futaba SBUS - BMP280 variometer bug

palomba01, 2023-03-03:
I'm trying to use the legacy arduino code with a BMP280 for altitude/vertical speed, on Frsky works great but on Futaba SBUS the vertical speed oscillates too much even when still and flying the values have the correct feeling but with exaggerated values, +30m/s on a real +1m/s thermal. 
[image]

https://user-images.githubusercontent.com/83312049/222806018-8036d642-38c1-4909-a8cb-c6040b2c26dc.mp4

zxd00276, 2023-03-10:
I am using MSRC (Regacy) on my FUTABA SBUS system. In my case, there is no abnormal change in data with BMP280.
Please refer to the attached video. Also, how about changing the value of averaging in CONFIG.H of MSRC? It may be oversampling in BMP280.

https://user-images.githubusercontent.com/102213916/224272321-0ab7eced-8318-4de5-b3be-6a34f10e276c.mp4

zxd00276, 2023-03-10:
In the previous report, I used FUTABA's R3001SB (T-FHSS), but I tested it with your FASSTest receiver (R7106SB), and there was no abnormal data fluctuation. Why don't you check if there is no temperature change etc. with Arduino pro mini?
Please refer to the video below.

https://user-images.githubusercontent.com/102213916/224447917-dd10fee1-20a5-45fa-8b74-def9f0a44011.mp4

zxd00276, 2023-03-21:
Dear, I am attaching the requested config.h.
I think it's data from your FUTABA T14SG's TELEMETRY
Are you updating to the latest version of the software? Why don't you check if there is no problem with T14SG once?
[config.zip] (https://github.com/dgatf/msrc/files/11024686/config.zip)

closed 2023-07-18 by dgatf