dgatf/msrc #120: having a similar problem to closed issue XBUS-Connection to AR8020T-Receiver #95

BatesKG, 2023-11-07:
Hello.  I'm trying to use this project with an analog speed sensor (FuriousFPV), an RP2040 Zero board, and an AR8020T receiver that is bound to an ix20.   In the BoardConfig gui, I can see changing values getting from the sensor to the RP2040 in the debug screen, but the IX20 is not seeing the airspeed sensor.  Are the code changes in issue #95 now part of the downloadable binary from the main page?  Or the binary that is 1K smaller from the issue 95 page?   I've tried both binaries and neither seem to work.
Thanks for any info you can provide.

KevinB

dgatf, 2024-02-23:
Check that pullups are installed (I've tested with 1k)
I've tested and it is working. It might be the binary was not the last one. I've added some additional debug output. You should see something like this:
[image]

Please, download and check the debug and post it

dgatf, 2024-03-08:
Faster boot time to improve xbus initialization. Commit a1bfc8d

closed 2024-03-08 by dgatf