dgatf/msrc #124: Wiring Diagramm hobbywing Frsky

picposter, 2024-04-26:
Hello,
I am not sure where to post it, and I didn't find it on the page. Connecting pin 0 to the frsky receiver, but where to connect the platinum ESC?
Do I have to connect it on the rp2040? 

I am sorry, I am confused but I like to have this in my setup.

Kind regards
Cedric

picposter, 2024-04-26:
Hello again,
found what i searched for in the Windows Program. A hint in the documentation would be great.

dgatf, 2024-04-26:
First, you should specify which esc model and protocol are you using

There is a picture of the circuit for every configuration in msrc_gui:

[image]

 And also in the readme table:

[image]

And you can see a description of esc types in section [6.1](https://github.com/dgatf/msrc?tab=readme-ov-file#61-esc)

Also, I'm not sure what do you mean with:

> found what i searched for in the Windows Program

And please, could you propose and explain an specific improvement of the documentation and/or musr_gui app , so other users can take advantage of your suggestion, rather than:

> A hint in the documentation would be great

closed 2024-04-28 by dgatf