dgatf/msrc #125: Schematics hard to see (too small) in msrc-gui

ignaciobfp, 2024-05-04:
Maybe I am missing something, but with my current screen/resolution I am unable to identify pins and resistance values in the MSRC-gui "circuit" tab. Opening in a new window does not improve the situation for me.

This can be solved by either allowing zoom (if there is a way, I could not figure it out), or allowing the user to save the image to zoom using an image viewer afterwards.

Also, it would be very welcome to check the circuit without the need of having a board connected, to be able to solder wiring in advance.

tontonlex, 2024-06-01:
I have the same problem.
I think it shoud be a good thing if button "open to new windows" open in full screen.
Actually we need decrypt resistor values and pinout.

dgatf, 2024-08-21:
Added zoom for circuit image with commit 3a0b64a

closed 2024-08-21 by dgatf