dgatf/msrc #126: Any intentions on adding Jeti EX ESC telemetry to convert it to S.Port

jasc76, 2024-05-13:
Hi

any plans to include Jeti Mezon EX telemetry ??

Regards
Jan

F-Po, 2024-06-09:
Same question from my side, I would really like to have it.

dgatf, 2024-06-17:
Jeti Ex telemetry is implemented. It was developed with arduino legacy and ported to rp2040. It should work as is now, but it haven't been tested on rp2040 yet. Please select _Jeti Ex Bus_ in receiver protocol. If any issue please post the debug output.

F-Po, 2024-06-17:
Seems to be a misunderstanding. What's needed is to read out the telemetry values from a Jeti Mezon Speed controller and send it to FrSky SmartPort receiver.

dgatf, 2024-10-07:
The Mezon sends the telemetry in Jeti Ex format which is implemented for the receiver. In Ex bus, the telemetry is requested from the receiver and it also sends the channels data.

I need to confirm this. For that I've added a serial monitor to msrc. Connect esc to GPIO5 as usual and select *Receiver protocol->Serial Monitor* with the following parameters:

- Baudrate: 115200
- Stop bits: 1
- Parity: None
- Timeout: 1 ms
- Inverted: Unchecked

dgatf, 2024-10-16:
Does Mezon lite 160 output rpms in pwm?

dgatf, 2024-10-16:
Jetiex for mezon is not implemented. Though you can select serial monitor in receiver protocol at 115200 and post output. Conbect elemetry line of the esc to gpio5 while connected to the receiver.

F-Po, 2024-10-16:
I don't know what the Mezon lite 160 output in telemetry. This log comes also if no esc is connected, when pwm is selected. So this was a wrong try by myself.

Where can I select the serial monitor and the baud rate? In msrc_gui I don't see this option and you said the lua doesn't work.

[image]

F-Po, 2024-10-16:
Please tell me what exactly I have to connect where and where and how I can get the serial log. I'm really baffled at the moment.

F-Po, 2024-10-17:
Got it. I had overseen that the serial monitor is selectable in the receiver protocol - you mentioned it - my fault. In the end the key was to connect the esc telemetry gnd to the RP2040 gnd. Here is your log: [Mezon 160 lite 1st log.txt](https://github.com/user-attachments/files/17415620/Mezon.160.lite.1st.log.txt) Please tell me if you need more!

dgatf, 2024-10-17:
The output doesn't look good. Almost all zeroes. To be sure, to reverse engineer the mezon, it is needed a jeti receiver. Which receiver are you using?

F-Po, 2024-10-17:
I am using a FrSky RX6R and I do not have a jeti receiver. But I can borrow a jeti system including different sensors from a friend next week. So we have to wait. I will give you feedback if I have more information.

dgatf, 2024-10-17:
Were you connecting the telemetry of the esc to the smartport? If not using jeti recevier, it must not be conected. If you did, that might be the reason it didn't. You can try connecting esc telemetry line only to gpio5 (not to smartport also) and gnd to gnd.

F-Po, 2024-10-17:
I did not connect the sport. The only connection to the RP2040 is the usb and signal and gnd to the telemetry port of the esc.

F-Po, 2024-10-17:
...but good question: If I'm using jeti, I must connect the esc telemetry to both the RP2040 **and** the receiver telemetry port?

dgatf, 2024-10-17:
When using jeti receiver, yes, you have to, because the telemetry is polled from the receiver.

F-Po, 2024-10-17:
Okay, I will do so.

Here a picture of my FrSky setup:
[image]

dgatf, 2024-10-17:
Ok

F-Po, 2024-10-21:
Here the first two logs from the jeti communication. The MULI is a 6S LiPo sensor which had 6 cells connected while logging.

[log 2 - jeti MULI 6S.log](https://github.com/user-attachments/files/17466547/log.2.-.jeti.MULI.6S.log)
[log 3 - jeti Mezon 160 lite.log](https://github.com/user-attachments/files/17466548/log.3.-.jeti.Mezon.160.lite.log)

dgatf, 2024-10-24:
I've made the baudrate editable, so you can set any baudrate. Download updated msrc_gui.