dgatf/msrc #127: Hobbywing Platinum V4 120A - RPM

Timo-72, 2024-06-05:
As with the Teensy I can see only RPM values like 100, 200, 300, 400, 1800, 1900 etc. 

I had main gear: 106
Pinion gear: 12

I changed Pinion gear to 120 and in my transmitter I changed resolution from 0,.. to 0.0

And now it show 1810.0 , 1920.0 , 2020.0 => so now it is OK

Timo-72, 2024-06-05:
I have 2040 board

dgatf, 2024-06-17:
ADC resolution is 4096. If adding main and pinion gear in transmitter, change sensor precision to maximum, 0.00 (3.30 will be the maximum, so you have 330 steps of resolution for rpm). Though it is recommended to add main and pinion gear in msrc, to use the full resolution for rpms.

closed 2024-06-17 by dgatf

dgatf, 2024-06-17:
Please add your receiver protocol and rpm sensor. I guess is smartport with esc rpms. Esc rpms in smartport has 100 rpm resolution, despite the sensor precision. An option could be added to select rpm (not esc rpms) as the rpm sensor, which has 1 rpm resolution. See [esc rpm](https://github.com/dgatf/msrc/blob/cf730bc45061ff50bea82d417e31007ac940b755/board/project/smartport.h#L84C9-L84C30) and [rpm](https://github.com/dgatf/msrc/blob/cf730bc45061ff50bea82d417e31007ac940b755/board/project/smartport.h#L46C3-L46C4)