dgatf/msrc #128: Analog current sensor offset issue

tontonlex, 2024-06-15:
Hello,
I try tu use msrc with a analog current sensor (ACS758LCB-100B)
This one is bidirectionnal, so this sensor 0A Vout is about half VCC input.

Auto offset and manual offset seem not work.
With 22mV/A and 1.7 manual offset, I get about 74A on debug and on my futaba transmitter.
So about the same when offset is set to 0.

I see there was a fix 14th april 2024 (https://github.com/dgatf/msrc/commit/0897e295cf3523399ecce1c198035c669d1af7fe) so I try build firmware, but the problem keep the same.

Does anyone have a solution to this problem ?

derFliegendeHamburger, 2024-06-15:
I have an old arduino based version modified to work with zero midpoint for one analog sensor. 
Cannot help with modern mcu

dgatf, 2024-06-16:
There were some issues with current and auto offset. Auto offset and manual offset should be ok now.

Fixed with 2158586

tontonlex, 2024-06-16:
Auto offset and manual offset works as expected now, but in debug mode only.
Auto Offset voltage is well mesured, I have theses values in debug :
> Current (98): Curr -0.18 Cons 98.65 Volt 1.65 VoltOffs 1.66 Multip 45.00

But on SBUS2 value is still wrong, I still have 74A.

dgatf, 2024-06-16:
Current is zero until auto offset (if auto offset is enabled). Commit 2158586

[Binary](https://drive.google.com/file/d/1pzIt4XdcLxiSUfSTfyoh_w5tLTFd9JN2/view?usp=drive_link) updated

closed 2024-06-17 by dgatf