dgatf/msrc #129: Hobbywing Platinum 150A V5

Timo-72, 2024-06-30:
There is an older issue showing problems with Hobbywing Platinum 150A V5.

What I use:
Frsky X-XSR Receiver = using smartport 
Hobbywing Platinum 150A V5

Available Firmware for Hobbying: 4.08 , 4.10 , 4.1.0 , 4.1.2
No information from Hobbywing on the different firmwares.
VBAR Forum says 4.1.0 and 4.1.2 is VBAR ready.
I tried all 4 diferemt firmwares

I tried in Msrc_gui exe Hobbywing 120A, 150A Flyfun 130A, 150A.
I have latest OpenTX 2.3.15
I can find senspor, but all sensors show 0, means 0V, 0 Temperature, 0 RPM

On the same hardware installation (Helicopter Nimbus 550) I used Hobbywing 120A V4 and Telemetry was OK.
So receiver, cables etc. are OK. 
- Issue with my V5 Platinum ESC? (defect?) 
- Or something wrong with this program?

dgatf, 2024-06-30:
It is may be a different protocol. V3 and V4 are different protocols. V5 may be different from V4 protocol.

You can try to read your esc with the attached sketch (you may need to change the baudrate) and an arduino uno (connect esc data to rx and gnd to gnd) 

[serial_read.zip](https://github.com/user-attachments/files/16044960/serial_read.zip)

Timo-72, 2024-06-30:
Thanks for the feedback, I will check if I can / want to support your offer within the next days.
I spend hours on the internet and looking for other options.

I found at flybarless system Microbeast that flybars need to be adjusted different for Hobbywing Platinum V4 and V5.
On this page you can find:
https://wiki.beastx.com/index.php?title=Manuals:MBPlusFblV52:Receiver_M/de#Hobbywing.C2.AE_Platinum_.28V4.2FV5.29

blinkt violett => Hobbywing® Platinum V4
violett => Hobbywing® Platinum V5 / OMP 65A ESC
blinkt rot => Kontronik® JIVE/PowerJIVE/HeliJIVE
rot => Kontronik® JIVE Pro/Kosmik
blinkt blau => YGE® Texy
blau => Scorpion® Tribunus ("UnSc" mode)

So it looks like it is different!

Timo-72, 2024-07-01:
Grüzi Daniel,
I have some bad and some good news.
I send an email to Hobbywing and this is the answer:
**The protocols of Platinum V4 and V5 are different.**
Hobbywing says: "You can read realdata from SBUS 2, but you will need a S.BUS2 Telemetry Adapter for Futaba module. Which you can find in our Website.
So can this also be read with arduino uno?

And attached protocol information from Hobbywing for V5 ESC:
[Platinum V5 protocol.pdf](https://github.com/user-attachments/files/16048590/Platinum.V5.protocol.pdf)

Is this good enough?
I see also good change to get more detailed information from OMP 65A ESC, if this helps and if this controller really
has the same protocol.

Let me know how to continue.
Thx Timo

dgatf, 2024-07-01:
That's good news.

I've created a preliminary version to support HW V5 protocol. Please download updated [binary](https://drive.google.com/file/d/1kOk22Qd9oZjJUaG1-pNzfVW4My3XwyOM/view?usp=sharing) for hw5 protocol.

In msrc_gui select hw4 (any model is ok). Check if you have telemetry in your transmitter. Otherwise enable debug and post output.

I don't know if this v5 protocol is firmware version dependent. You may need to try with different esc firmware version (vbar version or not vbar version ?)

Timo-72, 2024-07-01:
Great news also from your side and thanks for the binary.
And I can tell you your new binary is showing telemtry. Not the correct values, but it shows telemtry.
What is missing, old Hobbywing showed Temperature for BEC and ESC. Now only one Temperature.
But no issue if this cannot be fixed.

As said, Hobbywing PC tool does NOT show any information on the firmware. 
So it does not show with or without Vbar. I had only made assumptions from other pages on the internet.

I checked now with Firmware 4.0.8 (oldest available) and 4.1.2 (newest available).
Both show same data!
I choose HW V4 150A from MSRC_Gui.
I did not change values in MSRC_Gui, only gear information.

Attached screenshot what I can see:

[image]

Do you need now more information?

THX so far

dgatf, 2024-07-01:
I've added support for vbar protocol with [commit](25e7ab022c16d479cfad90c3d11676f202b2e546). Please download updated binaries: [msrc](https://drive.google.com/file/d/1pzIt4XdcLxiSUfSTfyoh_w5tLTFd9JN2/view?usp=drive_link) and msrc_gui for [linux](https://drive.google.com/file/d/1G5A5ucQUb3tXk_KQKVdSJNhs5huyyZiU/view?usp=drive_link) or [windows](https://drive.google.com/file/d/1WogFyMToZI1BenxwAIU2i_oVkPFHTB7D/view?usp=drive_link)

Timo-72, 2024-07-02:
Thx, I checked the new binary file with the new windows MSRC_gui
It shows now more sensors, but most of them are not active. No issue, I can delete them manually.

[image]

I set in MSRC_Gui 4 pair of poles (XNOVA Motor has 8 poles / 4 pairs) 
My nimbus 550 has 106 gears to 11 on the motor.

[image]

Result is what you can see.
RPM should be ca. 1120RPM
6,7A should be something between 20A and 40A (running a XNOVA 4020 1200KV at 40%, continous max 105A, short max 130A)
The EscC / currency counting should showing much more, if 30A at 40% ESC load should mean ca. 0,5A per Minute.
What you see on the screenshot is ca. 30sec to 60sec. So we should see ca. 0,2 to 0,5 Ah

I do not know if this works for all VBAR. This is the page where I had my knowledge from.
https://www.vstabi.info/forum/read.php?57,201333,201359
Maybe Vbar / Vstabi was adjustet for this new Hobbywing protocol.
Someone should maybe try that, installing VBAR version for example on a Hobbywing V4 with VBAR version.

dgatf, 2024-07-02:
>It shows now more sensors, but most of them are not active. 
Yes, there are four sensors not used in this esc: bec temperature, motor temperature, bec voltage and bec current. I leave them in case other esc use them as it is defined in the protocol specification

Temperature and voltage values seems to be ok.
Consumption is calculated by msrc with the current value, so we leave it for now.
The issue is with rpms and current. How did you measure rpms and current?

Could you try the HW 120A with a vbar firmware?

Timo-72, 2024-07-02:
RPM
I had the Hobbywing V4 120A before in this helicopter. RPMs where shown correctly at the time.
I know that 50% open ESC was ca. 1200RPM. In addition the hobbywing program box ( https://www.hobbywing.com/en/products/multifunction-lcd-program-box97.html ) shows my the RPM when I setup governor store. And this is simular value that the box is showing.
Today I did this with lowest value possible on the ESC (because in living room) what is 40% ESC open. So round about RPM is linear to how much I "open" ESC.

In parallel calculation is: Battery had ca. 22,5V x 1200KV motor = 27000 RPM on Motor
Gear 106:11 = 9,636
27.000 / 9,636 = 2800 RPM max 
ESC 40% open => 2800 x 0,4 = 1120RPM
For sure you have some loss and inefficency what means many some percentage smaller as 1120 RPM

Currency, I know Motor takes ca. 30A in hover mode. So today was less because Heli was on ground.
But round about not so far away. But I have no good formula if you need that to program. What I know, I have 6S 5000mAh battery. I could fly ca. 7 minutes for a consumption of 4200 to 4400mAh. So if the consumption from msrc shows 4100 and I can load into the battery again 4100 I know it is correct.

If you / we need more factors, than I need more time. I need to compare consumption showing by msrc and what I can load back in battery. And this can be compared.

Hobbywing 120A, I have 1 in the old helicopter with a Teensy. But I do not like to change that. Because with the Teensy everything was more complicated. I have 1 that I took out and replaced by the 150A. If you are really interested I could interim wise conntect it back to the nimbus helicopter, update to VBAR version and check if telemetry is available.

dgatf, 2024-07-02:
Sorry, I forgot to add the rpm multiplier in msrc. Please download updated [binary](https://drive.google.com/file/d/1pzIt4XdcLxiSUfSTfyoh_w5tLTFd9JN2/view?usp=sharing)

Rpms should be ok now.

Timo-72, 2024-07-02:
I checked and RPM is OK now. Shows 1200RPM now at 40% ESC usage. That should be OK.

[image]

Attached a picture of the telemtry. As said. the 6A to 8A are too low.
But there is also a calculation error.

You see usage is 7A for ca. 75 seconds
Used currency shows 150mAh.

Tomorrow I will try 120A Hobbywing V4.
And I will do easy test. I will load a battery to eg 3,9V. Than I will let it run for 2 or 3 minutes (without flying), afterwards I will load battery again and I will check how much consumption was.

dgatf, 2024-07-02:
Consumption calculation seems ok: 8A during 60s is 133mAh

6-8A seems reasonable if not flying,

Timo-72, 2024-07-02:
RPM is correct but accuracy ......
It looks like it rounds to 100, because it is too stable at 1200RPM.
I can try tomorrow play with the parameters. means increase the gear for example by x10 and
on openTX and can decrease by factor 10.

closed 2024-07-02 by Timo-72

dgatf, 2024-07-02:
To increase accuracy, set poles and gears on transmitter.

dgatf, 2024-07-02:
> It looks like it rounds to 100, because it is too stable at 1200RPM.

Yes, it is in steps of 100rpm if poles and gears are set in msrc. This is related to smartport only. The frsky telemetry sensor used for rpms is the esc sensor, which has an accuracy of 100rpm. See [esc sensor](https://github.com/dgatf/msrc/blob/13db69e071518f700cbdc7fb1dc3ba259d0b93b2/board/project/smartport.h#L85) . To increase accuracy, set gear and poles in the transmitter.

I may add an option in msrc to use an alternative telemetry [sensor for rpms](https://github.com/dgatf/msrc/blob/13db69e071518f700cbdc7fb1dc3ba259d0b93b2/board/project/smartport.h#L47) with an accuracy pf 1rpm.

Timo-72, 2024-07-03:
RPM precision, 
you do not need to program anything, see attached Power point / picture that you can use for your manual if you want.

[RPM precision fix.pptx](https://github.com/user-attachments/files/16078827/RPM.precision.fix.pptx)

[image]

dgatf, 2024-07-03:
Great. Thank you.

Timo-72, 2024-07-08:
Hello, when you / we checked current everything was OK. Now I was out for flying and current consumption after 7 minutes is ca. 3500mAh, but it should be ca. 4200 to 4400. I see it because 5000mah was checked and it has ca. 5300mAh. 
Form voltage battery was empty.

Before I changed to Hobbying 150A V5 I used for some weeks Hobbywing 120A V4 in this helicopter. At this time current consumption was shown correcty with 4200 to 4400maH before I had to land because voltage after landing was ca. 3,73 to 3,75V.

Is actual ampere load and current total consumption depending from each other in you calculation? If not, can I just increase the factore in openTX on the current values?

dgatf, 2024-07-08:
Consumption is calculated with the formula mAh = C(A) * t(ms) / 3600 for every current value received.

You can use a ratio (1.2) for consumption in opentx. If it is the V5, means that current values delivered are not very accurate.

Timo-72, 2024-07-11:
I made also some advertizing for your project at www.rc-heli.de
Maybe you can add this adress also to your github page at point 11. support 

One person already made a RP2040 board and it is running

[Telemetrie Hobbywing Platinum V5 - Protokoll Details _ ESC auslesen - RC-Heli Community.pdf](https://github.com/user-attachments/files/16175312/Telemetrie.Hobbywing.Platinum.V5.-.Protokoll.Details._.ESC.auslesen.-.RC-Heli.Community.pdf)

Timo-72, 2024-07-11:
> Consumption is calculated with the formula mAh = C(A) * t(ms) / 3600 for every current value received.
> 
> You can use a ratio (1.2) for consumption in opentx. If it is the V5, means that current values delivered are not very accurate.

1.2 was also logical for me. But at start ratio was 0. When I changed I had to go to a ratio of 300 ! It was crazy for me, and you think why 300? But on the other hand if my problem is fixed, go out flying