dgatf/msrc #130: VBAR version or Hobbywing V5?

Timo-72, 2024-07-03:
I checked now new "VBAR" version of MSRC with Hobbywing V4 120A, Smartport and Frsky X-XSR Receiver

1. I update Hobbywing 120A V4 to latest VBAR Firmware
[image]

2. I deleted all sensors in OpenTX and I searched new sensors, but no telemetry available
[image]

This means you should rename "VBAR" in the dropdown menue of MSRC:GUI to Hobbywing Platimum V5 (Maybe in addition 150A)
If you like, I can ask Hobbywing if all Platimum V5 have same protocol.

Interim wiring to check:
[image]

Kind regards and many thanks !!!

Timo-72, 2024-07-03:
As you know Vbar is the software of the flybarless system from Mikado (flybarless Vstabi).
In the "Spirit" flybarless you have to choose what ESC you use.
I think you do the same in the Mikado flybarless called Vstabi.

And than Mikado has programmed the new Hobbywing V5 telemetry.
The same like you did. But it is not a universal protocol from my understanding

dgatf, 2024-07-03:
I've been looking at the HW V4 spirit [page](https://manual.spirit-system.com/index.php?title=HobbyWing_Telemetry) and indeed they say not to use the vbar version for HW V4 to get the telemetry. Also there is a HW V5 telemetry [page](https://manual.spirit-system.com/index.php?title=HobbyWingV5_Telemetry/cs).

In the HW firmare [page](https://www.hobbywing.com/en/service/download.html?id=5&cid=81), for V4, they say that the vbar version update is related to telemetry. I'm curious what telemetry sends the vbar version for HW V4. I guess is similar to HW V5.

Could you enable debug (disconnect bec power to rp2040 when debugging) for the HW 120A to check if something is received.

Timo-72, 2024-07-03:
Hello, I see no benefit to this. I do not know why you hunt no for VBAR. Every VBAR user has a VSTABI Neo or EVO. This system is the "mercedes" and everything is perfectly integretaded  bewtween ESC, flybarless and transmitter. But, flybarless 280€ and transmitter 1200€. Everybody who has this system will not use msrc. All other people using a V4 ESC can use Hobbywing firmware without Vbar. For Hobbywing V5 new protocol is integrated. So where do you see the benefit?

For sure I can do the debug, but it means to again open 4 screws that are closed with loctide because USB-C is not reachable from RP2020. So do you see a benefit to Debug Vbar protovol from Hobbywing?

dgatf, 2024-07-03:
I know that. But the target is not to give support to vbar users. The target is to use a telemetry protocol that provide data that does not need to be transformed. No vbar firmware provide raw sensor data and vbar firmware provide telemetry in final units (V, A, C). The transformation is different depending on model, so you need to measure voltage and current to verify that telemetry values are correct. I see much benefit on this.

Note. I've assumed that the v5 protocol is vbar protocol, but that's not confirmed.

Timo-72, 2024-07-04:
Hello, I send now again an email to Hobbywing support. Maybe we get the information. Would be easier as debugging. I will tell you when I have feedback.

Timo-72, 2024-07-08:
Hello, no feedback so far from Hobbywing. But normally Hobbying support gives answer. It can take view days. This is just for your info.

Timo-72, 2024-07-13:
[HW 120A V4 Platinum latest VBAR Firmware.txt](https://github.com/user-attachments/files/16200174/HW.120A.V4.Platinum.latest.VBAR.Firmware.txt)

Attached log file HW Platinum 120A V4 latest Vbar Firmware
Please recognize, this is only interim connection to the other components , so no connection to motor, this means no Ampere measure and no RPM, but Volt and Temperature should be visible. But I do see nothing. I do not know if I can try that again. As said, I have to dissamble part always my heli. In addition this HW 120A I am selling if something buys it. And still no feedback from HW. best regards

closed 2024-10-07 by dgatf