dgatf/msrc #131: Analog Temperature Sensor

Pablo18011, 2024-07-17:
Hi, I have connected a 100kNTC and resistor circuit to my RP2040 as per the diagram.
I have confirmed this is works in the GUI in debug mode, however when I discover new sensors on my Taranis it only 
diiscovers Custom sensor EscT 0B70 and this does not out put temperature on the Taranis.
Can anyone help?
I have the voltage divider circuit and this works perfectly, I want to be able to monitor a glider motor temperature directley on
the motor housing and not the ESC.
My set up is a RX R9MM using F.Port flex firmware.

dgatf, 2024-07-17:
Please post the debug

Pablo18011, 2024-07-17:
Is tihis what you need?
This is with no lipo connected to the voltage sensor circuit.
USB. Debug enabled
USB (119)
Voltage (220): 3.34
Temperature (224): -24.20
USB (119)
Voltage (104): 1.31
Temperature (108): -24.46
USB (119)
Voltage (104): 2.87
Temperature (108): -24.91
USB (119)
Voltage (104): 1.02
Temperature (108): -25.04
USB (119)
Voltage (104): 3.19
Temperature (108): -25.15
USB (119)
Voltage (104): 2.58
Temperature (108): -25.60
USB (119)
Voltage (104): 3.81
Temperature (108): -25.85
USB (119)
Voltage (104): 2.13
Temperature (108): -25.93
USB (119)
Voltage (104): 2.32
Temperature (108): -25.76
USB (119)
Voltage (104): 2.04
Temperature (108): -25.62
USB (119)
Voltage (104): 1.58
Temperature (108): -25.50
USB (119)
Voltage (104): 3.76
Temperature (108): -25.53
USB (119)
Voltage (104): 2.29
Temperature (108): -25.54
USB (119)
Voltage (104): 4.24
Temperature (108): -25.93
USB (119)
Voltage (104): 1.67
Temperature (108): -25.99
USB (119)
Voltage (104): 2.80
Temperature (108): -25.81
USB (119)
Voltage (104): 1.34
Temperature (108): -26.28
USB (119)

Thanks

Pablo18011, 2024-07-17:
Rs is a 2K resistor instead off a 10k as the latter would max out at 32C.
With a 2k and 100kNTC I have enough range(output) to measure upto 100C.
I have tested this in debug mode and everthing works fine.
Basically I'm trying to measure the temperature of a motor on a glider, so I have warning when it gets over 70C.
As it's a pusher prop used for launch, it does not have the cooling advantage of a tractor configuration.

dgatf, 2024-07-17:
Msrc is not receiving telemetry requests from the receiver. Check the circuit at pins 0 & 1

Are you receiving RxBt in your transmitter?

Pablo18011, 2024-07-17:
Thanks, I'll check can get back to you.

Pablo18011, 2024-07-18:
OK I have attached another debug, the previous one I deliberately removed the 5V supply to the RX, this why there was no RxBt telemetry.This time I connected the RP2040 to the GUI and applied Vbat to the Rx R9M and ran debug.
My Taranis discovers the sensor EscT 0B70, however there is no variation in displayed "C", when I Offset it to read 20C, it just 
stays at this value.

USB. Debug enabled
USB (137)
Smartport (202) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport. Sensor (228) > 10 0 9 68 4 0 0 7A 
Smartport (128) < 7E 71 
Smartport. Sensor (244) > 10 70 B 0 0 0 0 74 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Voltage (220): 11.37
Temperature (224): -23.63
USB (137)
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport. Sensor (106) > 10 0 9 71 4 0 0 71 
Smartport (128) < 7E 71 
Smartport. Sensor (122) > 10 70 B 0 0 0 0 74 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Voltage (104): 11.28
Temperature (108): -23.88
USB (137)
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport. Sensor (102) > 10 0 9 68 4 0 0 7A 
Smartport (128) < 7E 71 
Smartport. Sensor (122) > 10 70 B 0 0 0 0 74 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Voltage (104): 11.58
Temperature (108): -24.19
USB (137)
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport. Sensor (102) > 10 0 9 86 4 0 0 5C 
Smartport (128) < 7E 71 
Smartport. Sensor (122) > 10 70 B 0 0 0 0 74 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Voltage (104): 11.36
Temperature (108): -24.57
USB (137)
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport. Sensor (102) > 10 0 9 70 4 0 0 72 
Smartport (128) < 7E 71 
Smartport. Sensor (122) > 10 70 B 0 0 0 0 74 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Smartport (128) < 7E 71 
Voltage (104): 11.27
Temperature (108): -24.76
[... 210 lines cut ...]
Smartport (120) < 7E 71 
Smartport (120) < 7E 71 
Smartport. Sensor (102) > 10 0 9 71 4 0 0 71 
Smartport (120) < 7E 71 
Smartport. Sensor (122) > 10 70 B 0 0 0 0 74 
Smartport (120) < 7E 71 
Smartport (120) < 7E 71 
Smartport (120) < 7E 71 
Smartport (120) < 7E 71 
Voltage (104): 11.31
Temperature (108): -28.97
USB (137)
Smartport (120) < 7E 71 
Smartport (120) < 7E 71 
Smartport (120) < 7E 71 
Smartport. Sensor (102) > 10 0 9 6B 4 0 0 77 
Smartport (120) < 7E 71 
Smartport. Sensor (122) > 10 70 B 0 0 0 0 74 
Smartport (120) < 7E 71 
Smartport (120) < 7E 71 
Smartport (120) < 7E 71 
Smartport (120) < 7E 71 
Voltage (104): 11.62
Temperature (108): -28.25
USB (137)

Pablo18011, 2024-07-18:
This is my circuit below, all I added is a 100K NTC and 2K as per the circuit in the GUI

[image]

dgatf, 2024-07-18:
MSRC is not sending negative values for smartport. If a telemetry value is less than 0, it sends 0.

Please upload the updated [binary](https://drive.google.com/file/d/1tT9Y5Io5eOpDXTEIZ1WsEnx9hMCWKb7Y/view?usp=sharing).

Let me know if this fixes the issue.

Pablo18011, 2024-07-18:
Thanks you so much for your time, I will test and get back to you :)

Pablo18011, 2024-07-18:
OK, a quick test and that's working, I'm now getting a C reading change on my Taranis radio, awesome!
I have attache a debug, it's bouncing around a bit and I need to play around with the Offset, but it's working...
Thanks again...

USB. Debug enabled
USB (137)
Smartport (200) < 7E 71 
Temperature (222): -27.85
Voltage (220): 11.28
Smartport (130) < 7E 71 
Smartport. Sensor (228) > 10 0 9 68 4 0 0 7A 
Smartport (130) < 7E 71 
Smartport. Sensor (244) > 10 70 B E4 FF FF FF 8F 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Temperature (110): -26.91
Voltage (108): 11.29
USB (137)
Smartport (130) < 7E 71 
Temperature (110): -26.56
Voltage (108): 11.32
Smartport (130) < 7E 71 
Smartport. Sensor (108) > 10 0 9 6C 4 0 0 76 
Smartport (130) < 7E 71 
Smartport. Sensor (122) > 10 70 B E5 FF FF FF 8E 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Temperature (110): -26.28
USB (137)
Smartport (130) < 7E 71 
Voltage (108): 11.27
Temperature (110): -25.83
Smartport (130) < 7E 71 
Smartport. Sensor (108) > 10 0 9 67 4 0 0 7B 
Smartport (130) < 7E 71 
Smartport. Sensor (122) > 10 70 B E6 FF FF FF 8D 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Voltage (108): 11.26
Temperature (110): -25.91
USB (137)
Smartport (130) < 7E 71 
Voltage (108): 11.30
Smartport (130) < 7E 71 
Smartport. Sensor (108) > 10 0 9 6A 4 0 0 78 
Smartport (130) < 7E 71 
Smartport. Sensor (122) > 10 70 B E6 FF FF FF 8D 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Smartport (130) < 7E 71 
Temperature (110): -25.84
Voltage (108): 11.61
USB (137)
Smartport (130) < 7E 71 
Temperature (110): -24.99
Voltage (108): 11.77
Smartport (130) < 7E 71 
Smartport. Sensor (108) > 10 0 9 99 4 0 0 49 
Smartport (130) < 7E 71 
[... 137 lines cut ...]
Smartport (122) < 7E 71 
Smartport (122) < 7E 71 
Smartport (122) < 7E 71 
Voltage (108): 11.76
Temperature (110): -24.21
USB (137)
Smartport (122) < 7E 71 
Smartport (122) < 7E 71 
Voltage (108): 11.28
Temperature (110): -23.48
Smartport (122) < 7E 71 
Smartport. Sensor (100) > 10 0 9 68 4 0 0 7A 
Smartport (122) < 7E 71 
Smartport. Sensor (120) > 10 70 B E9 FF FF FF 8A 
Smartport (122) < 7E 71 
Smartport (122) < 7E 71 
Smartport (122) < 7E 71 
Smartport (122) < 7E 71 
Voltage (108): 11.28
Temperature (110): -23.24
USB (137)
Smartport (122) < 7E 71 
Smartport (122) < 7E 71 
Temperature (110): -24.18
Voltage (108): 11.28

dgatf, 2024-07-18:
Great.

Added to main branch with commit 58b602145d09a89a8d9fe3656d04932c7bc4e5d9

Binary updated.

closed 2024-07-18 by dgatf

dgatf, 2024-07-18:
> it's bouncing around a bit 

To avoid that, you can increase the averaging elements for temperature from msrc_gui.

[image]

Pablo18011, 2024-07-18:
I have increased it to 10 and that has settled it down.
I will experiment over next few days and run the motor up with a larger prop and gauge the read out with a infrared thermonitor on the motor and compare output on the radio.It's exactly what I needed to monitor a pusher plane motor to keep it in the green and avoid burning out a motor or damaging the magnets.
Thanks again for the help, much appreciated :)