dgatf/msrc #132: Disable the Thermal Protection

ghiyats07, 2024-07-27:
Hi! I'm sorry,i am not really good at this.I have a question,can i change the max temperature of my thermal protection in HW V4 by following the steps? I'm still confused how to implement the script to my esc.Thank you!

dgatf, 2024-07-27:
I dont know if there is any way to change the maximum temperature protection. But MSRC doesnt make any change on the esc configuration.

closed 2024-07-27 by dgatf

ghiyats07, 2024-07-27:
> I dont know if there is any way to change the maximum temperature protection. But MSRC doesnt make any change on the esc configuration.

Thanks for the reply, so if i can't change the maximum temp. I also can't disable the thermal protection right?

dgatf, 2024-07-27:
I think so, but this is not related to MSRC. You may ask Hobbywing.