dgatf/msrc #134: Sending telemetry data to S.Port stops after around a min.

CarlouT, 2024-09-01:
Hi,

I manage to make this project partially work. The status led of the board blinks indicating that it is  sending telemetry. But this is not persistent. It stops at some point in time (around after 1 min.). If I press the reset button, it will start sending telemetry data again for around a min then it will stop again. 

Any help would be appreciated.
[image]

Regards,
Carlou

CarlouT, 2024-09-02:
Just an update on this issue. If I disconnect the TELE wire from the Hobbywing ESC to the RP2040 board, the sending of the telemetry data from the RP2040 board to the receiver does not cut-off (does not stop). If I put the TELE wire back, then the issue is back wherein it will only send data for aproximatly a min then it stops (Blue LED stops blinking). I looks like the root cause is coming from the hobbywing ESC. I already tried loading older firmware version of this ESC but it did not solve the issue. Kindly help.

CarlouT, 2024-09-02:
Hi,

I manage to fid the issue. The board is defective. I purchse a new board. Now its working as expected.

Regards,
Carlou

dgatf, 2024-10-06:
Hi,

Thanks for the feedback. Indeed there was an issue with current auto offset task (If auto offset is disabled there is no issue). Fixed with commit c4281ef96926d4eee9981873dd612cd397d0c6d6

closed 2024-10-06 by dgatf