dgatf/msrc #135: No data from Hobbywing FlyFun 80A

Kat-Man-DEW, 2024-09-12:
I have a Hobbywing FlyFun 80A and FrSky SR6 receiver and FrSky X20 Pro AW transmitter. The ESC telemetry fields are in the telemetry section in my transmitter but I'm not getting any data. I ran the config and tried both of the Hobbywing settings. I'm very familiar with wiring these type devices.

This is a great project. I have several Hobbywing ESC's I would like to use and I would donate if I could get it to work. I am unable to logon to the phpBB site. I have requested my password several times but I never receive an email. I checked my Junk and Deleted items folders.

dgatf, 2024-10-06:
Check that your HW esc does not have the vbar firmware version. You'll need a hw programmer for that.

If you know that you have the correct firmware version and still doesn't show esc telemetry, please disconnect the power from your model to msrc connect to usb, enable debug from msrc_gui and post the output.

dgatf, 2024-10-07:
I've added a serial monitor to msrc. You can also monitor whatever the esc is sending (if anything). Connect the esc to GPIO5 as usual and select *Receiver protocol->Serial Monitor* with the following parameters and post output:

- Baudrate: 19200
- Stop bits: 1
- Parity: None
- Timeout: 1 ms
- Inverted: Unchecked

Kat-Man-DEW, 2024-10-10:
Thank you very much for the reply. Do I need to connect power to the ESC? Does a receiver need to be connected? I only have the ESC connected to GPIO5.

Where do I select Receiver protocol->Serial Monitor?

Here is what I get when I enable Debug;
USB. Debug enabled
USB (139)
USB (138)
USB (138)
USB (138)
USB (138)
USB (138)
USB (138)
USB (138)
USB (138)
USB (138)
USB (138)

dgatf, 2024-10-10:
With your esc assembled in you model, connect the esc to gpio5 and gnd from telemetry port. Do not connect power from rp2040 to esc.  Connect rp2040 to usb and execute msrc_gui. From there select receiver protocol serial monitor and enable debug.

frutsch, 2024-10-10:
Same issue to me with a Hobbywing Platinum 80A V4. Latest VBar Software.Telemetrie Datas Always 0. In Transmitter (X20R)

dgatf, 2024-10-10:
> Latest VBar Software

You have to use the no vbar firmware version (04.0.04)

The vbar version is 04.0.05

This is for model platinum v4 80a

dgatf, 2024-10-10:
Chedk you have selected the correct model, Platinum PRO v4 80A. If not update config and reboot. 

If the current value is still wrong, adjust current multipler to get the correct value.

[image]

dgatf, 2024-10-10:
Regarding the reset, please try this [binary](https://drive.google.com/file/d/1faqcZKSOszqqAFpy5cTzvigTfy1Aoi6F/view?usp=sharing).

It delays the hw esc driver for 5 seconds at boot, in case the esc is sending something at boot that msrc doesn't like. If this works, will adjust the delay later.

dgatf, 2024-10-10:
> Regarding the reset, please try this [binary](https://drive.google.com/file/d/1faqcZKSOszqqAFpy5cTzvigTfy1Aoi6F/view?usp=sharing).

Just updated now.

frutsch, 2024-10-10:
Binary looked good, startet now allone before my rp2040 burned down... Have to build a new one, will come Back tomorrow

dgatf, 2024-10-10:
Damn, I forgot that there is already an option to delay hw4/flyfun. Just click _Init delay._ You can se in previous piccure

dgatf, 2024-10-10:
Damn, I forgot that there is already an option to delay hw4/flyfun, but it was deactivated. I've fixed it. Just click _Init delay_. You can se in the previous screenshot. When enabled it delays the esc driver for 15s

frutsch, 2024-10-11:
Now everything ist ok, except the Amp readings jump to 100A as i switch on Motor.
No Idea why, current amplifier ist 500 but can't change It to other numbers.

dgatf, 2024-10-11:
> Now everything ist ok, except the Amp readings jump to 100A as i switch on Motor.

That's  normal if you are using auto offset for the esc. It should go to 0A after 15 seconds. Alternatively you can set the offset manually if auto offset doesn't work.

>  No Idea why, current amplifier ist 500 but can't change It to other numbers.

You should be able to change it. If it is greyed out in msrc_gui, download again as that was  a bug.

closed 2024-10-11 by dgatf

Kat-Man-DEW, 2024-10-11:
> With your esc assembled in you model, connect the esc to gpio5 and gnd from telemetry port. Do not connect power from rp2040 to esc. Connect rp2040 to usb and execute msrc_gui. From there select receiver protocol serial monitor and enable debug.

Here is my serial monitor debug output. I have a Hobbywing LED Program Card and Multifunction LCD Program Box (but not the one with a wheel on their website), but I don't know how to check or change my firmware. And I haven't been able to find any ESC firmware on their website.

USB. Debug enabled
USB (126)
USB (104)
USB (104)
USB (104)
USB (104)

dgatf, 2024-10-11:
You probably have the vbar version. You need to flash a no vbar version with the programmer.

Kat-Man-DEW, 2024-10-12:
> You probably have the vbar version. You need to flash a no vbar version with the programmer.

I don't know how to do that.