dgatf/msrc #136: Vario with GPS and ESCno t working

Klippeneck, 2024-09-16:
Hello,
I have built a msr 2040 sensor for Jeti.
Configuration is for Fun Fly ESC, GPS and vario.
ESC and GPS seem to work, the vario does not show any current values.
If the GPS is not configured, the vario works, but the displayed values are not in the
usual Jeti form, i.e. with msr 1m/s as the smallest value, with Jeti
0.01 m/s, so not usable. Altitude seems to be correct.
Does anyone else have similar problems?

dgatf, 2024-10-06:
This is probably related to #134 . To check if it is the same issue, disable auto offset for the esc and the issue should dissapear.

If so, flash updated [binary](https://drive.google.com/file/d/1pzIt4XdcLxiSUfSTfyoh_w5tLTFd9JN2/view?usp=drive_link)that fixes the issue.

dgatf, 2024-10-06:
The number of sensors for Jeti was limited to 32. I've increased to the maximum per Jetiex spec: 256

I've also added 2 decimals for vspeed.
And added a string for msrc. It should appear the name of the sensor as MSRC in transmitter.

Flash updated [binary](https://drive.google.com/file/d/1pzIt4XdcLxiSUfSTfyoh_w5tLTFd9JN2/view?usp=drive_link).

Please provide feedback as I don't have a Jeti system to check.

dgatf, 2024-10-07:
I've been reviewing JetiEx spec and it turns out that a a _range of identifiers_ (which is MSRC for Jeti receiver) can have a maximum of 16 sensors. Which explains your issue: not more than 16 sensors are allowed under the same _range of identifiers_.

The only way to have add sensors is to use E2 or a Jeti Telemetry Expander. I suggest is to use a second rp2040 and connect it to E2 or a Jeti Telemetry Expander.

Implementing a second _range of identifiers_  on MSRC is not straight forward as it involves using an additional uart that would be connected to E2 or expander.

Also, I have removed the mute (it was muting 50% of packets in case other sensors were attached). But that's not the topology of the ex bus.

closed 2024-10-07 by dgatf