dgatf/msrc #138: Vario wroks not korekt with Jeti

Klippeneck, 2024-10-07:
I have tested the changes, unfortunately without success.
The sensor name does not appear in the telemetry display.
Only 2 values appear for the vario, air temperature and abs. altitude, vario value is missing.
The vario value only appears when the additional GPS is configured, but is somewhat unstable.
See screenshots of the display.

[image]
[image]
Jürgen

Klippeneck, 2024-10-07:
Sensor value jumps between +-0.5m/s.
Yes, the v Speed value now has 2 digits after the decimal point.
But the v Speed value only appears when the GPS is activated, with vario only there is no v Speed value. 
Jürgen

dgatf, 2024-10-07:
There was some issues with the buffer creation due to the changes. Hopefully it is fixed now with commit 30073882f457e076e86d986697b8759b7fdda04b

Please flash updated [binary](https://drive.google.com/file/d/1pzIt4XdcLxiSUfSTfyoh_w5tLTFd9JN2/view?usp=drive_link).

Klippeneck, 2024-10-08:
The change has not been successful, on the contrary.
Now there are no values at all for the vario and also for GPS.
In addition, the sensor does not always start at Power On and can only be brought back to life by pressing the reset button on the RP 2040 several times. Only then does the LED light up.
Jürgen

dgatf, 2024-10-08:
I've reverted some of the changes. Changes now are only: no mute and maximum sensors 15

If this works (it should), I'll leave as is an forget about the sensor name.

Flash updated [binary](https://drive.google.com/file/d/1pzIt4XdcLxiSUfSTfyoh_w5tLTFd9JN2/view?usp=sharing)

Klippeneck, 2024-10-08:
With this version, the vario works with all 3 values.
Temperature and altitude seem plausible, the vario value jumps between plus minus 0.56 m/s with steps of 0.18 m/s, i.e. 0.18, 0.37 and 0.57 m/s positive and negative respectively.
There are no intermediate values.
Jürgen

dgatf, 2024-10-08:
Ok. Good. So the only change removed is the sensor name.

I'll check the vario resolution. To smooth vario values you can increase the number of averaging elements for vario.

[image]

dgatf, 2024-10-08:
Vspeed works better when moving. But also if the interval used to calculate vspeed is low, it makes vspeed more sensitive to noise. For now this parameter is hardcoded but I could populate yo msrc_gui.

See the [interval](https://github.com/dgatf/msrc/blob/ce9af81b8b2b147f742d75e4fe3e0a495ce78b87/board/project/sensor/bmp280.c#L58)

The interval is 500ms, but it could be increased to 1 or 2 seconds to smooth vspeed

closed 2024-10-08 by dgatf