dgatf/msrc #140: Support for dualsky summit x (Kiss)

floflolapin, 2024-10-11:
Hi
I would like to use a dualsky summit x esc.
Apparently it follows kiss 24a telemetry protocol.
Is there way to make it work with msrc?
Thanks
Florian

dgatf, 2024-10-11:
Yes, it should work with *Esc Protocol->APD F*

In case it doesn't work, I've added a serial monitor at GPIO5. Connect esc to GPIO5 as usual and select *Receiver protocol->Serial Monitor* with the following parameters and post output:

- Baudrate: 115200
- Stop bits: 1
- Parity: None
- Timeout: 3 ms
- Inverted: Unchecked

floflolapin, 2024-10-13:
Hi
Thanks for your help.
It works except for current which stays at zero.

Here is the debug output with serial monitor. I hope I understood what you needed:

USB. Debug enabled
USB (104)
Serial monitor (243). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
USB (104)
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
Serial monitor (150). Length: 10 (31 ms):  0x20 0x5 0xEC 0x0 0x0 0x0 0x0 0x0 0x0 0xA3
[... 690 lines cut ...]
[... cut ...]
Serial monitor (142). Length: 10 (31 ms):  0x22 0x5 0xEA 0x0 0x0 0x0 0x0 0x0 0x0 0x9D
Serial monitor (142). Length: 10 (31 ms):  0x22 0x5 0xEA 0x0 0x0 0x0 0x0 0x0 0x0 0x9D
Serial monitor (142). Length: 10 (31 ms):  0x22 0x5 0xEA 0x0 0x0 0x0 0x0 0x0 0x0 0x9D
Serial monitor (142). Length: 10 (31 ms):  0x22 0x5 0xEA 0x0 0x0 0x0 0x0 0x0 0x0 0x9D
Serial monitor (142). Length: 10 (31 ms):  0x22 0x5 0xEA 0x0 0x0 0x0 0x0 0x0 0x0 0x9D
Serial monitor (142). Length: 10 (31 ms):  0x22 0x5 0xEB 0x0 0x0 0x0 0x0 0x0 0x0 0x42

dgatf, 2024-10-13:
Bytes for current and consumption are all 0. Maybe your model doesn't have a current sensor. Or the current is too low as I see that rpms are 0 as well.

floflolapin, 2024-10-13:
that's strange because RPM is working (although it's too high)
Here is another log where I change from Frsky S Port to serial monitor and went full throttle both time.
You can see that RPM is changing with S Port protocol:

USB. Debug enabled
USB (104)
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.01 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.75
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.01 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.75
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.03 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
USB (104)
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
Apd F (102) < Rpm: 0 Volt: 15.02 Curr: 0.00 Temp: 37 Cons: 0 CellV: 3.76
[... 882 lines cut ...]
Serial monitor (142). Length: 10 (31 ms):  0x29 0x5 0xC8 0x0 0x0 0x0 0x0 0x0 0x0 0xCF
Serial monitor (142). Length: 10 (31 ms):  0x29 0x5 0xC9 0x0 0x0 0x0 0x0 0x0 0x0 0x10
Serial monitor (142). Length: 10 (31 ms):  0x29 0x5 0xC9 0x0 0x0 0x0 0x0 0x0 0x0 0x10
Serial monitor (142). Length: 10 (31 ms):  0x29 0x5 0xC9 0x0 0x0 0x0 0x0 0x0 0x0 0x10
Serial monitor (142). Length: 10 (31 ms):  0x29 0x5 0xC9 0x0 0x0 0x0 0x0 0x0 0x0 0x10
Serial monitor (142). Length: 10 (31 ms):  0x29 0x5 0xC8 0x0 0x0 0x0 0x0 0x0 0x0 0xCF
USB (104)
Serial monitor (142). Length: 10 (31 ms):  0x29 0x5 0xC9 0x0 0x0 0x0 0x0 0x0 0x0 0x10
[... cut ...]
Serial monitor (142). Length: 10 (31 ms):  0x29 0x5 0xCB 0x0 0x0 0x0 0x0 0x0 0x0 0xA9
Serial monitor (142). Length: 10 (31 ms):  0x29 0x5 0xCA 0x0 0x0 0x0 0x0 0x0 0x0 0x76
Serial monitor (142). Length: 10 (31 ms):  0x29 0x5 0xCB 0x0 0x0 0x0 0x0 0x0 0x0 0xA9
Serial monitor (142). Length: 10 (31 ms):  0x29 0x5 0xCB 0x0 0x0 0x0 0x0 0x0 0x0 0xA9
Serial monitor (142). Length: 10 (31 ms):  0x29 0x5 0xCB 0x0 0x0 0x0 0x0 0x0 0x0 0xA9
Serial monitor (142). Length: 10 (31 ms):  0x29 0x5 0xCB 0x0 0x0 0x0 0x0 0x0 0x0 0xA9

dgatf, 2024-10-13:
It seems that your esc doesn't have a current sensor. For the rpms you have to set pair of poles.  Also main and pinion teeth if using an heli. You can do in msrc or in your transmitter. To get best resolution for rmps, set those values in transmitter.

closed 2024-10-13 by dgatf

floflolapin, 2024-10-13:
Ok thanks for your help.
I'll check with the manufacturer