dgatf/msrc #141: Any intentions on adding converting Futaba SBUS2 to Frsky S-Port

doloebig, 2024-11-19:
more Sensors and ESC talking only SBUS2. For ex. D-Power AVICON PRO 
Is there a possibility to convert the telemtrie data to S-Port? using SBUS2 as sensor and S-Port as the target protocol

Regards
Doro

dgatf, 2024-11-22:
It can be done. We need to read the communications between the esc and the receiver.

In msrc_gui receiver protocol select _Serial monitor_ with the following parameters:

- Baudrate: 100000
- Stop bits: 2
- Parity: even
- Inverted: true (checked)

Post msrc_gui log output.

Make the following circuit:

[image]

doloebig, 2024-11-22:
sorry i have no SBUS2 RX

doloebig, 2024-11-23:
the SBUS2 is well documented
https://sites.google.com/site/sbus2diy/futaba-sbus2-%D0%BF%D1%80%D0%BE%D1%82%D0%BE%D0%BA%D0%BE%D0%BB/sbus2-protocol-%D0%BF%D1%80%D0%BE%D0%B4%D0%BE%D0%BB%D0%B6%D0%B5%D0%BD%D0%B8%D0%B5

doloebig, 2024-11-23:
here is a libery
https://github.com/BrushlessPower/SBUS2-Telemetry
hope this is helpfull