dgatf/msrc #143: PWM speed reading problem

ahfynjj, 2024-12-01:
Does MSRC support Hall sensor, or directly connect to the engine's CDI to read the speed? I read the closed issue and it was mentioned in the previous content that GPIO5 is connected to CDI, but the figure in the MSRC config program shows that it is connected to GPIO6. I tried to connect it to the RPM line of hobbywing platium V4 60A and rotate the motor. The ESC RPM sent back did not show the reading. Is there any error in my operation? Please advise.
Please see the debug log

In addition, I made a simple PCB to convert the 8.4V of the ESC into 5V to power the RP2040, and integrated resistors to facilitate the connection of different remote control systems, as well as ESC and GPS interfaces. After it is improved, it will be open source for everyone to make and use. Thank you

[1.txt](https://github.com/user-attachments/files/17969157/1.txt)

[image]

[image]

dgatf, 2024-12-01:
Sorry, there is a typo in the docs. It is [GPIO4](https://github.com/dgatf/msrc/blob/139d373068d91d1269c0da32e6d7e80adbfc6857/board/project/constants.h#L16).

ahfynjj, 2024-12-02:
It worked, thanks