dgatf/msrc #153: Contact sensor for FrSky telemetry

Richrfl, 2025-04-17:
Is there any possibility to add to the code several independent contact sensors for FrSky telemetry? The idea behind the contact sensor is to be used, for example, for a stall sensor, gears fully up or down and maybe some other application that I do not have on top of my mind right now.

dgatf, 2025-04-17:
Yes, it is feasible as there are some spare pins. So, msrc reads the state (low/high) and sends a value (0/1)  for every pin. Is this correct?

Richrfl, 2025-04-17:
Yes,  that is the idea.

dgatf, 2025-04-17:
I plan to use 6 gpios (17 to 23) for this. I prefer to send all the info in one telemetry packet, rather than 1 packet per gpio. If the data is send in one telemetry packet, it will be masked. E.g. if gpios 17 and 19 are high, it will send 5 (in binary 000101). Will this work for you?

Richrfl, 2025-04-17:
Wow,  dgat! You are overestimating me. Thank you. I never tried to go into the telemetry code.  I do understand how data com protocols work,  but I never went into the frsky telemetry.  With that said,  I suggest your using the best procedure to your criteria and I will be happy to implement and test it.  I should have an extra board at hand for that purpose.

dgatf, 2025-04-17:
Ok. I've added gpio sensors with [commit](e3feac62d019e73ba7923293c639f93d9e56a765)

You can enable gpio sensor from msrc_gui and set the reading interval in ms (from 10 to 10000).

It will send the state of the pins (17 to 22) as dataId 0x5100. The value are the state of the pins masked.

Partially tested.

dgatf, 2025-04-17:
You can use logical switches and special functions in Opentx to play a sound when a telemetry value changes.

Or you can create a lua script to show individual pin states on the lcd.

Richrfl, 2025-04-17:
Well, I guess like I have to use the bigger board, as the zero does not have pins 16 through 25. Not a big deal, thought. Also, I am using Ethos, and like with OpenTx, I have the means to use voice or represent a widget on the screen.
Thank you, Daniel, for your help. If I have any issues, I will come back to this thread. I will order the big board, and I am supposed to get it by Monday.

[image]

dgatf, 2025-04-17:
You have those pins in the waveshare, on the backside, but without holes.

See [rp2040 waveshare pinout](https://grobotronics.com/images/companies/1/RP2040-Zero-details-7.jpg?1638179513717)

Richrfl, 2025-04-17:
Wow!! Thank you!!

dgatf, 2025-04-17:
> Daniel the GUI the MSRC config cannot be scrolled. I am missing pat of it. FYI I am running windows 11.

You have to connect to the rp2040 to be able to scroll.

Richrfl, 2025-04-18:
OK. Thank you.

Richrfl, 2025-04-18:
Daniel, there is a document from which I copied some parts that have all information about the board pin assignments, voltages, etc, as shown below. Can you please remind me of the link to such doc?
Status led of the board blinks when sending telemetry. If it doesn't blink check connections and config.

RP2040 Zero pinout

[image]

RP2040 Zero has to be powered with 5v or less. For higher voltage, a voltage regulator is needed.
It can be powered from the telemetry port or from BEC. Some receivers deliver 5v or less at the telemetry port, even if BEC voltage is higher than 5v. Check that voltage supplied to RP2040 is not higer than 5v.
Press BOOT button during startup and paste the binary file [MSRC-RP2040.uf2](https://drive.google.com/file/d/1pzIt4XdcLxiSUfSTfyoh_w5tLTFd9JN2/view?usp=sharing) to RP2040 folder

Richrfl, 2025-04-18:
Daniel, disregard my previous message I found the doc.

Richrfl, 2025-04-18:
I am almost there. I just need a little help: Which number should I use as Sensor ID (2 digits), and final questions: the only thing, besides the ID, that I have to set is the Refresh? Now, for the Refresh, there is a list of parameters (voltage, current, rpm, etc.) Which ones should I use? Also, as a Sensor, there are all the standard sensors, which one should I use?
Thank you

dgatf, 2025-04-18:
You can choose any sensorid if there are no other sensors connected to smartport. The sensorid is to avoid conflicts with other sensors. You can check the sensorid of a telemetry value in opentx.

For gpio, the interval is the time elapsed for every update of the pins state and to send the telemetry packet.

Richrfl, 2025-04-20:
Daniel, I am sorry to be a pain in your neck, but I have a problem with the Zero board. After trying to solder a cable to the 17, 18 and 19 pads, I have to give up with that board. Either I do not have the skills or the tools to solder into those pads. Therefore, I have a question for you: Would the MSRC gui support either of these boards?  RP2040 Core Board Type-C USB-C for Raspberry Core Board 4MB RP2 for Raspberry Pi Pico Micropython or Raspberry Pi Pico RP2040 microcontroller Dual-core ARM Cortex M0+ Processor?

dgatf, 2025-04-20:
It works with all rp2040 boards.

Richrfl, 2025-04-20:
Trying to understand how the contact sensor will work and must be configured after re-reading your explanation: "E.g. if gpios 17 and 19 are high, it will send 5 (in binary 000101)." from here I understand two things: 1) even if I use 1 single pin for my application, (let's say p[in 17) all other pins must be forced to GND or 5V. 2) When the sensor is discovered, it will be a single sensor, which, in your example above, will show the number 000101, and from this number, I will have to single out which contacts are closed or open. Is that correct? Just FYI, I previously used your design for an ESC, and it worked perfectly with FrSky Ethos telemetry. The sensors were discovered immediately.

dgatf, 2025-04-20:
No need to ground unused pins. I've added a mask. Select the pins you'll use under GPIO in msrc_gui. The pins not selected will be 0.

E.g. If you enable pins 17 and 19 it will send 000X0Y, were X and Y is the state of pins 17 and 19. The rest will always be 0.

dgatf, 2025-04-20:
If yon don't see a check box for every pin under GPIO, is because you downloaded the initial gpio version. If so, download again msrc_gui and firmware.

Richrfl, 2025-04-20:
Something is wrong, Daniel. I just downloaded again the files: msrc_gui.exe and MSRC-rp2040.uf2 created three days ago and re-loaded the uf2 file to the board, the gui connects to the board, but doesn't show any GPIO, (or I don't get the idea)

[image]

[image]

dgatf, 2025-04-20:
Sorry, I think I forgot to launch github actions to publish new binaries. Please download again.

Richrfl, 2025-04-20:
OK

Richrfl, 2025-04-20:
No luck. I downloaded the gui.exe and the uf2 and I still got the same.

dgatf, 2025-04-20:
I'll review it tomorrow.

Richrfl, 2025-04-20:
OK. Thank you

dgatf, 2025-04-21:
I tested windows version and it's ok. See screenshot:

[image]

You have download latest [binaries](https://github.com/dgatf/msrc?tab=readme-ov-file#1-binaries). Not binaries from released versions.

Richrfl, 2025-04-21:
It seems I am not on the same page. I downloaded from here:

[image]

dgatf, 2025-04-21:
Anytime. From there you download the development (latest) binaries. On the next release this will be included and you would be able to download from the release page.

Richrfl, 2025-04-21:
Ok, thanks a lot for the clarification. I will get the new boards this afternoon; either late today or tomorrow, I should be ready to test.

Richrfl, 2025-04-21:
The new board is configured OK, but Ethos' Discovery new sensors do not find the GPIOs this time. Here are the config screens:

[image]

[image]

I am sure I am missing something, but this should be right.

dgatf, 2025-04-21:
Enable voltage sensor and check if it is discovered. No need to attach physical sensor.

Richrfl, 2025-04-21:
Reboot??? or Reset? Reboot will delete the uf2 file.
This is what I did:

[image]

Not discovered

dgatf, 2025-04-21:
It working ok. See [video](https://youtube.com/shorts/HHOy-ZxfkcE?feature=share)

And photo:

[image]

5100 is the gpio (with gpio 18 high) and A3 voltage.

dgatf, 2025-04-21:
Check connections and be sure you also update the firmware. If using the released firmware it won't work with gpio.

Richrfl, 2025-04-21:
I downloaded the fw from the Binary that you suggested. I am using a different board than yours. I think I have a very basic problem, as when I go to the debug tab, I don't get anything, meaning that RX and board are not talking to each other. Can you please publish a picture of the breadboard? It is hard to see the connections in the video. Thank you.

Richrfl, 2025-04-21:
One little step ahead. Running the debug, without the Sport connected, I get this:

[image]
When I connect the SPort, I get this:

[image]

Now, the discover found ADC3, which has to be the voltage enabled on the board.

dgatf, 2025-04-21:
And 5100 is the gpio sensor. Is it discovered?

Richrfl, 2025-04-21:
No it didn't. I disabled the voltage and I got this:

[image]

GPIO shows 0x1 when pin 17 is 5V and 0x0 when it is 0V.

Richrfl, 2025-04-21:
Brillant!!!! It works!!!! Thanks a lot!!!!

closed 2025-04-21 by Richrfl